espressif/esp-idf #5470: Allow unlimited HCI Extended Features Pages by default

DerfJagged, 2020-06-19:
**Is your feature request related to a problem? Please describe.**

Yes. When you connect an Xbox One S controller to the ESP32 by running the esp_hid_host example, it errors out in `btm_read_remote_ext_features_complete` due to HCI_EXT_FEATURES_PAGE_MAX being set to 2 when the Xbox One controller has 3 pages. If you comment out the `return` line, it works great.

**Describe the solution you'd like**

To not `return` when pages > 3. HCI_EXT_FEATURES_PAGE_1 and HCI_EXT_FEATURES_PAGE_2 are unhandled in code anyway (empty functions), so I don't see the harm in allowing any number of pages to maximize compatibility. 

**Describe alternatives you've considered**

Give a warning instead of returning. 

Thank you!

Wth-Esp, 2020-06-22:
Hi @DerfJagged 

We agree with you about the fix, so would you like to commit a PR?

Thanks
Weitianhua

DerfJagged, 2020-06-22:
@Wth-Esp done! Thank you.

https://github.com/espressif/esp-idf/pull/5481

closed 2020-06-22 by DerfJagged