espressif/esp-idf #5471: unable to compile the on off server ble mesh example using nimble

ankit-thealchemist, 2020-06-19:
## Environment

- Development Kit:      esp devkit v1

- Module or chip used:  [ESP32-WROOM-32|
- IDF version (run ``git describe --tags`` to find it):
    // v4.0.1
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     [Windows]
- 
- Using an IDE?: |Yes (please give details)] vs code 
- Power Supply:         NA
## Problem Description

I was compiling the on off server ( https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server ) ble mesh using the nimble option enable.
I am unable to compile the program. The program throw the error in linking process. The error I got is 
[3/4] Linking CXX executable onoff_server.elf
FAILED: onoff_server.elf
cmd.exe /C "cd . && D:\IOTCode\framework\toolchain4.0\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\onoff_server.elf.rsp  -o onoff_server.elf  && cd ."
d:/iotcode/framework/toolchain4.0/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/bt/libbt.a(glue.c.obj): in function `net_buf_ref':
D:/IOTCode/framework/esp-idf-v4.0/components/bt/host/nimble/nimble/nimble/host/mesh/src/glue.c:95: multiple definition of `net_buf_ref'; esp-idf/bt/libbt.a(mesh_buf.c.obj):D:/IOTCode/framework/esp-idf-v4.0/components/bt/esp_ble_mesh/mesh_common/mesh_buf.c:287: first defined here
d:/iotcode/framework/toolchain4.0/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/bt/libbt.a(glue.c.obj): in function `net_buf_unref':
D:/IOTCode/framework/esp-idf-v4.0/components/bt/host/nimble/nimble/nimble/host/mesh/src/glue.c:111: multiple definition of `net_buf_unref'; esp-idf/bt/libbt.a(mesh_buf.c.obj):D:/IOTCode/framework/esp-idf-v4.0/components/bt/esp_ble_mesh/mesh_common/mesh_buf.c:301: first defined here
d:/iotcode/framework/toolchain4.0/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/bt/libbt.a(glue.c.obj): in function `net_buf_simple_pull_le16':
D:/IOTCode/framework/esp-idf-v4.0/components/bt/host/nimble/nimble/nimble/host/mesh/src/glue.c:166: multiple definition of `net_buf_simple_pull_le16'; esp-idf/bt/libbt.a(mesh_buf.c.obj):D:/IOTCode/framework/esp-idf-v4.0/components/bt/esp_ble_mesh/mesh_common/mesh_buf.c:172: first defined here

kindly help, how to compile the program correctly.

dhrishi, 2020-06-22:
@ankit-thealchemist Did you explicitly enable BLE Mesh from NimBLE configuration options? If yes, you should not do that. ESP BLE Mesh is Espressif's implementation of BLE Mesh which uses NimBLE (or Bluedroid) just as the BLE host. Please share your application's sdkconfig.

Also, in the environment you've mentioned that toolchain version is 5.2.0. Since, ESP-IDF v4.0 the toolchain is to be updated to v8.2.0. Please see the ESP-IDF documentation for more details.

ankit-thealchemist, 2020-06-22:
> @ankit-thealchemist Did you explicitly enable BLE Mesh from NimBLE configuration options? If yes, you should not do that. ESP BLE Mesh is Espressif's implementation of BLE Mesh which uses NimBLE (or Bluedroid) just as the BLE host. Please share your application's sdkconfig.
> 
> Also, in the environment you've mentioned that toolchain version is 5.2.0. Since, ESP-IDF v4.0 the toolchain is to be updated to v8.2.0. Please see the ESP-IDF documentation for more details.
> EDIT: The error messages show the toolchain version is correct, although please confirm.

thanks for the reply. Yes you are correct that I enable both esp ble mesh and also nimble mesh , thats why it wont compile.  I corrected my mistake and it is working fine now.

also about the toolchain I actually type the command xtensa-esp32-elf-gcc --version and whatever is the result I pasted without giving a thought.  Any ways very very thank you. I am closing the issue now.

Also I have little bit  query apart from this. Actually I am implementing he OOB method during provision. What ever I choose the OOB method I always get the option to provision the device with no OOB ( I am using the nrf mesh) do I missing something while enabling the the OOB.  I watched the video of the nrf mesh, where they show the the implementing the OOB method. In that video there is no option of provisioning with no OOB method.  for more information please look into the issue I open.  

https://github.com/espressif/esp-idf/issues/5281

closed 2020-06-22 by ankit-thealchemist

dhrishi, 2020-06-22:
@ankit-thealchemist Your question is now answered on that thread

ankit-thealchemist, 2020-06-22:
> @ankit-thealchemist Your question is now answered on that thread

thanks for the speedy resolution. You are awesome!!

kunj2308, 2025-01-02:
> ## Environment
> * Development Kit:      esp devkit v1
> * Module or chip used:  [ESP32-WROOM-32|
> * IDF version (run `git describe --tags` to find it):
>   // v4.0.1
> * Build System: [...]

Hello I have a question that for the nimble mesh , which changes I have to make in the setting or code of on_off_server which is on esp-idf?

ankit-bansal-oxit, 2025-01-02:
@kunj2308 I hardly remember, as it is 4.5 years old. But, please enable the nimble from the host side only not in the esp ble mesh. 
Thanks!

kunj2308, 2025-01-03:
> @kunj2308 I hardly remember, as it is 4.5 years old. But, please enable the nimble from the host side only not in the esp ble mesh. Thanks!

Thanks for reply , I get the solution by enabling the below option too and its work 

[image]

Currently, I am working on implementing the esp-ble-mesh network and I want to use it for home automation and now I am searching for BLE-OTA , if you have any code or material please provide to me , 

thanks

ankit-thealchemist, 2025-01-03:
@kunj2308 I suppose they have the component for ble fuota. Although I never tried by myself.

https://components.espressif.com/components/espressif/ble_ota/versions/0.1.14

Its not hard to implement the ota via ble by yourself. just have two characterstersics one for command and other for data. Like when you send start to the command characterstics, device would prepare itself for ota. then you can send the packets one by one on data characterstics which would be received by ota api itself. This would give you better control. 

Thanks!

kunj2308, 2025-01-03:
thanks for reply ,

I tried this but I got some errors while OTA though the mobile app is there any different method you know ?, else it's okay 

thank you !

ankit-thealchemist, 2025-01-03:
@kunj2308 For a client I have designed the file transfer library for esp32, using ble from above mehtod describe.  I donot remember how we transfer the file from app or python script. However the concept remains the same. 

Have you tired with bluedriod stack? maybe stack problem.