espressif/esp-idf #5472: BLE - GATT_INSUF_AUTHENTICATION

hassance3, 2020-06-19:
Code: Examples\bluetooth\bluedroid\ble\gatt_security_server

1 - Line 558: Set IO capability to ESP_IO_CAP_OUT -> `esp_ble_io_cap_t iocap = ESP_IO_CAP_OUT;`
2 - Bonding a BLE device (Phone) using PIN 123456.
3 - After successful bonding / connection, just disconnect.
4 - Remove all bonded device -> `remove_all_bonded_devices();`
5 - Use the phone and try to connect.

So, the phone (central) still bonding ESP32 (peripheral), but the ESP has just unbound the phone.

What will happen:
- The phone can connect without need of PIN, can get full services/profiles but can't perform any Read/Write/Indict/Notify request due to insufficient authentication. -> `BT_GATT: GATT_INSUF_AUTHENTICATION`

What should happen:
- The phone show pairing PIN message to bond/connect again.

Thank you,

hassance3, 2020-06-19:
Fixed by: `uint8_t auth_option = ESP_BLE_ONLY_ACCEPT_SPECIFIED_AUTH_ENABLE;`

closed 2020-06-19 by hassance3

chegewara, 2020-06-19:
If you would try to connect one more time then phone also should be unbouded.
Bluetooth specs says its not required to be paired/bounded to discover services.
iPhone specs says that phone should be able to connect without pairing and respond on pairing request only when try to read/write characteristics/descriptors that require pairing.

hassance3, 2020-06-19:
@chegewara Noted, Thank you sir.