espressif/esp-idf #5473: ethernet basic example not compile

capantojar3, 2020-06-20:
Environment
Development Kit: esp devkit v1

Module or chip used: [ESP32-WROOM-32|

IDF version : v4.0.1

Build System: idf.py

Compiler version :
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0

Operating System: [linux ubuntu]

Using an IDE?: |Yes VS Code

Power Supply: NA

Problem Description
I am unable to compile the ethernet basic example. The error I got is:

../main/ethernet_example_main.c: In function 'app_main':
../main/ethernet_example_main.c:85:15: error: 'eth_mac_config_t' {aka 'struct <anonymous>'} has no member named 'smi_mdc_gpio_num'
     mac_config.smi_mdc_gpio_num = CONFIG_EXAMPLE_ETH_MDC_GPIO;
               ^
../main/ethernet_example_main.c:86:15: error: 'eth_mac_config_t' {aka 'struct <anonymous>'} has no member named 'smi_mdio_gpio_num'
     mac_config.smi_mdio_gpio_num = CONFIG_EXAMPLE_ETH_MDIO_GPIO;
               ^
In file included from /home/insepet/Documentos/esp/esp-idf/components/esp_common/include/esp_timer.h:44,
                 from /home/insepet/Documentos/esp/esp-idf/components/freertos/include/freertos/portmacro.h:84,
                 from /home/insepet/Documentos/esp/esp-idf/components/freertos/include/freertos/portable.h:96,
                 from /home/insepet/Documentos/esp/esp-idf/components/freertos/include/freertos/FreeRTOS.h:105,
                 from ../main/ethernet_example_main.c:11:
../main/ethernet_example_main.c:127:49: error: implicit declaration of function 'esp_eth_new_netif_glue'; did you mean 'esp_netif_is_netif_up'? [-Werror=implicit-function-declaration]
     ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));
                                                 ^~~~~~~~~~~~~~~~~~~~~~
/home/insepet/Documentos/esp/esp-idf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
../main/ethernet_example_main.c:127:49: warning: passing argument 2 of 'esp_netif_attach' makes pointer from integer without a cast [-Wint-conversion]
     ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));
                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/insepet/Documentos/esp/esp-idf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
In file included from ../main/ethernet_example_main.c:13:
/home/insepet/Documentos/esp/esp-idf/components/esp_netif/include/esp_netif.h:112:78: note: expected 'esp_netif_iodriver_handle' {aka 'void *'} but argument is of type 'int'
 esp_err_t esp_netif_attach(esp_netif_t *esp_netif, esp_netif_iodriver_handle driver_handle);
                                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
In file included from /home/insepet/Documentos/esp/esp-idf/components/esp_common/include/esp_timer.h:44,
                 from /home/insepet/Documentos/esp/esp-idf/components/freertos/include/freertos/portmacro.h:84,
                 from /home/insepet/Documentos/esp/esp-idf/components/freertos/include/freertos/portable.h:96,
                 from /home/insepet/Documentos/esp/esp-idf/components/freertos/include/freertos/FreeRTOS.h:105,
                 from ../main/ethernet_example_main.c:11:
../main/ethernet_example_main.c:129:21: error: implicit declaration of function 'esp_eth_start'; did you mean 'esp_restart'? [-Werror=implicit-function-declaration]
     ESP_ERROR_CHECK(esp_eth_start(eth_handle));
                     ^~~~~~~~~~~~~
/home/insepet/Documentos/esp/esp-idf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
cc1: some warnings being treated as errors
[805/859] Linking CXX static library esp-idf/soc/libsoc.a
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

how to compile the program correctly?
thanks

suda-morris, 2020-06-21:
Please update your ESP-IDF to latest release/v4.0 (or later), your IDF is missing some important API like "esp_eth_start".

Alvin1Zhang, 2020-08-05:
@capantojar3 hi, would you please help share if any updates for this issue? Thanks.

Alvin1Zhang, 2020-08-14:
Thanks for reporting, will close due to short of feedback, feel free to reopen if the issue still happens or please help share more updates. Thanks.

closed 2020-08-14 by Alvin1Zhang