espressif/esp-idf #5474: Reboot after ret esp_vfs_fat_sdspi_mount

chuledeco, 2020-06-20:
## Environment

- Development Kit:     Custom board with ESP32-pico-D4
- Module or chip used:  ESP32-PICO-D4
- IDF version ESP-IDF v4.2-dev-1415-ga2263571b-dirty 2nd stage bootloader
- Build System:         idf.py
- Operating System:     Windows
- Using an IDE?: Eclipse
- Power Supply:        external 3.3V

## Problem Description

Hello! I loaded the sd_car_example and it worked with the previous version of esp-idf but now with 4.2 after 
`ret = esp_vfs_fat_sdspi_mount(mount_point, &host, &slot_config, &mount_config, &card);`
 the system reboot with no error message.

Could it be a problem with the latest version of esp-idf?

Thanks

## Debug Logs

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7100
load:0x40078000,len:13696
load:0x40080400,len:4000
entry 0x40080688
I (30) boot: ESP-IDF v4.2-dev-1415-ga2263571b-dirty 2nd stage bootloader
I (30) boot: compile time 18:10:35
I (31) boot: chip revision: 1
I (35) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 2MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (95) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0a35c ( 41820) map
I (131) esp_image: segment 1: paddr=0x0001a384 vaddr=0x3ffb0000 size=0x022b8 (  8888) load
I (135) esp_image: segment 2: paddr=0x0001c644 vaddr=0x40080000 size=0x00404 (  1028) load
I (138) esp_image: segment 3: paddr=0x0001ca50 vaddr=0x40080404 size=0x035c8 ( 13768) load
I (152) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x1d144 (119108) map
I (201) esp_image: segment 5: paddr=0x0003d16c vaddr=0x400839cc size=0x07e40 ( 32320) load
I (221) boot: Loaded app from partition at offset 0x10000
I (221) boot: Disabling RNG early entropy source...
I (222) cpu_start: Pro cpu up.
I (225) cpu_start: Application information:
I (230) cpu_start: Project name:     sd_card
I (235) cpu_start: App version:      1
I (240) cpu_start: Compile time:     Jun 20 2020 18:11:32
I (246) cpu_start: ELF file SHA256:  fb3d03082c472c70...
I (252) cpu_start: ESP-IDF:          v4.2-dev-1415-ga2263571b-dirty
I (259) cpu_start: Starting app cpu, entry point is 0x400816ec
I (0) cpu_start: App cpu up.
I (269) heap_init: Initializing. RAM available for dynamic allocation:
I (276) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (282) heap_init: At 3FFB2B58 len 0002D4A8 (181 KiB): DRAM
I (288) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (295) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (301) heap_init: At 4008B80C len 000147F4 (81 KiB): IRAM
[... 208 lines cut ...]
    // Open renamed file for reading
    ESP_LOGI(TAG, "Reading file");
    f = fopen(MOUNT_POINT"/foo.txt", "r");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for reading");
        return;
    }
    char line[64];
    fgets(line, sizeof(line), f);
    fclose(f);
    // strip newline
    char* pos = strchr(line, '\n');
    if (pos) {
        *pos = '\0';
    }
    ESP_LOGI(TAG, "Read from file: '%s'", line);

    // All done, unmount partition and disable SDMMC or SPI peripheral
    esp_vfs_fat_sdcard_unmount(mount_point, card);
    ESP_LOGI(TAG, "Card unmounted");
#ifdef USE_SPI_MODE
    //deinitialize the bus after all devices are removed
    spi_bus_free(host.slot);
#endif
}`

Alvin1Zhang, 2020-06-22:
Thanks for reporting, we will look into.

chuledeco, 2020-06-24:
Hi! Any news? Just to tell that commenting :
`#define USE_SPI_MODE`
and uncommenting:
`slot_config.width = 1;`

it works fine

closed 2020-07-02 by chuledeco