espressif/esp-idf #5476: rmt_set_tx_thr_intr_en can't be used with own interrupt handler

bbulkow, 2020-06-21:
As per issue #4583 , rmt_set_tx_thr_intr_en() causes panic on recent masters. 

I am using release/v4.2, and the behavior is still there. However, it is unclear how to use your own interrupt handler with the RMT devices anymore.

The documentation clearly states that this function may be used with your own interrupt handler. That section of the documentation makes it very clear that when you register an interrupt handler, you then specify which interrupts you want. You MUST NOT call rmt_driver_install, because that installs the system interrupt handler.

The "Use Interrupts" section:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html

Prior code which worked in 4.0 would register an interrupt handler with either esp_intr_alloc(), or with rmt_isr_register, then this function to set the rmt_set_tx_thr_intr_en() will fail. The reason, presumably, is that rmt_set_tx_thr_intr_en takes a channel, and that channel has not been configured. 

At that point, how is it possible in RMT after this checkin to use your own interrupt handler? You need to set the threshold for when the RMT buffer should be calling the interrupt handler using this function, and the answer to #4583 is to always use the system driver --- which specifically states that one must not be using or registering ones own handler.

Perhaps the documentation is wrong and you aren't suposed to call rmt_set_tx_thr_intr_en, but then, how are you supposed to set the interrupt threshold of the hardware?

Please advise on how we are supposed to use our own interrupt handlers instead of the system ones. At this point, only the system driver, not your own interrupt handler, seems to be operational.

To see sample code, please visit:
https://github.com/bbulkow/FastLED-idf . 

The necessary code is in: components/FastLED-idf/latforms/esp/32/clockless_rmt_esp32.h . This is the section where the interrupt handler is. The code works acceptably with the system driver, but will work more efficiently with the hand-built one.

wilhelmw201, 2020-06-22:
well a quick and dirty fix: find and open rmt.c in components/driver and replace 
```
RMT_CHECK(rmt_contex.rmt_driver_channels == 0, "RMT driver installed, can not install generic ISR handler", ESP_FAIL);
```

with 

```
if (rmt_contex.rmt_driver_channels != 0) 
rmt_isr_deregister(rmt_contex.rmt_driver_intr_handle);
```
then call rmt_isr_register after rmt_driver_install to overwrite the default handler but keep the remaining behavior in the rmt_driver_install

bbulkow, 2020-06-22:
Thanks for taking a look into this. Interesting change, because it would mean you'd have to set all your rmt_set_tx_thr_intr_en before registering your interupt handler, and then never change. That's fine for the code I've got. I look forward to your change, I can't see how the function signature for that function is going to work in the long run.... but I've only been looking at that unit for a little bit.

wilhelmw201, 2020-06-23:
rmt_set_tx_thr_intr_en only writes some registers and shouldn't cause a crash. You should be able to call it any time you want. The rmt_write_items should be the direct reason of crashes. It accesses some mutexes, arrays, etc. that are allocated in the rmt_driver_install, so you have to call rmt_driver_install first or you are accessing null pointers.

bbulkow, 2020-06-24:
The current master calls RMT_CHECK if the RMT driver is not installed, and RMT_CHECK causes a crash (aka reboot) is there is no rmt driver installed. The code does not crash if I apply the change as you suggested.

Thus your prior response seemed correct, are you going to make that change to master and 4.2 release?

bbulkow, 2020-06-24:
My code only calls rmt_write_items if using the built-in interrupt handler, so that isn't the issue.

bbulkow, 2020-06-26:
What's super interesting here is my code works great on 4.0, using my own interrupt driver.

Can you suggest what might have changed between 4.0 and 4.2 in the interrupt handling system that would likely result in crashes?

bbulkow, 2020-06-27:
Your proposed change of referencing "rmt_contex", there is no such variable defined anywhere.

Similar code is in one place in rmt.c, and it is like this:
```
	// In Github 5476, I was told to make this change
    _lock_acquire_recursive(&rmt_driver_isr_lock);
    s_rmt_driver_channels &= ~BIT(channel);
    if (s_rmt_driver_channels == 0) {
        // all channels have driver disabled
        rmt_isr_deregister(s_rmt_driver_intr_handle);
        s_rmt_driver_intr_handle = NULL;
    }
    _lock_release_recursive(&rmt_driver_isr_lock);

```

bbulkow, 2020-06-30:
I have determined my problem. It appears to focus around the initialization of the rmt_config_t structure.

The change ESP-IDF has is there is now an initialization #define in 4.1++. I use the initialization #define, my code works, if I simply set the obvious fields, it doesn't. I have looked at that code and there is a new field, and that field is covered by the sdkconfig variable, which would change the size of the structure, so it's important to get the build to match correctly. I have also placed a memset(0) for the config structure in 4.0. Since this is one config structure one time, the performance loss isn't important.

I propose you document the new and very handy RMT_DEFAULT_CONFIG_TX  My code that works in all 4.x releases looks like this:

```
// -- RMT configuration for transmission --- different in different ESP versions
#if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4,1,0)
            rmt_config_t rmt_tx = RMT_DEFAULT_CONFIG_TX(mPin, fastled_get_rmt_channel(i) );
#else
            rmt_config_t rmt_tx;
            memset(&rmt_tx, 0, sizeof(rmt_tx));
            rmt_tx.channel = fastled_get_rmt_channel(i);
            rmt_tx.gpio_num = mPin;
            rmt_tx.mem_block_num = 1;
            rmt_tx.tx_config.carrier_level = RMT_CARRIER_LEVEL_LOW;
#endif // version before 4.1

            rmt_tx.clk_div = DIVIDER;
            // don't wish to have a carrier applied. Therefore carrier_en is false and the extra parameters don't matter.
            rmt_tx.tx_config.loop_en = false;
            rmt_tx.tx_config.carrier_en = false;

            rmt_tx.tx_config.idle_level = RMT_IDLE_LEVEL_LOW;
            rmt_tx.tx_config.idle_output_en = true;
 
```
Since this is no longer an issue with ESP-IDF --- other than the documentation --- I propose the issue should be closed.

PS. My driver code still doesn't work, but it doesn't work in all releases, and it's probably my bug.

wilhelmw201, 2020-07-07:
The config struct did change but I did not expect it to have such a large impact on your program :( I am very sorry. I suggest using the dot-before-variables-and-braces-around-everything struct initialization aka designated initializers like the macros in RMT.h so that any new fields are all set to 0 instead of surprising values (also it looks cool).

The rmt_isr_register will cause the default ISR to become bypassed, and as a result the all functions in the driver may not function as expected. It would probably be better to directly call on HAL functions if you use your own ISR, only reusing those functions that do not cause errors.

bbulkow, 2020-07-11:
The issue here is that the struct initialization you are saying is "suggested" and you think "looks cool" is now the only way to write stable code. From my experience, this is a "landmine" in the code at this point, where 4.0 code appears to work silently in 4.1++ but your internal code uses uninitialized variables. If you place such a landmine, documenting it makes your platform pleasant, in that it becomes required to check the version as I've shown, and use the right inits for the version. 
Regarding style points, it is generally poor form to do the following:
* Depend on a _platform_ version instead of a _unit_ version. If you add a field to a struct, and that changes the API ( even as simple as it must be 0'ed ), one should be able to determine the correct code through a _unit_ version.
* requiring zeroing an allocated structure then setting the bits you need. In embedded code, that's inefficient, and also in high performance server code. It is better at some point to have an init function that takes _ALL_ variables ( perhaps with overloads for the common ones ) so avoid the anti-pattern of bzero the entire size.

In any case, document, document, document, please.

closed 2020-07-12 by bbulkow