espressif/esp-idf #5479: esp-tls: mbedtls_ssl_handshake returned -0x10

shenyong-iot01, 2020-06-22:
您好，
     我这边设备在ssl连接时遇到以下问题：
esp_get_minimum_free_heap_size():224
heap_caps_get_largest_free_block():39396
esp_get_free_heap_size:49264

【2020-06-22 16:28:24:375】E (7795) esp-tls: mbedtls_ssl_handshake returned -0x10
E (7795) esp-tls: Failed to open new connection
E (7795) TRANS_SSL: Failed to open a new connection
E (7795) MQTT_CLIENT: Error transport connect
[APP MQTT] MQTT_EVENT_ERROR
[APP MQTT] MQTT_EVENT_DISCONNECTED

请协助分析下，谢谢！

shenyong-iot01, 2020-06-22:
补充：握手之前剩余11K左右，握手前消耗了38000左右？
ssl连接需要这么多内存
【2020-06-22 18:43:54:331】E (99087) esp-tls: esp_get_free_heap_size: 11344
[image]

shenyong-iot01, 2020-06-23:
【2020-06-23 09:29:41:601】E (47827) esp-tls: esp_get_free_heap_size: 12380
E (47827) esp-tls: esp_get_minimum_free_heap_size: 464
E (47827) esp-tls: heap_caps_get_largest_free_block: 10252
I (47827) mbedtls: ssl_tls.c:8084 => handshake
[image]

esp_get_minimum_free_heap_size 变小后连接ssl失败？esp_get_minimum_free_heap_size这个变化规律是？

AdityaHPatwardhan, 2020-07-27:
@shenyong-iot01 can you please send the full debug_log also your sdkconfig file, Thank you

Alvin1Zhang, 2020-08-14:
Thanks for reporting, would you please help share the full debug_log and also your sdkconfig file? Thanks.

Alvin1Zhang, 2020-10-15:
Thanks for reporting, will close due to short of updates, feel free to reopen with more details, thanks.

closed 2020-10-15 by Alvin1Zhang