espressif/esp-idf #5482: ESP-IDF v4.0.1 crashes during lwip name resolution in pthread_local_storage library

maikeriva, 2020-06-22:
As suggested by @dobairoland I am opening a new issue for this particular problem.
In ESP-IDF v4.0.1 I experience crashes spawning from the pthread_local_storage library when attempting to perform an HTTPS connection with esp-http-client.

I believe this could be linked with setting  `component config`->`FreeRTOS`->`number of thread local storage pointers` equal to 10 in menuconfig, as I recently began to use local storage pointers in my application and I didn't experience this issue before.

The issue may be related to https://github.com/espressif/esp-idf/issues/5267 , in which a very similar thread local storage occurs.

Not being able to perform HTTPS connections is pretty critical for any kind of IoT application, and IDF v4.0.1 is labelled as stable. Is there a quick way to fix this?

Here's the traceback:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401d2cf8  PS      : 0x00060530  A0      : 0x800d3a20  A1      : 0x3ffe3260  
0x401d2cf8: find_value at /home/micheleriva/esp-idf/components/pthread/pthread_local_storage.c:192

A2      : 0xc2004136  A3      : 0x00000003  A4      : 0x00060520  A5      : 0x00000001  
A6      : 0x00060520  A7      : 0x3ffbb884  A8      : 0x8009a8c6  A9      : 0x3ffe3240  
A10     : 0x3ffbb4fc  A11     : 0x00000000  A12     : 0x3ffb63ec  A13     : 0x00000001  
A14     : 0x00060523  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xc2004136  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: acb09f97e4a349dc

Backtrace: 0x401d2cf5:0x3ffe3260 0x400d3a1d:0x3ffe3280 0x401667b3:0x3ffe32a0 0x40167771:0x3ffe32c0 0x40152d95:0x3ffe3300 0x4014033e:0x3ffe3350 0x4014038c:0x3ffe33a0 0x40140b6a:0x3ffe33f0 0x40140e7f:0x3ffe3420 0x4013f07f:0x3ffe3450 0x401d3bfd:0x3ffe3470 0x4013dc3d:0x3ffe3490 0x4013de85:0x3ffe34c0 0x400d8298:0x3ffe34e0 0x400dcedb:0x3ffe3510 0x400dcfb2:0x3ffe3990 0x40099505:0x3ffe39b0
0x401d2cf5: find_value at /home/micheleriva/esp-idf/components/pthread/pthread_local_storage.c:191

0x400d3a1d: pthread_getspecific at /home/micheleriva/esp-idf/components/pthread/pthread_local_storage.c:206

0x401667b3: sys_thread_sem_get at /home/micheleriva/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:455

0x40167771: netconn_gethostbyname_addrtype at /home/micheleriva/esp-idf/components/lwip/lwip/src/api/api_lib.c:1329 (discriminator 2)

0x40152d95: lwip_getaddrinfo at /home/micheleriva/esp-idf/components/lwip/lwip/src/api/netdb.c:342

0x4014033e: resolve_host_name at /home/micheleriva/esp-idf/components/esp-tls/esp_tls.c:62 (discriminator 6)

0x4014038c: esp_tcp_connect at /home/micheleriva/esp-idf/components/esp-tls/esp_tls.c:101

0x40140b6a: esp_tls_low_level_conn at /home/micheleriva/esp-idf/components/esp-tls/esp_tls.c:570

0x40140e7f: esp_tls_conn_new_sync at /home/micheleriva/esp-idf/components/esp-tls/esp_tls.c:697

0x4013f07f: ssl_connect at /home/micheleriva/esp-idf/components/tcp_transport/transport_ssl.c:74

0x401d3bfd: esp_transport_connect at /home/micheleriva/esp-idf/components/tcp_transport/transport.c:165

0x4013dc3d: esp_http_client_connect at /home/micheleriva/esp-idf/components/esp_http_client/esp_http_client.c:1016

0x4013de85: esp_http_client_open at /home/micheleriva/esp-idf/components/esp_http_client/esp_http_client.c:1172
```

maikeriva, 2020-06-22:
After some experiments it seems that the issue arises when the task which calls esp_http_client_open() has thread_local_storage pointers set. Not setting them prevents the crash.

Alvin1Zhang, 2020-06-23:
Thanks for reporting,  we will look into.

mahavirj, 2020-06-23:
@maikeriva Can you please share your application code, specifically changes around thread local storage in task that is attempting TLS connection? Just to note that, index 0 in thread local storage is reserved and it should not be used by application, please refer: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/thread-local-storage.html

Alvin1Zhang, 2020-07-01:
@maikeriva Thanks for reporting, would you please help share if any updates for this issue? Thanks.

maikeriva, 2020-07-06:
Thanks for your help!
Indeed it turned out that I was using the first slot of local storage pointers in some tasks. I switched to a task-argument-based approach to pass parameters, letting the local storage pointers alone, and the issue got resolved. I suspect that using local storage pointers starting from the second slot would have solved it as well, but I did not test such approach.

mahavirj, 2020-07-06:
Thanks for confirmation, will close the issue then.

closed 2020-07-06 by mahavirj