espressif/esp-idf #5483: BLE Mesh when LPN re-creates a friendship it should register for all current subscriptions

mishafarms, 2020-06-22:
## Environment

- Development Kit:      ESP32-Wrover-Kit, But any will have the issue
- Kit version v3
- Module or chip used:  
- IDF version v4.1-beta2-141-g84b51781c
- Build System:        eclipse build from eclipse plugin
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: YES Eclipse with eclipse plugin in a docker
- Power Supply:        USB

## Problem Description
When re-creating a friendship, the LPN should still receive all group subscriptions that have been provisioned.

I am playing with BLE Mesh and LPN. If my node already has a friendship and I provision a group subscription, when I send to the group the node gets the message. But if the node loses the friendship either because the node or the friend is reset or because I lose network connectivity, When the new friendship is established, I no longer get the group messages.

### Expected Behavior
Any subscriptions that the LPN has should be registered when it created a friendship.

### Actual Behavior
Only group subscriptions that are created after we have the current friendship created are registered with the friend.
 
### Steps to reproduce
Create a friendship between the LPN and a friend, Add a group subscription. Make sure that the subscription works. Reboot the friend and let a new friendship be created either with the original friend or a new friend. Try to send to the subscribed group. The LPN will not receive the message.

This is because the group subscriptions are not re-registered with the friend. I believe that the LPN should do this at a low layer. But I currently solved it by calling this code in my main.c in the provisioning callback. I think this is a bug in the LPN code. Am I correct? or should this be done in the prov_cb?

```
static void register_sub_with_friend(void)
 {
	// need a pointer to the composition data
	u16_t myGroups[CONFIG_BLE_MESH_LPN_GROUPS];
	int groupIdx = 0;

	const esp_ble_mesh_comp_t *myComp = esp_ble_mesh_get_composition_data();

	if (myComp == NULL)
	{
		ESP_LOGE(TAG, "No composite data found");
		// we are done now
		return;
	}

	// I got the composite. now walk the element list.

	for (int i = 0; i < myComp->element_count; i++)
	{
		// access the element
		const esp_ble_mesh_elem_t *myElement = &myComp->elements[i];

		// walk the SIG models

		for (int j = 0; j < myElement->sig_model_count; j++)
		{
			// now I need to look to see which groups are registered for this model

			const esp_ble_mesh_model_t *myModel = &myElement->sig_models[j];

			if (myModel == NULL)
			{
				// this shouldn't happen
				ESP_LOGE(TAG, "Model is NULL");
				continue;
			}

			for (int k = 0; k < CONFIG_BLE_MESH_MODEL_GROUP_COUNT; k++)
			{
				if (myModel->groups[k])
				{
					if (groupIdx < CONFIG_BLE_MESH_LPN_GROUPS)
					{
						myGroups[groupIdx++] = myModel->groups[k];
						ESP_LOGI(TAG, "Group %04x registered", myModel->groups[k]);
					}
				}
			}
		}

		// walk the SIG models

		for (int j = 0; j < myElement->vnd_model_count; j++)
		{
			// now I need to look to see which groups are registered for this model

			const esp_ble_mesh_model_t *myModel = &myElement->vnd_models[j];

			if (myModel == NULL)
			{
				// this shouldn't happen
				ESP_LOGE(TAG, "Model is NULL");
				continue;
			}

			for (int k = 0; k < CONFIG_BLE_MESH_MODEL_GROUP_COUNT; k++)
			{
				if (myModel->groups[k])
				{
					if (groupIdx < CONFIG_BLE_MESH_LPN_GROUPS)
					{
						myGroups[groupIdx++] = myModel->groups[k];
						ESP_LOGI(TAG, "Group %04x registered", myModel->groups[k]);
					}
				}
			}
		}
	}

	// delete all the groups, then add them again, this should be no problem and make sure that we subscribe at the
	// friend

	bt_mesh_lpn_group_del(myGroups, groupIdx);

	// now do it one at a time

	for (int i = 0; i < groupIdx; i++)
	{
		bt_mesh_lpn_group_add(myGroups[i]);
	}
 }
```

Alvin1Zhang, 2020-06-23:
@mishafarms Thanks for reporting, we will look into.

Campou, 2020-06-23:
Hi @mishafarms 

Thanks for this. 

Currently when a friendship is terminated, the subscription list on the friend node will be cleared. And i tried to find the corresponding description in the mesh spec (i.e. keep the subscription list when friendship is terminated), currently didn't find anything for this.

So in this condition, two solutions can be used.
1. When the friendship is re-established, you can send the Configuration Model Subscription Add to the LPN. When the LPN received the configuration message, it will send Friend Subscription List Add to the friend node;
2. Currently on the latest master branch, I have added some funcitons for model subscription locally.  The commit is https://github.com/espressif/esp-idf/commit/9aae4d3ba3e7c7f35841beeea8e6f0ce10929efd. If this functionality is not on the branch you are using, please cherry-pick this commit and have a try. Once the local model subscription function is invoked, the group address will be subscribed by a model. And if LPN is enabled, it will send Friend Subscription List Add to the friend node.

Thanks.

mishafarms, 2020-06-23:
@Campou,

You should not keep the subscription list on the friend node, You might create a new friendship with a different friend node and that info would not be on that friend node. 

I will try the commit you recommend.

Thanks

Campou, 2020-06-23:
Hi @mishafarms 

Yes, you're right. So the key point here is that when LPN established a new friendship, it needs to add the subscription list to the friend node.

Thanks.

mishafarms, 2020-06-23:
@Campou 

I put some code in the first message that does exactly that. I was thinking it belongs in the lower layer. I was calling from the prov_callback, It worked great.

Thanks.

Shane555, 2022-08-24:
Any idea if this is fixed?

Shane555, 2022-08-25:
> Hi @mishafarms
> 
> Thanks for this.
> 
> Currently when a friendship is terminated, the subscription list on the friend node will be cleared. And i tried to find the corresponding description in the mesh spec (i.e. keep the subscription list when friendship is terminated), currently didn't find anything for this.
> 
> So in this condition, two solutions can be used.
> 
> 1. When the friendship is re-established, you can send the Configuration Model Subscription Add to the LPN. When the LPN received the configuration message, it will send Friend Subscription List Add to the friend node;
> 2. Currently on the latest master branch, I have added some funcitons for model subscription locally.  The commit is [9aae4d3](https://github.com/espressif/esp-idf/commit/9aae4d3ba3e7c7f35841beeea8e6f0ce10929efd). If this functionality is not on the branch you are using, please cherry-pick this commit and have a try. Once the local model subscription function is invoked, the group address will be subscribed by a model. And if LPN is enabled, it will send Friend Subscription List Add to the friend node.
> 
> Thanks.

Currently, i am also calling it from prov_callback. But i don't think this makes any sense even if the mesh specs did not specify what to do in a scenario where the LPN and friend node breaks connection for whatever reason. Perhaps there could be a feature in idf menuconfig to allow user to check if they want the LPN to continue receiving message after breaking friendship. This would require modifying the lower layer like lpn.c or friend.c for example. Most of the time, we want to ensure LPN continue receiving message from a friend after breaking friendship as mesh nodes can easily lose power/connection for unforeseen reasons in an actual network and we would want the connection to resume to the same state

mishafarms, 2022-08-25:
I have not played with this in quite a while.

When the LPN and the friend lose the connection, The friend should clear all info about the LPN. The LPN can make a friendship
with any other friend. Once the new friendship is created, the LPN should inform the new friend that it is subscribed to any groups that it has been provisioned to subscribe to. 

This should be at some low layer (when the friendship is established). 

What I did was to cause the group subscriptions to be sent to the new friend when the new friendship was established, buy
calling a function at the top layer.

I was hoping that the MESH spec had something about this. I have not looked at the spec in a while either. But since the
LPN is not re-provisioned every time we make a new friendship, the LPN needs to inform that friend of all the messages
that the friend needs to forward to the LPN.

The LPN surely want to receive all the messages that it was provisioned to subscribe to.

Shane555, 2022-08-26:
That is odd. I couldn't get the group subscription to work on the friend node. It doesn't send the message meant for the LPN after friendship is established until i manually sent the group subscription again in the application layer after resetting the node. Anyways, i wanted to use the esp LPN for low powered uses. Currently the current consumption is still too high with light sleep and deep sleep is not really working well with LPN. What the deep sleep wake up does is to reboot the node and although it stores bluetooth connection configuration(able to connect to the other mesh nodes when it wakes up), it does not store the friendship connection during the deep sleep which also means its not a viable method for deep sleep.

Paramesh19, 2023-03-13:
Can anyone share the example code for LPN and friend node 
I am trying so hard didn't fine any reference codes for esp32 ble mesh lowpower operation examples

Paramesh19, 2023-04-01:
@Shane555 could you help me in establishing lpn and friend features in esp ble mesh,i am struggling to establish the connection,
i am getting warning like No acceptable friend offers received at lpn node,
kindly help me with this
thank you

Shane555, 2023-04-02:
> @Shane555 could you help me in establishing lpn and friend features in esp ble mesh,i am struggling to establish the connection, i am getting warning like No acceptable friend offers received at lpn node, kindly help me with this thank you

That's very vague and could be because of many reasons. Like replay protection list for eg. What's the log like for the friend node?