espressif/esp-idf #5484: LPN should be allowed to go to DEEP Sleep mode and when woken up just do the Friendship Poll

mishafarms, 2020-06-22:
In order to reduce the power requirements when an LPN turns off the BT radio, it should have the option to go into DEEP SLEEP mode with a timeout just less then the poll timeout and when it awakes, it should be able to just do a friendship poll.

I would like for the needed BLE mesh info to be stored in the RTC memory so that it is not lost and if we wake up from a DEEP SLEEP, the code would not have to do the total mesh init, It could call a different function that would not re-init everything and re-initiate a friendship, it should be able to just do a friendship poll and if the friend answers with a message then process it and if the friend responds with no_message (I believe this is where we would again disable the BT radio) enter DEEP SLEEP until just before the poll timeout.

This would allow a LPN to run at the lowest power consumption. I would expect this to be a config option for LPN.

Alvin1Zhang, 2020-06-23:
@mishafarms Thanks for raising this feature request, we will check.

Campou, 2020-06-23:
Hi @mishafarms 

Previously I have tried with light sleep on the LPN, it works fine. For using Deep Sleep mode and RTC memory, this will need some evaluation.
I will look into this, and get back to you soon.

Thanks.

mishafarms, 2020-06-23:
@Campou 
Using Light Sleep mode will use .8ma (according to the documentation I have), If I use a CR2032 battery (240 mah) I will get at best 12.5 days. If we can get Deep Sleep working it should use .01ma I would get at best 1000 days.

Now I know those times are without doing anything else. But assuming it takes 250ms to perform a poll with no message being transferred and 160ma to do that. I figure it uses about 4mah (160ma / (3600 / .250) * 360 polls per hour) that should give me about 60 days at a 10 second poll timeout and I can probably go to a 30 or 60 second poll timeout. This would now be more useful in a product.

Having said that. Where in the code did you put the light sleep? As I would like to try that anyway just to understand things a little better.

Thanks

Campou, 2020-06-23:
Hi @mishafarms 

I just added part of the implementations in the [light sleep example](https://github.com/espressif/esp-idf/tree/master/examples/system/light_sleep) to the mesh example, and enable bluetooth modem sleep in the menuconfig. No other big changes.

Thanks.

mishafarms, 2020-06-23:
@Campou 

Shouldn't you only go into light sleep when the friend responds to the poll with a the msg->md cleared?

Thanks

InfiniteYuan, 2021-09-26:
You can implement deep sleep in the application layer and wake up before each poll (you need to consider the wake-up time of deep sleep).

closed 2025-03-14 by Alvin1Zhang