espressif/esp-idf #5485: lwIP freeing memory it's not supposed to

KaeLL, 2020-06-22:
- Development Kit:      none
- Module or chip used:  ESP32D0WDQ6 (revision 1)
- IDF version: v4.2-dev-1905-g625bd5eb1
- Build System:         CMake
- Compiler version: esp-2020r2-8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 5V

I'm porting the W5100 Ethernet chip to work with esp-idf, and for some reason, lwIP is not copying the buffer delivered to it by the [stack_input](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/include/esp_eth_com.h#L175) function, but rather copying the address which the buffer is located at instead. This is a problem because if I handle the [receive](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/include/esp_eth_mac.h#L123)ing of data correctly, the app breaks because of double `free`'s. And there's no doubt the buffer should be freed by layer 2 client code here. I even tried enabling `LWIP_L2_TO_L3_COPY`, but apparently that works only for the WiFi driver.
There are 3 comments in the log block below on the middle to right side of the block, starting with `<-`
```
I (192) cpu_start: Application information:
I (192) cpu_start: Project name:     w5100_port
I (192) cpu_start: App version:      295c8d8-dirty
I (193) cpu_start: Compile time:     Jun 22 2020 17:34:17
I (194) cpu_start: ELF file SHA256:  d22d5db4f5fad967...
I (194) cpu_start: ESP-IDF:          v4.2-dev-1905-g625bd5eb1-dirty
I (195) cpu_start: Starting app cpu, entry point is 0x40081830
0x40081830: call_start_cpu1 at /esp-idf/components/esp32/cpu_start.c:286

I (0) cpu_start: App cpu up.
I (197) heap_init: Initializing. RAM available for dynamic allocation:
I (198) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (198) heap_init: At 3FFB3E10 len 0002C1F0 (176 KiB): DRAM
I (199) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (200) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (201) heap_init: At 4008C120 len 00013EE0 (79 KiB): IRAM
I (201) cpu_start: Pro cpu start user code
I (210) spi_flash: detected chip: generic
I (211) spi_flash: flash io: qio
I (211) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
netif: netmask of interface  set to 255.0.0.0
netif: GW address of interface  set to 127.0.0.1
netif_set_ipaddr: netif address being changed
netif: added interface lo IP addr 127.0.0.1 netmask 255.0.0.0 gw 127.0.0.1
tcpip_task_hdlxxx : 3ffb5c50, prio:18,stack:3584
I (0) system_api: Base MAC address is not set
I (0) system_api: read default base MAC address from EFUSE
I (0) esp_eth.netif.glue: 3c:71:bf:42:e6:eb
I (0) esp_eth.netif.glue: ethernet attached to netif
tcpip_thread: API message 0x3ffc8560
netif: added interface en IP addr 0.0.0.0 netmask 0.0.0.0 gw 0.0.0.0
I (219) eth_example: Ethernet Started
etharp_timer
dhcp_coarse_tmr()
pbuf_alloc(length=16)
pbuf_alloc(length=16) == 0x3ffb789c
pbuf_add_header: old 0x3ffb78ec new 0x3ffb78c4 (40)
pbuf_alloc(length=56)
pbuf_alloc(length=56) == 0x3ffb7900
pbuf_copy(0x3ffb7900, 0x3ffb789c)
pbuf_copy: end of chain reached.
pbuf_free(0x3ffb789c)
pbuf_free: deallocating 0x3ffb789c
tcpip_thread: CALLBACK 0x3ffb4c44
pbuf_free(0x3ffb7900)
pbuf_free: deallocating 0x3ffb7900
etharp_timer
dhcp_coarse_tmr()
tcpip_thread: API message 0x3ffc8510
netif: setting default interface en
tcpip_thread: API message 0x3ffc8510
dhcp_start(netif=0x3ffc88a8) en1
dhcp_start(): restarting DHCP configuration
dhcp_start(): starting DHCP configuration
dhcp_discover()
pbuf_alloc(length=308)
pbuf_alloc(length=308) == 0x3ffb78f8
transaction id xid(52bc3e71)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
pbuf_add_header: old 0x3ffb795c new 0x3ffb7954 (8)
pbuf_add_header: old 0x3ffb7954 new 0x3ffb7940 (20)
ip4_output_if: en1
IP header:
+-------------------------------+
| 4 | 5 |  0x00 |       336     | (v, hl, tos, len)
+-------------------------------+
|        0      |000|       0   | (id, flags, offset)
+-------------------------------+
|  255  |   17  |    0xba9d     | (ttl, proto, chksum)
+-------------------------------+
|    0  |    0  |    0  |    0  | (src)
+-------------------------------+
|  255  |  255  |  255  |  255  | (dest)
+-------------------------------+
ip4_output_if: call netif->output()
pbuf_add_header: old 0x3ffb7940 new 0x3ffb7932 (14)
ethernet_output: sending packet 0x3ffb78f8
dhcp_discover: deleting()ing
pbuf_free(0x3ffb78f8)
pbuf_free: deallocating 0x3ffb78f8
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 500 msecs
I (2299) w5100_mac: ADDRESS : 0x3ffb78f8       <- This is the buffer I allocated and later will attempt to free
pbuf_alloc(length=320)
pbuf_alloc(length=320) == 0x3ffb98fc
tcpip_thread: PACKET 0x3ffb9918
ethernet_input: dest:3c:71:bf:42:e6:eb, src:e8:20:e2:c5:e5:04, type:800
pbuf_remove_header: old 0x3ffb78f8 new 0x3ffb7906 (14)
ip_input: iphdr->dest 0xcd00a8c0 netif->ip_addr 0x0 (0x0, 0x0, 0xcd00a8c0)
ip_input: iphdr->dest 0xcd00a8c0 netif->ip_addr 0x100007f (0xc0, 0x7f, 0xcd00a800)
ip4_input: UDP packet to DHCP client port 68
ip4_input: DHCP packet accepted.
ip4_input: 
IP header:
+-------------------------------+
| 4 | 5 |  0x00 |       306     | (v, hl, tos, len)
+-------------------------------+
|        0      |000|       0   | (id, flags, offset)
+-------------------------------+
|   64  |   17  |    0xf79c     | (ttl, proto, chksum)
+-------------------------------+
|  192  |  168  |    0  |    1  | (src)
+-------------------------------+
|  192  |  168  |    0  |  205  | (dest)
+-------------------------------+
ip4_input: p->len 306 p->tot_len 306
pbuf_remove_header: old 0x3ffb7906 new 0x3ffb791a (20)
pbuf_remove_header: old 0x3ffb791a new 0x3ffb7922 (8)
dhcp_recv(pbuf = 0x3ffb98fc) from DHCP server 192.168.0.1 port 67
pbuf->len = 278
pbuf->tot_len = 278
searching DHCP_OPTION_MESSAGE_TYPE
DHCP_OFFER received in DHCP_STATE_SELECTING state
dhcp_handle_offer(netif=0x3ffc88a8) en1
dhcp_handle_offer(): server 0x0100a8c0
dhcp_handle_offer(): offer for 0xcd00a8c0
dhcp_select(netif=0x3ffc88a8) en1
pbuf_alloc(length=308)
pbuf_alloc(length=308) == 0x3ffb78f8
transaction id xid(52bc3e71)
pbuf_add_header: old 0x3ffb795c new 0x3ffb7954 (8)
pbuf_add_header: old 0x3ffb7954 new 0x3ffb7940 (20)
ip4_output_if: en1
IP header:
+-------------------------------+
| 4 | 5 |  0x00 |       336     | (v, hl, tos, len)
+-------------------------------+
|        1      |000|       0   | (id, flags, offset)
+-------------------------------+
|  255  |   17  |    0xba9c     | (ttl, proto, chksum)
+-------------------------------+
|    0  |    0  |    0  |    0  | (src)
+-------------------------------+
|  255  |  255  |  255  |  255  | (dest)
+-------------------------------+
ip4_output_if: call netif->output()
pbuf_add_header: old 0x3ffb7940 new 0x3ffb7932 (14)
ethernet_output: sending packet 0x3ffb78f8
pbuf_free(0x3ffb78f8)                                  <-------- Caught with the pants down :P
pbuf_free: deallocating 0x3ffb78f8
dhcp_select: REQUESTING
dhcp_select(): set request timeout 2000 msecs
pbuf_free(0x3ffb98fc)
pbuf_free: deallocating 0x3ffb98fc
CORRUPT HEAP: multi_heap.c:540 detected at 0x3ffb78f4    <------- Double free

abort() was called at PC 0x40088e59 on core 1
0x40088e59: multi_heap_assert at /esp-idf/components/heap/multi_heap_platform.h:67
 (inlined by) multi_heap_free_impl at /esp-idf/components/heap/multi_heap.c:540

Backtrace:0x40084ebb:0x3ffb5930 0x40085559:0x3ffb5950 0x40089396:0x3ffb5970 0x40088e59:0x3ffb59e0 0x4008145e:0x3ffb5a00 0x400893c1:0x3ffb5a20 0x400f1ccd:0x3ffb5a40 0x400e0b7b:0x3ffb5a60 0x400e994a:0x3ffb5a80 0x400e7aa9:0x3ffb5ab0 0x400eba92:0x3ffb5af0 0x400f1812:0x3ffb5b20 0x400de3b9:0x3ffb5b90 0x400de451:0x3ffb5bb0
0x40084ebb: panic_abort at /esp-idf/components/esp_system/panic.c:330

0x40085559: esp_system_abort at /esp-idf/components/esp_system/system_api.c:100

0x40089396: abort at /esp-idf/components/newlib/abort.c:46

0x40088e59: multi_heap_assert at /esp-idf/components/heap/multi_heap_platform.h:67
 (inlined by) multi_heap_free_impl at /esp-idf/components/heap/multi_heap.c:540

0x4008145e: heap_caps_free at /esp-idf/components/heap/heap_caps.c:305

0x400893c1: free at /esp-idf/components/newlib/heap.c:47

0x400f1ccd: ethernet_free_rx_buf_l2 at /esp-idf/components/lwip/port/esp32/netif/ethernetif.c:68

0x400e0b7b: pbuf_free at /esp-idf/components/lwip/lwip/src/core/pbuf.c:783

0x400e994a: dhcp_recv at /esp-idf/components/lwip/lwip/src/core/ipv4/dhcp.c:1997

0x400e7aa9: udp_input at /esp-idf/components/lwip/lwip/src/core/udp.c:404

0x400eba92: ip4_input at /esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:778

0x400f1812: ethernet_input at /esp-idf/components/lwip/lwip/src/netif/ethernet.c:186

0x400de3b9: tcpip_thread_handle_msg at /esp-idf/components/lwip/lwip/src/api/tcpip.c:180 (discriminator 1)

0x400de451: tcpip_thread at /esp-idf/components/lwip/lwip/src/api/tcpip.c:154

ELF file SHA256: d22d5db4f5fad967
```

Alvin1Zhang, 2020-06-23:
@KaeLL Thanks for reporting, we will look into.

KaeLL, 2020-06-23:
I have a PoC ready: https://github.com/KaeLL/esp32_w5100

Running it a few minutes ago gave me this output and error:
<details>
    <summary>Click to expand idf monitor log</summary>

```
9) cpu_start: Pro cpu up.
I (230) cpu_start: Application information:
I (230) cpu_start: Project name:     w5100_port
I (230) cpu_start: App version:      0cda5ae
I (230) cpu_start: Compile time:     Jun 23 2020 17:23:12
I (231) cpu_start: ELF file SHA256:  780798e4d379d9f0...
I (232) cpu_start: ESP-IDF:          v4.2-dev-1905-g625bd5eb1
I (233) cpu_start: Starting app cpu, entry point is 0x4008185c
0x4008185c: call_start_cpu1 at /esp-idf/components/esp32/cpu_start.c:286

I (0) cpu_start: App cpu up.
I (234) heap_init: Initializing. RAM available for dynamic allocation:
I (235) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (236) heap_init: At 3FFB3F38 len 0002C0C8 (176 KiB): DRAM
I (236) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (237) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (238) heap_init: At 4008C23C len 00013DC4 (79 KiB): IRAM
I (239) cpu_start: Pro cpu start user code
I (249) spi_flash: detected chip: generic
I (249) spi_flash: flash io: dio
I (250) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (0) system_api: Base MAC address is not set
I (0) system_api: read default base MAC address from EFUSE
I (0) esp_eth.netif.glue: 3c:71:bf:42:e6:eb
I (0) esp_eth.netif.glue: ethernet attached to netif
I (258) main: Ethernet Started
I (3268) main: Ethernet Got IP Address
I (3268) main: ~~~~~~~~~~~
I (3268) main: ETHIP:192.168.0.205
I (3268) main: ETHMASK:255.255.255.0
I (3268) main: ETHGW:192.168.0.1
I (3268) main: ~~~~~~~~~~~
I (3278) esp_netif_handlers: eth ip: 192.168.0.205, mask: 255.255.255.0, gw: 192.168.0.1
I (3278) MQTTS_EXAMPLE: [APP] Startup..
I (3278) MQTTS_EXAMPLE: [APP] Free memory: 207180 bytes
I (3278) MQTTS_EXAMPLE: [APP] IDF version: v4.2-dev-1905-g625bd5eb1
I (3278) MQTTS_EXAMPLE: [APP] Free memory: 206764 bytes
I (3278) MQTTS_EXAMPLE: Other event id:7
I (3888) HTTP_CLIENT: HTTP GET Status = 200, content_length = 290
I (3888) HTTP_CLIENT: 7b 0a 20 20 22 61 72 67 73 22 3a 20 7b 0a 20 20 
I (3888) HTTP_CLIENT: 20 20 22 65 73 70 22 3a 20 22 22 0a 20 20 7d 2c 
I (3888) HTTP_CLIENT: 20 0a 20 20 22 68 65 61 64 65 72 73 22 3a 20 7b 
I (3888) HTTP_CLIENT: 0a 20 20 20 20 22 43 6f 6e 74 65 6e 74 2d 4c 65 
I (3888) HTTP_CLIENT: 6e 67 74 68 22 3a 20 22 30 22 2c 20 0a 20 20 20 
I (3888) HTTP_CLIENT: 20 22 48 6f 73 74 22 3a 20 22 68 74 74 70 62 69 
I (3888) HTTP_CLIENT: 6e 2e 6f 72 67 22 2c 20 0a 20 20 20 20 22 55 73 
I (3888) HTTP_CLIENT: 65 72 2d 41 67 65 6e 74 22 3a 20 22 45 53 50 33 
I (3888) HTTP_CLIENT: 32 20 48 54 54 50 20 43 6c 69 65 6e 74 2f 31 2e 
I (3898) HTTP_CLIENT: 30 22 2c 20 0a 20 20 20 20 22 58 2d 41 6d 7a 6e 
I (3898) HTTP_CLIENT: 2d 54 72 61 63 65 2d 49 64 22 3a 20 22 52 6f 6f 
I (3898) HTTP_CLIENT: 74 3d 31 2d 35 65 66 32 36 35 30 38 2d 30 61 33 
I (3898) HTTP_CLIENT: 33 64 66 32 62 65 37 63 32 63 61 64 61 66 39 65 
I (3898) HTTP_CLIENT: 32 34 36 34 38 22 0a 20 20 7d 2c 20 0a 20 20 22 
I (3898) HTTP_CLIENT: 6f 72 69 67 69 6e 22 3a 20 22 31 38 39 2e 36 33 
I (3898) HTTP_CLIENT: 2e 34 38 2e 32 33 22 2c 20 0a 20 20 22 75 72 6c 
I (3898) HTTP_CLIENT: 22 3a 20 22 68 74 74 70 3a 2f 2f 68 74 74 70 62 
I (3898) HTTP_CLIENT: 69 6e 2e 6f 72 67 2f 67 65 74 3f 65 73 70 22 0a 
I (3898) HTTP_CLIENT: 7d 0a 
I (4838) HTTP_CLIENT: HTTP POST Status = 200, content_length = 422
I (5448) HTTP_CLIENT: HTTP PUT Status = 200, content_length = 421
I (5768) HTTP_CLIENT: HTTP PATCH Status = 200, content_length = 334
I (6068) HTTP_CLIENT: HTTP DELETE Status = 200, content_length = 335
[... 51 lines cut ...]

0x400df5a5: mem_malloc at /esp-idf/components/lwip/lwip/src/core/mem.c:237

0x400df61b: do_memp_malloc_pool at /esp-idf/components/lwip/lwip/src/core/memp.c:254

0x400df68c: memp_malloc at /esp-idf/components/lwip/lwip/src/core/memp.c:350 (discriminator 2)

0x400e01d1: pbuf_alloc_reference at /esp-idf/components/lwip/lwip/src/core/pbuf.c:336

0x400e03a9: pbuf_alloc at /esp-idf/components/lwip/lwip/src/core/pbuf.c:237

0x400efdf2: ethernetif_input at /esp-idf/components/lwip/port/esp32/netif/ethernetif.c:168

0x40127d7d: esp_netif_receive at /esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:747

0x400dbeda: eth_input_to_netif at /esp-idf/components/esp_eth/src/esp_eth_netif_glue.c:30

0x400db60b: eth_stack_input at /esp-idf/components/esp_eth/src/esp_eth.c:95

0x400d9368: emac_w5100_task at /esp32_w5100/components/ex_eth/w5100_esp32/emac/eth_mac.c:137

ELF file SHA256: 780798e4d379d9f0
```

</details>

KaeLL, 2020-06-26:
Ok. So I've been doing some more testing, and purposefully __not__ `free`ing the buffer used to receive data from the chip __isn't__ enough to stop the crashes, although they happen way __less__ often. If I free the buffer, it crashes in a consistent, punctual way. Without the free, it crashes intermittently, erratically.
Here's another log
<details><summary>Click to expand</summary>

```
CORRUPT HEAP: multi_heap.c:439 detected at 0x3ffbb8e4

abort() was called at PC 0x4009402b on core 0
0x4009402b: multi_heap_assert at /esp-idf/components/heap/multi_heap_platform.h:67
 (inlined by) multi_heap_malloc_impl at /esp-idf/components/heap/multi_heap.c:439

Backtrace:0x4009004f:0x3ffd8980 0x400906ed:0x3ffd89a0 0x4009473e:0x3ffd89c0 0x4009402b:0x3ffd8a30 0x40085b09:0x3ffd8a50 0x40085b3d:0x3ffd8a70 0x40094749:0x3ffd8a90 0x4017a2cc:0x3ffd8ab0 0x4017a34b:0x3ffd8ad0 0x4017a3c8:0x3ffd8af0 0x4017af29:0x3ffd8b10 0x4017b109:0x3ffd8b30 0x4018dd1a:0x3ffd8b60 0x401ca15d:0x3ffd8b80 0x4016d5ba:0x3ffd8ba0 0x4016cbcb:0x3ffd8bc0 0x401067ce:0x3ffd8be0
0x4009004f: panic_abort at /esp-idf/components/esp_system/panic.c:330

0x400906ed: esp_system_abort at /esp-idf/components/esp_system/system_api.c:100

0x4009473e: abort at /esp-idf/components/newlib/abort.c:46

0x4009402b: multi_heap_assert at /esp-idf/components/heap/multi_heap_platform.h:67
 (inlined by) multi_heap_malloc_impl at /esp-idf/components/heap/multi_heap.c:439

0x40085b09: heap_caps_malloc at /esp-idf/components/heap/heap_caps.c:145

0x40085b3d: heap_caps_malloc_default at /esp-idf/components/heap/heap_caps.c:177

0x40094749: malloc at /esp-idf/components/newlib/heap.c:32

0x4017a2cc: mem_malloc at /esp-idf/components/lwip/lwip/src/core/mem.c:237

0x4017a34b: do_memp_malloc_pool at /esp-idf/components/lwip/lwip/src/core/memp.c:254

0x4017a3c8: memp_malloc at /esp-idf/components/lwip/lwip/src/core/memp.c:350 (discriminator 2)

0x4017af29: pbuf_alloc_reference at /esp-idf/components/lwip/lwip/src/core/pbuf.c:336

0x4017b109: pbuf_alloc at /esp-idf/components/lwip/lwip/src/core/pbuf.c:237

0x4018dd1a: ethernetif_input at /esp-idf/components/lwip/port/esp32/netif/ethernetif.c:168

0x401ca15d: esp_netif_receive at /esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:747

0x4016d5ba: eth_input_to_netif at /esp-idf/components/esp_eth/src/esp_eth_netif_glue.c:30

0x4016cbcb: eth_stack_input at /esp-idf/components/esp_eth/src/esp_eth.c:95

0x401067ce: emac_w5100_task at /firmware/components/network/eth/w5100_esp32/emac/eth_mac.c:137

ELF file SHA256: 2ae35fc8a78b3f85

Rebooting...
```

<details>

KaeLL, 2020-06-29:
Here are a few more error logs. The 3rd log shows the LwIP thread accessing invalid memory, while the rest is caused by the task I created. They occur intermittently, probably due to something related to me not freeing the buffer I pass to `esp_eth_mediator_s::stack_input` like I mentioned in the previous comment, because if I do, the app crashes steadily.

<details>
<summary>Click to expand 1</summary>

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40093fac  PS      : 0x00060533  A0      : 0x80085ad8  A1      : 0x3ffd8650  
0x40093fac: is_free at /esp-idf/components/heap/multi_heap.c:138
 (inlined by) multi_heap_malloc_impl at /esp-idf/components/heap/multi_heap.c:439

A2      : 0x3ffe0440  A3      : 0x00000018  A4      : 0x3ffe1f04  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffe38f8  A8      : 0x00000000  A9      : 0x80aefc63  
A10     : 0x3ffe38f8  A11     : 0x00000020  A12     : 0x3ffe38fc  A13     : 0x3ffe1eb4  
A14     : 0x00000001  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x80aefc63  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x40093fa9:0x3ffd8650 0x40085ad5:0x3ffd8670 0x40085b09:0x3ffd8690 0x400946dd:0x3ffd86b0 0x401763b4:0x3ffd86d0 0x40176433:0x3ffd86f0 0x401764b0:0x3ffd8710 0x40177011:0x3ffd8730 0x401771f1:0x3ffd8750 0x40189a2e:0x3ffd8780 0x401c426d:0x3ffd87a0 0x4016af1e:0x3ffd87c0 0x4016a52f:0x3ffd87e0 0x40106326:0x3ffd8800
0x40093fa9: multi_heap_assert at /esp-idf/components/heap/multi_heap_platform.h:67
 (inlined by) multi_heap_malloc_impl at /esp-idf/components/heap/multi_heap.c:438

0x40085ad5: heap_caps_malloc at /esp-idf/components/heap/heap_caps.c:145

0x40085b09: heap_caps_malloc_default at /esp-idf/components/heap/heap_caps.c:177

0x400946dd: malloc at /esp-idf/components/newlib/heap.c:32

0x401763b4: mem_malloc at /esp-idf/components/lwip/lwip/src/core/mem.c:237

0x40176433: do_memp_malloc_pool at /esp-idf/components/lwip/lwip/src/core/memp.c:254

0x401764b0: memp_malloc at /esp-idf/components/lwip/lwip/src/core/memp.c:350 (discriminator 2)

0x40177011: pbuf_alloc_reference at /esp-idf/components/lwip/lwip/src/core/pbuf.c:336

0x401771f1: pbuf_alloc at /esp-idf/components/lwip/lwip/src/core/pbuf.c:237

0x40189a2e: ethernetif_input at /esp-idf/components/lwip/port/esp32/netif/ethernetif.c:168

0x401c426d: esp_netif_receive at /esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:747

0x4016af1e: eth_input_to_netif at /esp-idf/components/esp_eth/src/esp_eth_netif_glue.c:30

0x4016a52f: eth_stack_input at /esp-idf/components/esp_eth/src/esp_eth.c:95

0x40106326: emac_w5100_task at /firmware/components/network/eth/w5100_esp32/emac/eth_mac.c:129

ELF file SHA256: bd5e8766016ee0c9

Rebooting...
```

</details>
<details>
<summary>Click to expand 2</summary>

```
CORRUPT HEAP: multi_heap.c:438 detected at 0x3ffb62b4

abort() was called at PC 0x40093fa9 on core 1
0x40093fa9: multi_heap_assert at /esp-idf/components/heap/multi_heap_platform.h:67
 (inlined by) multi_heap_malloc_impl at /esp-idf/components/heap/multi_heap.c:438

Backtrace:0x4008ffe3:0x3ffd85a0 0x40090681:0x3ffd85c0 0x400946d2:0x3ffd85e0 0x40093fa9:0x3ffd8650 0x40085ad5:0x3ffd8670 0x40085b09:0x3ffd8690 0x400946dd:0x3ffd86b0 0x401763b4:0x3ffd86d0 0x40176433:0x3ffd86f0 0x401764b0:0x3ffd8710 0x40177011:0x3ffd8730 0x401771f1:0x3ffd8750 0x40189a2e:0x3ffd8780 0x401c426d:0x3ffd87a0 0x4016af1e:0x3ffd87c0 0x4016a52f:0x3ffd87e0 0x40106326:0x3ffd8800
0x4008ffe3: panic_abort at /esp-idf/components/esp_system/panic.c:330

0x40090681: esp_system_abort at /esp-idf/components/esp_system/system_api.c:100

[... 189 lines cut ...]

0x40176433: do_memp_malloc_pool at /esp-idf/components/lwip/lwip/src/core/memp.c:254

0x401764b0: memp_malloc at /esp-idf/components/lwip/lwip/src/core/memp.c:350 (discriminator 2)

0x40177011: pbuf_alloc_reference at /esp-idf/components/lwip/lwip/src/core/pbuf.c:336

0x401771f1: pbuf_alloc at /esp-idf/components/lwip/lwip/src/core/pbuf.c:237

0x40189a2e: ethernetif_input at /esp-idf/components/lwip/port/esp32/netif/ethernetif.c:168

0x401c426d: esp_netif_receive at /esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:747

0x4016af1e: eth_input_to_netif at /esp-idf/components/esp_eth/src/esp_eth_netif_glue.c:30

0x4016a52f: eth_stack_input at /esp-idf/components/esp_eth/src/esp_eth.c:95

0x40106326: emac_w5100_task at /firmware/components/network/eth/w5100_esp32/emac/eth_mac.c:129

ELF file SHA256: bd5e8766016ee0c9

Rebooting...
```

</details>

david-cermak, 2020-07-03:
Hi @KaeLL 

With the current IDF lwip ethernet port implementation, you're not supposed to free the L2 buffer. And yes, the `LWIP_L2_TO_L3_COPY` works only for the wifi port, as it supports both referencing and copying the buffer to L3. I know that it's not very readable, but the emac receive function calls the [ethernet_input](https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/netif/ethernetif.c#L168) where the pbuf of `PBUF_REF` type is created. at the same time we setup the [`netif->l2_buffer_free_notify`](https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/netif/ethernetif.c#L230) to free the L2 buffer when reference type of pbuf deallocates. (Note, that this is an esp-lwip patch and is already planned for refactoring into standard external pbufs).

As for the other issue, the intermittent crash on heap corruption, it seems to be related to some incorrect pbuf management (very unlikely your not freeing the allocated buffer -- should be cleared [here](https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/netif/ethernetif.c#L68)), which I cannot tell reading your code. What looks suspicious, though, is that you allocate the buffer in the `eth_mac.c` with standard malloc:
```
I (2299) w5100_mac: ADDRESS : 0x3ffb78f8       <- This is the buffer I allocated and later will attempt to free
```
And the same buffer is freed as using `pbuf_free()`!
```
pbuf_free(0x3ffb78f8)                                  <-------- Caught with the pants down :P
```
instead of [`free()`](https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/netif/ethernetif.c#L68) here.

There must be a problem in misuse the data pointer you allocate as `struct pbuf *` pointer, but this seems like you can easily track it down when deliberately freeing the buffer and let the double-delete issue occur.

KaeLL, 2020-07-03:
I'm sorry, I'm not following. Let's me brake down my understanding of what you're saying.

> With the current IDF lwip ethernet port implementation, you're not supposed to free the L2 buffer.

Right. I learnt that the hard way. But isn't that not ideal? Who allocates should deallocate, no? Of course, the argument stems from the whole good practices stuff, which I'm more than happy to ignore so long as you guarantee the stability of this behavior.

> the emac receive function calls the ethernet_input where the pbuf of PBUF_REF type is created.

Dabbling through that code is how I managed to test my hypothesis that I should indeed not free the buffer I was allocating. And it's not really that bad.

> at the same time we setup the netif->l2_buffer_free_notify to free the L2 buffer when reference type of pbuf deallocates.

As in, the buffer is reference counted?

> (Note, that this is an esp-lwip patch and is already planned for refactoring into standard external pbufs)

Perhaps as a consequence of not getting the previous statement, I also don't understand what exactly this one is referring to too.

> (very unlikely your not freeing the allocated buffer -- should be cleared here)

True. I am not. In fact, you can assume I didn't free the buffer for all all but the first original log in this thread, because every time I tried to free the L2 buffer I allocated, I'd consistently get a double free error that pointed to my free instead of lwIP's.

> which I cannot tell reading your code

Perhaps that's due to the link I posted being invalid because of the changes I made. Sorry about that, I'll update it. [Here](https://github.com/KaeLL/w5100_esp32/blob/34cb0f6fc3da9d2b517a126eb233ea4c28755448/eth_mac.c#L117-L129)'s the link to the code we're talking about. If you have any doubts, let me know.

> There must be a problem in misuse the data pointer **you** allocate as struct pbuf * pointer

I don't. Like it's being shown in the code linked above, I'm allocating using standard `malloc` and storing that on a `uint8_t *`.

> but this seems like you can easily track it down when deliberately freeing the buffer and let the double-delete issue occur.

Which would lead me to

> every time I tried to free the L2 buffer I allocated, I'd consistently get a double free error that pointed to my free instead of lwIP's.

And the log would go straight to `emac_w5100_task` -> `free` -> newlib stuff isntead of the `eth_stack_input` -> lwIP internals.

KaeLL, 2020-08-24:
I guess I didn't nag you more over this issue because I stopped having this problem after a few changes, which include not freeing the L2 buffer. Thanks for the clarifications. I'll close the issue.

closed 2020-08-24 by KaeLL