espressif/esp-idf #5487: advanced_https_ota is missing the README

ferrarimarco, 2020-06-23:
## Environment

N/A

## Problem Description

README for the `advanced_https_ota` example is missing.

### Expected Behavior

I was expecting a README that explains the example, and high-level guidance about that example in https://github.com/espressif/esp-idf/blob/master/examples/system/ota/README.md.

### Actual Behavior

### Steps to reproduce

1. Open https://github.com/espressif/esp-idf/tree/master/examples/system/ota/advanced_https_ota
2. :(

Alvin1Zhang, 2020-06-24:
@ferrarimarco Thanks for reporting and pointing out, we will look into.

shubhamkulkarni97, 2020-07-03:
@ferrarimarco, README files in examples directory typically provide guidance regarding usage and setup required to run a particular example.

As the setup for simple_ota, advanced_https_ota and native_ota examples is same, single README file is provided in [OTA](https://github.com/espressif/esp-idf/tree/master/examples/system/ota) directory.

Moreover, advanced_https_ota example rely on esp_https_ota component for performing OTA updates.

Can you please elaborate on what information would you like to be added in README?

Thanks,
Shubham

ferrarimarco, 2020-07-03:
It's strange that `advanced_https_ota` is not mentioned in https://github.com/espressif/esp-idf/blob/master/examples/system/ota/README.md. What are the differences between advanced_https_ota and esp_https_ota?

To gather the information you reported, the only way is to look at the code.

So, what I think it's missing is

1. A README for `advanced_https_ota`, that explains what it does, and how it augments `esp_https_ota`
1. A mention of `advanced_https_ota` in https://github.com/espressif/esp-idf/blob/master/examples/system/ota/README.md, otherwise users are left wondering why you didn't describe the `advanced_https_ota`. My first thought was that `advanced_https_ota` was a work-in-progress, but it's not the case.

shubhamkulkarni97, 2020-07-07:
Thanks for providing your thoughts, we will improve README to avoid confusion.
> It's strange that `advanced_https_ota` is not mentioned in https://github.com/espressif/esp-idf/blob/master/examples/system/ota/README.md. What are the differences between advanced_https_ota and esp_https_ota?
> 

`esp_https_ota` is a component (abstraction layer) which uses `app_update` component to perform OTA updates. While `advanced_https_ota` is an example application which uses `esp_https_ota` component.

closed 2020-09-16 by espressif-bot