espressif/esp-idf #5488: btc_transfer_context failes in btc_gattc_cback due to BT_STATUS_NOMEM

elad30300, 2020-06-23:
I use esp-idf v3.3 with esp32-wroom-32 chip. In my firmware i use ble gatt client to get data from sensor. After a while i cannot receive messages and got the error BT_BTC: btc_gattc_cback transfer failed. After change the log in this function I found out that the failure is due to BT_STATUS_NOMEM in btc_transfer_context. I also started to log the heap size in every callback call, and i can see it is decreased until something like 200 bytes and then it stopped to call the callback because the btc_transfer_context couldn't allocate what it needs so it didn't call my callback.
Is there any way to fix this problem?
It looks like memory like in the esp-idf version i use.

Alvin1Zhang, 2020-06-24:
@elad30300 Thanks for reporting and letting us know, we will look into. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID and etc. would help us debug further. Thanks.

elad30300, 2020-06-24:
Hi @Alvin1Zhang, I put the info there

Alvin1Zhang, 2020-06-24:
Thanks for updates, we will look into.

elad30300, 2020-06-25:
Alright, thank you, looking forward for updates.

WCCWCC, 2020-08-14:
Hi @elad30300 

* The device needs to perform flow control when sending notify or write no response packets.

* You can refer to the example program we provide. That example provides a way to reach the maximum rate using minimum memory.

https://github.com/espressif/esp-idf/tree/release/v3.3/examples/bluetooth/ble_throughput.

closed 2025-03-14 by Alvin1Zhang