espressif/esp-idf #5492: [CMake Error] xtensa-esp32-elf-gcc is not a full path and was not found in the PATH.

NguyenMinhTri, 2020-06-24:
**Environment**
Operating System: Win 10 Pro - 64bit
Editor: Visual code.
[
[image]
](url)
**Log**
```
Executing action: menuconfig
Running cmake in directory d:\research\esp32_workstation\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 d:\research\esp32_workstation\hello_world"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.21.0.windows.1") 
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: xtensa-esp32-elf-gcc
CMake Error at C:/esp32/esp-idf/tools/cmake/project.cmake:307 (__project):   
  The CMAKE_C_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

CMake Error at C:/esp32/esp-idf/tools/cmake/project.cmake:307 (__project):
  The CMAKE_CXX_COMPILER:

    xtensa-esp32-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

CMake Error at C:/esp32/esp-idf/tools/cmake/project.cmake:307 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "D:/Research/ESP32_WorkStation/hello_world/build/CMakeFiles/CMakeOutput.log".
See also "D:/Research/ESP32_WorkStation/hello_world/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1
```

igrr, 2020-06-24:
@NguyenMinhTri Could you clarify, are you running `idf.py menuconfig` in the same command prompt window where you have executed `export.bat`, or a different one?

NguyenMinhTri, 2020-06-24:
> @NguyenMinhTri Could you clarify, are you running `idf.py menuconfig` in the same command prompt window where you have executed `export.bat`, or a different one?

I am running this command in example project.
[image]

igrr, 2020-06-24:
I understand that these are in different directories, but are these two commands executed in the same command prompt window? Or are you running export.bat in one window, and then idf.py menuconfig in another one?

If you are running them in the same window, could you run `where xtensa-esp32-elf-gcc` and `echo %PATH%` commands after `idf.py menuconfig` fails?

NguyenMinhTri, 2020-06-24:
Thank you for the prompt response!! There is no error when I ran these in the same command prompt window.

closed 2020-06-24 by NguyenMinhTri