espressif/esp-idf #5493: aws_iot: failed! mbedtls_ssl_handshake returned -0x6800

lxdemon, 2020-06-24:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it): v3.3
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- (Windows only) environment type: 
- Using an IDE?: [No]
- Power Supply:         [external 3.3V]

## Problem Description
After power up, sometimes device can not connect to AWS, then i make it restart use esp_restart, usually it is work well, i din't know where is the problem

### Expected Behavior

### Actual Behavior

### Steps to reproduce

just power up device, usually i can get this problem

### Code to reproduce this issue

```cpp

void app_main()
{
    somecode;
    xTaskCreate(aws_iot_task, "aws_iot_task", 5120, NULL, 8, &pTaskAWSIotHandle);
}

void aws_iot_task(void *param) 
{
    somecode;
    connectParams.keepAliveIntervalInSec = 60;
    connectParams.isCleanSession = true;
    connectParams.MQTTVersion = MQTT_3_1_1;
    /* Client ID is set in the menuconfig of the example */
    connectParams.pClientID = thingID;
    connectParams.clientIDLen = (uint16_t) strlen(thingID);
    connectParams.isWillMsgPresent = false;

    ESP_LOGI(TAG, "Connecting to AWS...");
    ESP_LOGD(TAG, "Stack remaining for task '%s' is '%d' bytes", pcTaskGetTaskName(NULL), uxTaskGetStackHighWaterMark(NULL));
    ESP_LOGD(TAG, "Free memory for application is '%d' bytes", esp_get_free_heap_size());
    // uint8_t conn_retry_cnt = 0;
    do {
        rc = aws_iot_mqtt_connect(&client, &connectParams);
        if(SUCCESS != rc) {
            ESP_LOGE(TAG, "Error(%d) connecting to %s:%d", rc, mqttInitParams.pHostURL, mqttInitParams.port);
            vTaskDelay(1000 / portTICK_RATE_MS);

            // if (++conn_retry_cnt > 15){
                // ESP_LOGE(TAG, "Failed to connecting to AWS more than 15 times, AWS task is going to be delete");
                ESP_LOGE(TAG, "Failed to connecting to AWS, reboot device");
                vTaskDelay(1000 / portTICK_RATE_MS);
                esp_restart();
            // }
        }
    } while(SUCCESS != rc);
}

```

## Debug Logs

```
[... 122 lines cut ...]
I (702) wifi: new:<2,0>, old:<1,0>, ap:<255,255>, sta:<2,0>, prof:1
I (1842) wifi: state: init -> auth (b0)
I (1872) wifi: state: auth -> assoc (0)
I (1922) wifi: state: assoc -> run (10)
I (2062) wifi: connected with HeyGoogle, channel 2, HT20, bssid = ec:41:18:ef:6a:66
I (2142) wifi: pm start, type: 1

I (2862) event: sta ip: 192.168.18.55, mask: 255.255.255.0, gw: 192.168.18.1
I (2862) main: SYSTEM_EVENT_STA_GOT_IP: 3712a8c0
I (2862) main: Device connected to ap
I (2862) AWS Iot: AWS IoT SDK Version 3.0.1-
I (2862) tcp handle: master socket create success, socket fd: 54
I (2882) AWS Iot: Connecting to AWS...
I (2882) tcp handle: master socket bind success
I (2892) tcp handle: master socket is listening on port 10000
D (2892) AWS Iot: Stack remaining for task 'aws_iot_task' is '3396' bytes
I (2892) tcp handle: Waiting for TCP connections ...
D (2902) tcp handle: update readfds:        0
D (2912) AWS Iot: Free memory for application is '4340520' bytes
D (18832) AWS Iot: sub topic: $aws/things/BW8453_550831/shadow/update/accepted
D (18832) AWS Iot: pub topic: $aws/things/BW8453_550831/shadow/update
I (18832) AWS Iot: Subscribing...
D (18982) AWS Iot: Stack remaining for task 'aws_iot_task' is '1252' bytes
I (18982) AWS Iot: Subscribed, Free memory for application is '4313284' bytes
```

Alvin1Zhang, 2020-06-24:
@lxdemon Thanks for reporting, we will look into.

alibasta, 2020-07-08:
@lxdemon same problem here.

shahpiyushv, 2020-07-10:
@lxdemon , the below lines indicate SSL timed out which generally can occur due to a network issue.

```
E (15291) aws_iot: failed! mbedtls_ssl_handshake returned -0x6800
E (15291) AWS Iot: Error(-4) connecting to a2f8ob0zg2vz65.iot.ap-northeast-1.amazonaws.com:8883
```

Did you have the 15 attempts logic always commented out? What happens if you keep retrying for some time?

Meanwhile, on a completely unrelated note, you need to use the new iot:Data-ATS MQTT endpoint as per [Amazon's recommendations](https://docs.aws.amazon.com/cli/latest/reference/iot/describe-endpoint.html#options).

Alvin1Zhang, 2020-10-15:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates, thanks.

closed 2020-10-15 by Alvin1Zhang

MaazSk, 2021-08-14:
Hie @Alvin1Zhang, I am facing the same issue is there been any progress or is the issue been resolved because I am getting the same issue I am using esp-idf version 4.3. 

```
I (6773) aws_iot: Shadow Connect
D (6773) aws_iot: Seeding the random number generator...
D (6783) aws_iot: Loading CA root certificate from file ...
D (6783) aws_iot: ok (0 skipped)
D (6793) aws_iot: Loading client cert from file...
D (6793) aws_iot: Loading client private key from file...
D (6803) aws_iot: ok
D (6803) aws_iot: Connecting to a3nwac7rt2fwp8-ats.iot.us-west-2.amazonaws.com/8883...
D (8493) aws_iot: ok
D (8493) aws_iot: Setting up the SSL/TLS structure...
D (8493) aws_iot: SSL state connect : 0
D (8493) aws_iot: ok
D (8493) aws_iot: SSL state connect : 0
D (8493) aws_iot: Performing the SSL/TLS handshake...
E (13503) aws_iot: failed! mbedtls_ssl_handshake returned -0x6800
E (13503) aws_iot: Shadow Connection Error
E (13503) run_device: Device exited with error from the manage_device_states function.
```