espressif/esp-idf #5494: This problem still exists in ESP-IDF v4.1-beta2-141-g84b51781c.

vvhh2002, 2020-06-24:
This problem still exists in ESP-IDF v4.1-beta2-141-g84b51781c.
I use the btstack hfp_hs example code. when the SCO connection is established, the following error appears:

```

I (73122) BTS: hfp_hf.c.625: HFP: sending hci_accept_connection_request, sco_voice_setting 0x43
I (73252) BTS: hci.c.2247: Synchronous Connection Complete (status=0) XX:XX:XX:XX:XX:XX
I (73252) BTS: hfp.c.699: eSCO Connection established. 

I (73262) BTS: hfp.c.707: sco_handle 0x181, address E0:33:8E:A8:81:35, transmission_interval 12 slots, retransmission_interval 2 slots,  rx_packet_length 60 bytes, tx_packet_length 60 bytes, air_mode 0x 3 (0x02 == CVSD)

Audio connection established with SCO handle 0x0181.
Using mSBC codec.
I (732ASSERT_PARAM(1 0), in rwbt.c at line 314
82) sco_demo: data: size 63
I (73292) sco_demo: 81 01 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 
I (73292) sco_demo: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
I (73292) sco_demo: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
I (73292) sco_demo: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
I (73292) sco_demo: SCO CRC Error: 2 - data: 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400859d0  PS      : 0x00060034  A0      : 0x8008a4f8  A1      : 0x3ffbe290  
0x400859d0: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffbdbb4  A7      : 0x60008050  A8      : 0x800859cd  A9      : 0x3ffbe270  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40085905  LEND    : 0x4008590c  LCOUNT  : 0x00000000  
0x40085905: r_assert_param at ??:?

0x4008590c: r_assert_param at ??:?

Core 0 was running in ISR context:
EPC1    : 0x40132106  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400859d0
0x40132106: esp_pm_impl_waiti at /esp/esp-idf/components/esp32/pm_esp32.c:486

0x400859d0: r_assert_param at ??:?

ELF file SHA256: 630e8609643bb6a6

Backtrace: 0x400859cd:0x3ffbe290 0x4008a4f5:0x3ffbe2b0 0x4008a913:0x3ffbe2d0 0x40084df5:0x3ffbe2f0 0x40132103:0x3ffca9c0 0x400d668b:0x3ffca9e0 0x40095121:0x3ffcaa00 0x40092e81:0x3ffcaa20
0x400859cd: r_assert_param at ??:?

0x4008a4f5: r_rwbt_isr at ??:?

0x4008a913: r_rwbtdm_isr_wrapper at intc.c:?

0x40084df5: _xt_lowint1 at /esp/esp-idf/components/freertos/xtensa_vectors.S:1153

0x40132103: esp_pm_impl_waiti at /esp/esp-idf/components/esp32/pm_esp32.c:484

0x400d668b: esp_vApplicationIdleHook at /esp/esp-idf/components/esp_common/src/freertos_hooks.c:86

0x40095121: prvIdleTask at /esp/esp-idf/components/freertos/tasks.c:4691

0x40092e81: vPortTaskWrapper at /esp/esp-idf/components/freertos/port.c:436

Core 1 register dump:
PC      : 0x40132106  PS      : 0x00060234  A0      : 0x800d668e  A1      : 0x3ffcafc0  
0x40132106: esp_pm_impl_waiti at /esp/esp-idf/components/esp32/pm_esp32.c:486

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x800d60aa  A9      : 0x3ffcaf90  
A10     : 0x00000000  A11     : 0x00060623  A12     : 0x00060620  A13     : 0x3ffc2500  
A14     : 0x3ffbbdb0  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 630e8609643bb6a6

Backtrace: 0x40132103:0x3ffcafc0 0x400d668b:0x3ffcafe0 0x40095121:0x3ffcb000 0x40092e81:0x3ffcb020
0x40132103: esp_pm_impl_waiti at /esp/esp-idf/components/esp32/pm_esp32.c:484

0x400d668b: esp_vApplicationIdleHook at /esp/esp-idf/components/esp_common/src/freertos_hooks.c:86

0x40095121: prvIdleTask at /esp/esp-idf/components/freertos/tasks.c:4691

0x40092e81: vPortTaskWrapper at /esp/esp-idf/components/freertos/port.c:436

Rebooting...
ets Jun  8 2016 00:22:57

```

_Originally posted by @vvhh2002 in https://github.com/espressif/esp-idf/issues/1322#issuecomment-648947186_

Alvin1Zhang, 2020-06-29:
Thanks for reporting, we will look into.

Wth-Esp, 2020-07-09:
Does this appear every time?

vvhh2002, 2020-07-15:
yes.
you could try btstack hpf demo.

boblane1, 2020-10-30:
Hi, @vvhh2002 

The issue is caused by AES-CCM decryption, the previous packet can not be decrypted completely before the next packet is received by the controller. For now, ESP32 solve this problem by use of E0 encryption.

mringwal, 2021-01-26:
I'd like to confirm that sending SCO with BTstack's HPF HF Demo works if BR/EDR Secure Connections are disabled with `gap_secure_connections_enable(false);`

Any chance for this to get fixed? Thanks!
(BR/EDR Secure Connections isn't a big thing, but nice to have )

xiongweichao, 2024-07-09:
Hi @mringwal , I'm very sorry for replying to you so late. There are currently no plans to fix it on ESP32. Suggest disabling BR/EDR secure connections. Thanks!

mringwal, 2024-07-10:
BR/EDR Secure Connections allows for Cross-Transport Key Derivation, which is an important features since iOS and newer Android devices support it.

It would be quite disappointing if this feature could not be used at all, although it's officially supported by ESP32 and documented in the Bluetooth SIG listing.