espressif/esp-idf #5495: Bluedroid - limit of handles/services

chegewara, 2020-06-24:
Hi,
this question comes from arduino users, but i tested it with esp-idf master and i cant answer on it.
Why there is limit in services we can create with bluedroid? When we are trying to create more than 6 services we are getting this error:
```
W (1401) my_gatts_event_handler: custom gatts event handler, event: 0
W (1401) my_gatts_event_handler: custom gatts event handler, event: 7
I (1401) : Service 0 defined
W (1411) my_gatts_event_handler: custom gatts event handler, event: 7
I (1411) : Service 1 defined
W (1421) my_gatts_event_handler: custom gatts event handler, event: 7
I (1421) : Service 2 defined
W (1431) my_gatts_event_handler: custom gatts event handler, event: 7
I (1431) : Service 3 defined
W (1441) my_gatts_event_handler: custom gatts event handler, event: 7
I (1441) : Service 4 defined
W (1451) my_gatts_event_handler: custom gatts event handler, event: 7
I (1451) : Service 5 defined
E (1461) BT_GATT: GATTS_ReserveHandles: no free handle blocks 

E (1461) BT_APPL: service creation failed.
W (1471) my_gatts_event_handler: custom gatts event handler, event: 7
I (1471) : Service 6 defined
I (1481) : All Services defined
W (1481) my_gatts_event_handler: custom gatts event handler, event: 12
I (1481) : Service 0 started
W (1491) my_gatts_event_handler: custom gatts event handler, event: 12
I (1491) : Service 1 started
I (1511) : Service 2 started
W (1511) my_gatts_event_handler: custom gatts event handler, event: 12
W (1521) my_gatts_event_handler: custom gatts event handler, event: 12
I (1521) : Service 3 started
W (1531) my_gatts_event_handler: custom gatts event handler, event: 12
I (1531) : Service 4 started
W (1541) my_gatts_event_handler: custom gatts event handler, event: 12
I (1541) : Service 5 started
E (1551) BT_APPL: service not created
```
Important is this error:
`E (1461) BT_GATT: GATTS_ReserveHandles: no free handle blocks `
It does not matter how many handles i will use for each service, 3, 15 or 50, the result is always the same.

Thanks.

https://github.com/nkolban/esp32-snippets/issues/955

chegewara, 2020-06-24:
One more thing.
`esp_ble_gatts_create_service` is returning `ESP_OK` even if no free handles error occurred and service creation failed.

WCCWCC, 2020-06-29:
Hi,  @chegewara 
* BLE STACK uses GATT_MAX_SR_PROFILES to create the number of gatt servers now, the default value is 8.
* BLE STACK will also create GAP SERVICE itself, leaving the actual number of users to 6.
* If you want to create more services, you can increase this macro definition as needed

chegewara, 2020-06-29:
That explains all, but i still think its a bug, because i believe that handle returned for 7th service should be 0xffff rather than some random value. Also `esp_ble_gatts_create_service` should return error in such case.

Thanks.

WCCWCC, 2020-06-30:
Hi,  @chegewara 
  
* There are two restrictions in the process of creating a service:
   
   * If limited by the maximum number of services, the following log will appear
           **GATTS_ReserveHandles: no free handle blocks**
   * If the maximum number of services meets the conditions,If the maximum number of services meets the conditions, the following log will appear
           **GATTS_ReserveHandles: no handles, s_hal: needed**

 * You can use different logs to distinguish what happened.

chegewara, 2020-06-30:
Yes, i eventually found those logs:
```
E (1461) BT_GATT: GATTS_ReserveHandles: no free handle blocks 
E (1461) BT_APPL: service creation failed.
```
but problem is with arduino.
Of course i dont even need such logs now if i know that i am limited to 6 services.
The question is how this `esp_ble_gatts_create_service` can return `ESP_OK` if service cant be created due to lack of free handles? Services is not created, is that correct?
Here is documentation:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gatts.html#_CPPv428esp_ble_gatts_create_service13esp_gatt_if_tP18esp_gatt_srvc_id_t8uint16_t
```
Return
ESP_OK : success

other : failed
```
which clearly says it should return value other than `ESP_OK` if is not success.

WCCWCC, 2020-06-30:
Hi, @chegewara 

* Bluetooth handles API requests asynchronously, server creation succeeds or fails, you need to check the status of event ESP_GATTS_CREATE_EVT and API(esp_ble_gatts_create_service) return value .

* The API returns successfully. It does not mean that the corresponding operation was successfully executed. It only means that the API call was successful.

chegewara, 2020-06-30:
Ok, i think i found the issue. This code in library is not checking status:
```
		case ESP_GATTS_CREATE_EVT: {
			if (getUUID().equals(BLEUUID(param->create.service_id.id.uuid)) && m_instId == param->create.service_id.id.inst_id) {
				setHandle(param->create.service_handle);
				m_semaphoreCreateEvt.give();
			}
			break;
		} // ESP_GATTS_CREATE_EVT
```

Thanks.

closed 2020-06-30 by chegewara