espressif/esp-idf #5498: ESP IDF Mesh Broadcast feature not working

Suraj-singh999898, 2020-06-25:
ESP IDF version:   release/v4.0
Development Kit:    ESP WROOM 32 MCU Module
Build System:       make
Compiler version:   xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Operating System:   Ubuntu 19.10
Using IDE with Arduino esp32 as a component

Expected Behavior
    Message should receive in ESP noded which has the same Broadcast group id.

Actual Behavior
    Message does not receive when changing broadcast group id. only working 
    Setting a new broadcast group id in all ESP32 devices, then broadcasting by root node message didn't receive but when I use 0xff:0xff:0xff:0xff:0xff:0xff broadcast Id then all node get data.

basically I want to divide my One mesh network into 2 Groups.
Below GroupID Setting code

```
    mesh_addr_t GroupID = {
            .addr = {0xff, 0xff, 0xff, 0xff, 0xff, 0x00}
    };

    esp_err_t errReturn = esp_mesh_set_group_id(&GroupID, 1);
    if (errReturn){
        Serial.print ( "Group ID can't set" );
        Serial.println ( esp_err_to_name (errReturn) );
    }else {
        Serial.print ( "Group ID set successfully" );
    }

    ESP_ERROR_CHECK(esp_mesh_start());  /* mesh start */
```

Below Broadcast sending code

```
mesh_addr_t broadcast_group_id = {
            .addr = {0xff, 0xff, 0xff, 0xff, 0xff, 0xff}    // Working
//          .addr = {0xff, 0xff, 0xff, 0xff, 0xff, 0x00}    // not working
    };

    mesh_data_t data = {};
    data.data   = (uint8_t *) msg.c_str();
    data.size   = msg.length();
    data.proto  = MESH_PROTO_BIN;
    data.tos    = MESH_TOS_P2P;

//  esp_err_t errReturn = esp_mesh_send( &broadcast_group_id, &data, MESH_DATA_GROUP, NULL, 1);                 // not working
//  esp_err_t errReturn = esp_mesh_send( &broadcast_group_id, &data, MESH_DATA_P2P, NULL, 1);                   // working
    esp_err_t errReturn = esp_mesh_send( &broadcast_group_id, &data, MESH_DATA_P2P, NULL, MESH_OPT_SEND_GROUP); // not working

    if (errReturn)
    {
        Serial.println ("---------------> broadcastToNodes\t: esp_mesh_sending fail");
        Serial.println ( esp_err_to_name (errReturn) );
    }else{
        Serial.println ("---------------> broadcastToNodes\t: Mesh Broadcast to all nodes success");
    }

```

ESP-iPENCIL, 2020-06-29:
@Suraj-singh999898  Please change to the below and try again.

`esp_err_t errReturn = esp_mesh_send( &broadcast_group_id, &data, MESH_DATA_GROUP, NULL, 0);   `
 

I tested your code excpet the above change in the IDF/v4.0 commit: e9dcba0d (wifi firmware version: d74ef33) , the group sending works well.

Please let me know your test result. If you fails again, please tell me your commit ID and the error code returned by the `esp_mesh_send`,  thanks.

Alvin1Zhang, 2020-08-11:
@Suraj-singh999898 Thanks for reporting, would you please help share if any updates for this issue? Thanks.

Suraj-singh999898, 2020-08-11:
Thanks, its working!

closed 2020-08-11 by Suraj-singh999898