espressif/esp-idf #5499: esp_http_client_cleanup() corrupts heap when LWIP stats enabled

brianbienvenu, 2020-06-25:
## Environment

- Development Kit:      ESP32-Ethernet-Kit
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.0.1
- Build System:         idf.py (? vscode-esp-idf-extension)
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: vscode-esp-idf-extension
- Using an IDE?: VS Code
- Power Supply:         USB

## Problem Description

Building the esp-http-client example code from esp-idf and changing enabling the LWIP Statistics menuconfig option causes a reliable crash at esp_http_client_cleanup().

I first noticed this in my own application, but the below seems to be the minimal example. I hope they are both coming from the same underlying issue! Turning off LWIP Stats on my application does seem to run without crashing. 

Adding a liberal number of calls to `heap_caps_check_integrity_all()` helped determine the cause was heap corruption (bad head), although the addresses and overwritten data change from build to build.

### Expected Behavior
It doesn't crash.

### Actual Behavior
It crashes reliably.

### Steps to reproduce

1. Copy esp-http-client example, open directory in VS Code.
1. Change menuconfig settings:
    1. Enable Connect Using -> "Ethernet"
    1. Enable "Enable LWIP statistics"
1. Build, flash, monitor
1. Crash on first call to `esp_http_client_cleanup(client);` (line 160)

## Debug Logs

```
<snip>
I (6524) HTTP_CLIENT: Connected to AP, begin http example
I (7054) HTTP_CLIENT: HTTP GET Status = 200, content_length = 270
I (8524) HTTP_CLIENT: HTTP POST Status = 200, content_length = 444
I (9024) HTTP_CLIENT: HTTP PUT Status = 200, content_length = 443
I (9274) HTTP_CLIENT: HTTP PATCH Status = 200, content_length = 335
I (9504) HTTP_CLIENT: HTTP DELETE Status = 200, content_length = 336
I (9774) HTTP_CLIENT: HTTP HEAD Status = 200, content_length = 270
I (9774) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
abort() was called at PC 0x40082a4a on core 0
0x40082a4a: lock_acquire_generic at C:/Users/Brian/esp-idf/components/newlib/locks.c:143

ELF file SHA256: 1d503f88f9e07ee5

Backtrace: 0x4008540d:0x3ffc0200 0x40085791:0x3ffc0220 0x40082a4a:0x3ffc0240 0x40082b6d:0x3ffc0270 0x4011c131:0x3ffc0290 0x40117cbd:0x3ffc0550 0x40117c45:0x3ffc05a0 0x40087587:0x3ffc05d0 0x4008788e:0x3ffc05f0 0x4008236a:0x3ffc0610 0x4008aea1:0x3ffc0630 0x400f73e2:0x3ffc0650 0x400f91f9:0x3ffc0670 0x400f9161:0x3ffc06a0 0x400f7f4a:0x3ffc06c0 0x400e3a5e:0x3ffc06e0 0x400e3a89:0x3ffc0700 0x400e52fc:0x3ffc0730 0x400d4dbf:0x3ffc0750 0x4000bd4b:0x3ffc0770 0x40001781:0x3ffc0790 0x400d99ee:0x3ffc07b0 0x40126adb:0x3ffc07d0 0x400d7cb9:0x3ffc07f0 0x400d7cd1:0x3ffc0810 0x400d68d1:0x3ffc0830 0x400d7193:0x3ffc08c0 0x40087d31:0x3ffc08e0
0x4008540d: invoke_abort at C:/Users/Brian/esp-idf/components/esp32/panic.c:155

0x40085791: abort at C:/Users/Brian/esp-idf/components/esp32/panic.c:172

0x40082a4a: lock_acquire_generic at C:/Users/Brian/esp-idf/components/newlib/locks.c:143

0x40082b6d: _lock_acquire_recursive at C:/Users/Brian/esp-idf/components/newlib/locks.c:171

0x4011c131: _vfiprintf_r at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)

0x40117cbd: fiprintf at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fiprintf.c:48

0x40117c45: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:58 (discriminator 8)

0x40087587: get_next_block at C:/Users/Brian/esp-idf/components/heap/multi_heap.c:124
 (inlined by) assert_valid_block at C:/Users/Brian/esp-idf/components/heap/multi_heap.c:164

0x4008788e: multi_heap_free_impl at C:/Users/Brian/esp-idf/components/heap/multi_heap.c:476

0x4008236a: heap_caps_free at C:/Users/Brian/esp-idf/components/heap/heap_caps.c:268

0x4008aea1: free at C:/Users/Brian/esp-idf/components/newlib/heap.c:47

0x400f73e2: sys_mbox_free at C:/Users/Brian/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:355

0x400f91f9: netconn_drain at C:/Users/Brian/esp-idf/components/lwip/lwip/src/api/api_msg.c:888

0x400f9161: netconn_free at C:/Users/Brian/esp-idf/components/lwip/lwip/src/api/api_msg.c:831

0x400f7f4a: netconn_delete at C:/Users/Brian/esp-idf/components/lwip/lwip/src/api/api_lib.c:257

0x400e3a5e: free_socket_free_elements at C:/Users/Brian/esp-idf/components/lwip/lwip/src/api/sockets.c:637

0x400e3a89: free_socket at C:/Users/Brian/esp-idf/components/lwip/lwip/src/api/sockets.c:673

0x400e52fc: lwip_close at C:/Users/Brian/esp-idf/components/lwip/lwip/src/api/sockets.c:870

0x400d4dbf: esp_vfs_close at C:/Users/Brian/esp-idf/components/vfs/vfs.c:487 (discriminator 4)

0x400d99ee: tcp_close at C:/Users/Brian/esp-idf/components/tcp_transport/transport_tcp.c:167

0x40126adb: esp_transport_close at C:/Users/Brian/esp-idf/components/tcp_transport/transport.c:214

0x400d7cb9: esp_http_client_close at C:/Users/Brian/esp-idf/components/esp_http_client/esp_http_client.c:1206

0x400d7cd1: esp_http_client_cleanup at C:/Users/Brian/esp-idf/components/esp_http_client/esp_http_client.c:616

0x400d68d1: http_rest_with_url at C:\Users\Brian\Documents\ESP_Play\esp-http-client\esp_http_client\build/../main/esp_http_client_example.c:160

0x400d7193: http_test_task at C:\Users\Brian\Documents\ESP_Play\esp-http-client\esp_http_client\build/../main/esp_http_client_example.c:517

0x40087d31: vPortTaskWrapper at C:/Users/Brian/esp-idf/components/freertos/port.c:143

Rebooting...
<snip>
```

Alvin1Zhang, 2020-06-29:
Thanks for reporting, we will look into.

david-cermak, 2020-07-03:
Hi @brianbienvenu 

Could you please post your full configuration? Does this crass occur every time or just intermittently?
We were not able to reproduce your issue, just enabling ethernet and lwip statistics. There must be something we miss.

Alvin1Zhang, 2020-07-29:
@brianbienvenu Thanks for reporting, feel free to reopen with more details, thanks.

closed 2020-07-29 by Alvin1Zhang