espressif/esp-idf #5500: Intermittently fails to boot despite 3.3V and GND pins set correctly

lr399, 2020-06-25:
## Environment

- Development Kit:      [ESP32-DevKitC|none own board now]
- Kit version (for WroverKit/PicoKit/DevKitC): [|v4]
- Module or chip used:  ESP32-WROOM-32D
- IDF version  framework-espidf 3.30300.190916 (3.3.0)
- Build System:         platformIO
- Compiler version toolchain-xtensa32 2.50200.80 (5.2.0)
- Operating System:     [indows
- (Windows only) environment type: platformIO
- Using an IDE?: [No|Yes (please give details)] VSC with platformIO
- Power Supply:         external 3.3V (stepped down from 9V battery using regulator)

## Problem Description

The above board is being used on a custom PCB. One of the primary roles of the PCB is a power latching circuit to ensure no power is pulled from the battery when the device is off.

When the ESP is turned off, esp_restart() is called and we drive a GPIO pin high which unlatches the power supply. This means as the board shuts down the power is physically disconnected.

When we turn the device on, the power is supplied to the ESP once again and it turns on. However one time in approximately 250 on/off cycles, it fails to boot. When this happens, I've checked with the multimeted and there's 3.3V being supplied to the 3.3V pin, ground is grounded. The serial connection shows no attempt at boot at all- no startup code, no errors, nothing.

When in this state of not booting but with connected power, a successful boot can be trigger by simply grounding the EN pin. It then starts up correctly. 

A theory was that the EN pin may not be high/low at the correct time during boot, however we are using the recommended RC circuit to connect the EN pin to 3.3V. I have also tried holding the EN pin at 3.3V and grounded and repeating the on/off cycle tests and still seen the same problem. 

The ESP is not getting excessively hot, and the occasions it fails to boot appear to be completely random. I also removed the esp_restart() call from the shutdown code so it was shutdown solely by physical disconnect of the power, same problem.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4831421/sdkconfig.txt)

[image]

[firmware_elf.txt](https://github.com/espressif/esp-idf/files/4831496/firmware_elf.txt)

My colleagues and I are out of ideas. Thanks!

### Expected Behavior
Should boot up every time.

### Actual Behavior
Fails to boot approximately one every 250 on/off cycles despite correctly set 3.3V and ground pins.

### Steps to reproduce
N/A

Alvin1Zhang, 2020-06-29:
Thanks for reporting, we will look into.

projectgus, 2020-06-29:
Hi @lr399,

Thanks for the detailed report. This looks like a hardware issue rather than anything with ESP-IDF, but we can try to help you here:

The circuit connected to EN looks is a little different to the peripheral schematic shown in the WROOM module datasheet (section 7), or the circuit used on Espressif dev boards. What is J7 pin 6 connected to when the ESP32 fails to start up?

Angus

lr399, 2020-06-29:
Hi @projectgus, thanks for your help. J7 is either connected to a usb serial cable or left unconnected. Either way, the same problem occurs with the same regularity. J7 pin 6 is always at 0.0V when the power is disconnected and rises to 3.3V when the power is connected. I've also tried using a voltage supply to hold J7 pin 6 at 0.0V and 3.3V continuously without any affect.

I saw online about 'The discharge circuit can be applied in scenarios where ESP32 is powered on and off repeatedly by switching thepower rails, and there is a large capacitor on the VDD33 rail' and wondered if this had any relevance to this issue?

projectgus, 2020-06-30:
Hi @lr399,

The net at J7 pin 6 needs to be connected to ground for the RC delay circuit on the EN pin to function as intended. If it's floating (either because nothing is connected here, or because whatever is at the other end of the "USB serial cable link" isn't driving it to ground) then the capacitor at C6 is effectively removed from the circuit.

You can quickly test if this is the problem by jumpering J7 Pin 6 directly to ground on the board (J7 pin 1, I think?).

Please check  the example circuits for an Espressif development board like the [DevKitC](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/modules-and-boards.html#esp32-devkitc-v4) closely, and also the peripheral schematics section from the WROOM datasheet. All of these documents are available on the Espressif website.

> I saw online about 'The discharge circuit can be applied in scenarios where ESP32 is powered on and off repeatedly by switching thepower rails, and there is a large capacitor on the VDD33 rail' and wondered if this had any relevance to this issue?

This sentence is from the ESP32-WROOM datasheet section 7. It refers to the problem that if the VDD power rail has a lot of capacitance, and external power is removed, then the capacitor may take a long time to discharge back to 0V. For the period of time that the VDD voltage is outside the specified voltage range for ESP32, the chip may not behave correctly.

It's possible this is the root cause of the problem you're seeing. You can verify if you need the discharge circuit by measuring the VDD voltage at the WROOM module immediately after disconnecting power. If it drops immediately to 0V, you probably don't need the discharge circuit.

Suggest also checking that the other aspects of the design match the WROOM datasheet (such as ensuring sufficient capacitance for the VDD power rail).

projectgus, 2020-07-14:
Hopefully you've made some progress with troubleshooting this issue.

As it's a hardware issue rather than something related to ESP-IDF then there is a limit to how much help we can give via this channel.

Espressif does offer a [paid hardware design review service](https://www.espressif.com/en/contact-us/technical-inquiries/hardware-issues), you can contact via the website. Or the [esp32.com](https://esp32.com) forum is a good place to ask technical questions or get input on a design.

One thing I didn't notice in the original post is that the design is powered from a 9V battery. If this is a consumer 9V battery then these have high internal resistance, the ESP32 draws a low of current briefly during initial power up and this may be browning out the battery voltage and causing intermittent glitches. To confirm, use an oscilloscope to watch the various power traces during power-on. You may be able to work around this by increasing capacitance on the 3.3V rail.

Best of luck resolving the issue with the design.

closed 2020-07-14 by projectgus