espressif/esp-idf #5501: Import_prebuilt example not compiling

TerelleM, 2020-06-25:
Hi, I'm new to using the ESP-IDF format and want to get a project up and running on an ESP32. The project requires an extended source and header file and I am aware a CMakeLists is requred in each component folder as well as the main file. I was hoping to use the Import_prebuilt example as a reference for my own code however, it does not compile and throws a ninja error.

**Main CMakeLists:**
idf_component_register(SRCS "main.c"
                    INCLUDE_DIRS ".")

add_prebuilt_library(prebuilt "libprebuilt.a"
             PRIV_REQUIRES spi_flash app_update log)

target_link_libraries(${COMPONENT_LIB} PRIVATE prebuilt)

**Component CMakeLists:**
idf_component_register(SRCS prebuilt.c
                       INCLUDE_DIRS "."
                       PRIV_REQUIRES app_update spi_flash log)

add_custom_command(TARGET ${COMPONENT_LIB}
                  POST_BUILD
                  COMMAND ${CMAKE_COMMAND} -E copy $<TARGET_FILE:${COMPONENT_LIB}> ${CMAKE_SOURCE_DIR}/../main
                  COMMAND ${CMAKE_COMMAND} -E copy ${CMAKE_CURRENT_LIST_DIR}/prebuilt.h ${CMAKE_SOURCE_DIR}/../main
                  COMMENT "Copying built archive file and header to parent example directory...")

**Error Thrown:**
Python requirements from C:\Users\terel\Documents\esp-idf\requirements.txt are satisfied.
ninja: error: '../main/libprebuilt.a', needed by 'import_prebuilt.elf', missing and no known rule to make it
Checking Python dependencies...
Adding flash's dependency "all" to list of actions
Executing action: all (aliases: build)
Running ninja in directory c:\users\terel\eclipse-workspace\import_prebuilt\build
Executing "ninja all"...
ninja failed with exit code 1

projectgus, 2020-06-29:
Hi @TerelleM,

I'm afraid I don't fully understand what your goal is. Do you already have a prebuilt static library that you want to include in your ESP-IDF project, or is there another reason why you picked the "import_prebuilt" example as a base?

The error you're seeing is that same one that the `import_prebuilt` example gives if you don't run the separate build step to produce the example `libprebuilt.a` and copy this into the example project's main directory.  The README for the example gives some more information about this.

The `import_prebuilt` example is designed to demonstrate one specific situation: you have a binary library that is already being built somewhere else, and you need to include this binary library in your ESP-IDF project. Is that a match for your use case?

The more specifics you can give about exactly what your goal is, the more likely we can point you in the right direction.

Angus

PS Please also let us know which ESP-IDF version you're using.

TerelleM, 2020-06-29:
@projectgus ,

No sorry I am trying to include a separate source file with a header to my esp-32 build. I've looked online through allot of resources about adding separate components in addition to the main source file however i can't seem to code the CMakeLists.txt correctly.

projectgus, 2020-06-30:
Hi @TerelleM ,

If you only have a single source file and header then suggest starting simple by adding it your project's "main" component.

1. Download the [idf-template](https://github.com/espressif/esp-idf-template) project and make a copy of it (or use the existing project you have).

2. Copy the source file and header into the `main/` directory

3. Edit the `main/CMakeLists.txt` file - this is the CMakeLists.txt file for the main component.

4. Assuming you're using ESP-IDF v4.x, make this CMakeLists.txt file look something like this:
https://docs.espressif.com/projects/esp-idf/en/v4.0.1/api-guides/build-system.html#minimal-component-cmakelists

Where `SRCS` has a list of the source file names to include in the `main` component (probably `main.c` and the name of your other file), and change the `INCLUDE_SRCS` part to say `INCLUDE_SRCS .` (Because the header is in the same directory, not in an `include` subdirectory).

Once this project can build OK, if you want to split the extra source file and header out from main into its own component then you can do this as well. The docs page linked above has some examples of how to structure additional components as part of your project.

I'm going to close this issue because this is more of a question about how to use ESP-IDF, rather than something we can fix in ESP-IDF. If you need more help then the best place to post questions is on https://esp32.com/ forum .

If posting on the forum about this then please be as specific as you can - include file names, file contents, exact versions or everything, and the full output when it fails to build.

Hope you have everything building smoothly soon.

closed 2020-06-30 by projectgus