espressif/esp-idf #5502: 通过mqtt连接服务器，esp_mqtt_client_config_t mqtt_cfg = {         .host =域名时报错：MQTT_CLIENT: Error transport connect

iot1688, 2020-06-26:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

iot1688, 2020-06-26:
esp_mqtt_client_config_t mqtt_cfg = {
        //.host = hosttemp,//sysCfg.mqtt_host,            //MQTT服务器IP
        .uri = hosttemp,//sysCfg.mqtt_host,  用uri的方式，改成mqtt://mqtt.eclipse.org类似这个，还是报同样的错误。**
        .event_handle = mqtt_event_handler, //MQTT事件
        .port=sysCfg.mqtt_port,

iot1688, 2020-06-30:
有人帮忙看下吗？
是不是我们menuconfig配置哪个参数漏了？

david-cermak, 2020-06-30:
Hi @iot1688 

Could you please try to apply this patch to IDF and check if this fixes the issue?
[0001-optimization-DNS-get-function.patch.txt](https://github.com/espressif/esp-idf/files/4852586/0001-optimization-DNS-get-function.patch.txt)

There's been a recent DNS related bugfix, which hasn't made its way to `v4.0` yet.
Also without this patch the DNS lookup might fail just intermittently, not constantly.Do you see this issue every time you connect or does it work just sometimes?

iot1688, 2020-07-01:
@david-cermak 
Thanks, it appears all the time, not intermittently. In addition, in the official example of MQTT TCP, the domain name of the proxy mqtt server is normally resolved, and now the problem is in the Bluetooth engineering example.

iot1688, 2020-07-01:
static int resolve_dns(const char *host, struct sockaddr_in *ip) 
{
    const struct addrinfo hints = {
        .ai_family = AF_INET,
        .ai_socktype = SOCK_STREAM,
    };
    struct addrinfo *res;

    int err = getaddrinfo(host, NULL, &hints, &res);
    if(err != 0 || res == NULL) {
        ESP_LOGE(TAG, "DNS lookup failed err=%d res=%p", err, res);
         return ESP_FAIL;
     }
     ip->sin_family = AF_INET;
    memcpy(&ip->sin_addr, &((struct sockaddr_in *)(res->ai_addr))->sin_addr, sizeof(ip->sin_addr));
    freeaddrinfo(res);
     return ESP_OK;
 }
打了补丁后，还是一样无法域名解析出代理MQTT服务器域名：
 CT_MQTT: Other event id:7[0m
[0;31mE (1973) TRANS_TCP: DNS lookup failed err=202 res=0x0[0m  这个是IPV6的处理
[0;31mE (1983) MQTT_CLIENT: Error transport connect test.mosquitto.org:1883[0m
[0;32mI (1983) CT_MQTT: MQTT_EVENT_ERROR[0m
[0;32mI (1993) CT_MQTT: MQTT_EVENT_DISCONNECTED[0m

david-cermak, 2020-07-01:
@iot1688 
Thanks for testing the patch and sharing the results. Also when you mentioned that you're getting a constant DNS failure, it means that the problem must be somewhere else.

Could you please share some part of your application, where the issue could be reproduced? I've checked the standard `examples/bluetooth/bluedroid/ble/gatt_server/` example and the hostname resolution works okay for me with the same broker:
```
D (10111) event: running post MQTT_EVENTS:7 with handler 0x400d7744 on loop 0x3ffde3ec
0x400d7744: mqtt_event_handler at /Users/david/dev7/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server/build/../main/gatts_demo.c:807

E (10111) GATTS_DEMO: Event dispatched from event loop base=MQTT_EVENTS, event_id=7
I (10121) GATTS_DEMO: Other event id:7
D (10141) TRANS_TCP: [sock=54],connecting to server IP:5.196.95.208,Port:1883...
D (10231) MQTT_CLIENT: Transport connected to mqtt://test.mosquitto.org:1883
```

iot1688, 2020-07-01:
@david-cermak 
thanks for your help!

code such as:

#define CONFIG_BROKER_URL "mqtt://test.mosquitto.org"
void ct_mqtt_init(char *clientid, char *username, char *password)
{
    ESP_LOGI(GATTS_TAG, "%s","mqtt init\n");

   //ESP_LOGI(GATTS_TAG, "%s\n",hosttemp);
    esp_mqtt_client_config_t mqtt_cfg = {
        //.host = "5.196.95.208",//sysCfg.mqtt_host,            //MQTT服务器IP
        .uri = CONFIG_BROKER_URL,//"mqtt://test.mosquitto.org",//hosttemp,//sysCfg.mqtt_host,
        //.event_handle = mqtt_event_handler, //MQTT事件
        // .port=sysCfg.mqtt_port,                         //端口
        // //.username = username,//sysCfg.mqtt_user,                //用户名
        // //.password = password,//sysCfg.mqtt_pass,               //密码
        // .client_id= clientid,//sysCfg.device_id,
        // .keepalive=sysCfg.mqtt_keepalive,
        // .lwt_qos=sysCfg.mqtt_qos,
        // .lwt_retain=sysCfg.mqtt_retain,
        // //.disable_clean_session=true,
        // //.disable_auto_reconnect=true,
        // // .user_context = (void *)your_context
    };

 

    mqtt_client = esp_mqtt_client_init(&mqtt_cfg);
    esp_mqtt_client_register_event(mqtt_client, ESP_EVENT_ANY_ID, mqtt_event_handler, mqtt_client);
    ESP_ERROR_CHECK(esp_mqtt_client_start(mqtt_client));
}
上面的是连接mqtt域名的函数
下面的是连接第三路由器，连接成功后调用上边的函数连接MQTT代理服务器
void wifi_init_softap_sta(bool bAPSta)
{
    s_wifi_event_group = xEventGroupCreate();//sta

    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    //STA模式需要的
    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &event_handler, NULL));

    // wifi_config_t sta_config = {
    //     .sta = {
    //         .ssid = EXAMPLE_ESP_WIFI_SSID,
    //         .password = EXAMPLE_ESP_WIFI_PASS
    //     },
    // };
    wifi_config_t sta_config;
    memset(&sta_config,0,sizeof(wifi_config_t));
    //memcpy(sta_config.sta.ssid,"JAY",3);
    //memcpy(sta_config.sta.password,"20110812",8);
    memcpy(sta_config.sta.ssid,sysCfg.sta_ssid,strlen(sysCfg.sta_ssid));
    memcpy(sta_config.sta.password,sysCfg.sta_pwd,strlen(sysCfg.sta_pwd));
    //end

    if(bAPSta)
    {      
        //带SOFTAP模式 
        ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL));
        // wifi_config_t ap_config = {
        //     .ap = {
        //         .ssid = EXAMPLE_ESP_WIFI_SSID,
[... 80 lines cut ...]
    {
        mConnectAPStatus = true;
        ESP_LOGI(GATTS_TAG, "connected to ap SSID:%s password:%s",sysCfg.sta_ssid, sysCfg.sta_pwd);

       ct_mqtt_init(sysCfg.device_id, sysCfg.mqtt_user, sysCfg.mqtt_pass);
              
    } 
    else if (bits & WIFI_FAIL_BIT) 
    {
        mConnectAPStatus = false;
        ESP_LOGI(GATTS_TAG, "Failed to connect to SSID:%s, password:%s",
                 sysCfg.sta_ssid, sysCfg.sta_pwd);

    }
    else 
    {
        mConnectAPStatus = false;
        ESP_LOGE(GATTS_TAG, "UNEXPECTED EVENT");
    }

    ESP_ERROR_CHECK(esp_event_handler_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, &event_handler));
    ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler));
    vEventGroupDelete(s_wifi_event_group);
    
}

iot1688, 2020-07-02:
@david-cermak 
你好，有没有帮我们看下，这个问题，对我们来说真的是不知道哪个地方漏了什么还是，折腾一周多了，比较烦恼。
ESP32感觉上没有ESP8266那么好开发。

iot1688, 2020-07-02:
发现下面的代码不行，但是这个在ESP8266平台上可以用：
//设置静态IP地址，netmask,网关之类的
    tcpip_adapter_ip_info_t ipInfo;
    
    //STA下的静态IP等
    ipInfo.ip.addr=ipaddr_addr("192.168.0.200");
    ipInfo.netmask.addr=ipaddr_addr("255.255.255.0");
    ipInfo.gw.addr=ipaddr_addr("192.168.0.1");
    ESP_ERROR_CHECK(tcpip_adapter_dhcpc_stop(TCPIP_ADAPTER_IF_STA));
    ESP_ERROR_CHECK(tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_STA,&ipInfo));

这是为何呢，网关等等都配置对了

david-cermak, 2020-07-02:
Hi @iot1688 

Thanks for sharing the code. I was able to reproduce your issue locally. Indeed, it is related to setting the static IP as you've outlined in your last post. This code however works perfectly and sets the static IP including the gateway. What it doesn't configure, though, is the dns server information and that's why the DNS lookup fails.
You have to set DNS server, as well, similar to the following:
``` c
tcpip_adapter_dns_info_t dns_info;
dns_info.ip.u_addr.ip4.addr=ipaddr_addr("8.8.8.8");
dns_info.ip.type = IPADDR_TYPE_V4;
tcpip_adapter_set_dns_info(TCPIP_ADAPTER_IF_STA, TCPIP_ADAPTER_DNS_MAIN, &dns_info);
```
Configure the DNS server IP according to your preferences, possibly this might equals to your router IP `192.168.0.1`.
Please note that with DHCP client on, this gets setup automatically. I'm not sure why the same code works in 8266 codebase, my assumption is that it just uses older lwIP, where the DNS is set automatically to some value, you can check the actual settings by calling the `tcpip_adapter_get_dns_info()` to find out the reason.

iot1688, 2020-07-03:
@david-cermak 
TKS, it's ok.

closed 2020-07-03 by iot1688