espressif/esp-idf #5503: blink example crashes on ESP32 U4wdh

syu-source, 2020-06-26:
blink example crashes on ESP32 U4wdh

## Environment
Custom board. [Moving from esp32d0wd with 4MB external flash to ESP32-U4WD]
Build with VSCODE + ESP-IDF Visual Studio Code Extension v0.3.0

- Development Kit:      [none]
- Kit version : N/A
- Module or chip used: ESP32-U4WD
- IDF version (run ``git describe --tags`` to find it):
    v4.1-dev-2182-gdc14d027c
- Build System:        CMAKE
- Compiler version:
xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [Windows|Linux|macOS]
- MSYS2 mingw32
- Using an IDE: vscode
- Power Supply:         USB, 3.3v.

## Problem Description
failed to start blink example on esp32-u4wdh.
Panic after 
```
I (274) cpu_start: Pro cpu start user code
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
```
Same code and binary works on esp32-d0wd with external 4MB flash.

Flashing and custom code in bootloader works fine, I assume that PCB [hardware wring] works fine.
I've googled and checked with reference/datasheet pdfs, no luck so far.

sdkconfig modification:
 `CONFIG_FREERTOS_UNICORE=y`

### Expected Behavior
It is supposed to work.

### Actual Behavior
Loop
Panic at start_cpu0_default() and reboot.

### Steps to reproduce
VSCODE with idf extention
1. Press F1, ESP-IDF: show Example**s** Projects. <--- TYPO? 
2. Select BLINK and click "Create project using example blink"
3. Change sdkconfig: `CONFIG_FREERTOS_UNICORE=y`
4. Build
5. Flash and Monitor

### Code to reproduce this issue
Same as example code, not changed.
Though I haved tried to build custom bootloader, add debug print and GPIO output, works as expected.

## Debug Logs

```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7204
load:0x40078000,len:14220
load:0x40080400,len:4584
0x40080400: _init at ??:?

entry 0x400806f8
I (30) boot: ESP-IDF v4.1-dev-2182-gdc14d027c 2nd stage bootloader
I (30) boot: compile time 13:14:02
I (30) boot: chip revision: 3
I (34) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 3, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x04cd8 ( 19672) map
I (121) esp_image: segment 1: paddr=0x00014d00 vaddr=0x3ffb0000 size=0x01960 (  6496) load
I (125) esp_image: segment 2: paddr=0x00016668 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at D:/workspace/vscode/esp-idf.v4/components/freertos/xtensa/xtensa_vectors.S:1789

I (132) esp_image: segment 3: paddr=0x00016a74 vaddr=0x40080404 size=0x08948 ( 35144) load
I (156) esp_image: segment 4: paddr=0x0001f3c4 vaddr=0x00000000 size=0x00c54 (  3156) 
I (157) esp_image: segment 5: paddr=0x00020020 vaddr=0x400d0020 size=0x127bc ( 75708) map
0x400d0020: _stext at ??:?

I (195) boot: Loaded app from partition at offset 0x10000
I (195) boot: Disabling RNG early entropy source...
I (195) cpu_start: Pro cpu up.
I (199) cpu_start: Application information:
I (204) cpu_start: Project name:     blink
I (209) cpu_start: App version:      1
I (213) cpu_start: Compile time:     Jun 26 2020 13:15:39
I (219) cpu_start: ELF file SHA256:  72a158c022c47e58...
I (225) cpu_start: ESP-IDF:          v4.1-dev-2182-gdc14d027c
I (231) cpu_start: Single core mode
I (236) heap_init: Initializing. RAM available for dynamic allocation:
I (243) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (249) heap_init: At 3FFB2050 len 0002DFB0 (183 KiB): DRAM
I (255) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (262) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (268) heap_init: At 40088D4C len 000172B4 (92 KiB): IRAM
I (274) cpu_start: Pro cpu start user code
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000be32  PS      : 0x00060130  A0      : 0x80001e11  A1      : 0x3ffe3af0
A2      : 0x3ffae0ac  A3      : 0x00001800  A4      : 0x3ffae6fc  A5      : 0x00000000  
A6      : 0x3ffae7bc  A7      : 0x3ffae7c8  A8      : 0x00000000  A9      : 0x3ffe3a30  
A10     : 0x3ffae0ac  A11     : 0x00001800  A12     : 0x3ffae6fc  A13     : 0x00000000
A14     : 0x0000001f  A15     : 0x00000001  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000074  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

ELF file SHA256: 72a158c022c47e58e61104155ec8625f12ea2e594fd4ba7faa22ace14dcfb724

Backtrace: 0x4000be2f:0x3ffe3af0 0x40001e0e:0x3ffe3b10 0x40001e93:0x3ffe3b30 0x400d51e2:0x3ffe3b50 0x400d5298:0x3ffe3b80 0x40080dd9:0x3ffe3ba0 0x40080ffb:0x3ffe3bf0 0x4007945b:0x3ffe3c30 0x40079515:0x3ffe3c60 0x40079533:0x3ffe3ca0 0x4007982d:0x3ffe3cc0 0x4008072a:0x3ffe3df0 0x40007c15:0x3ffe3eb0 0x4000073d:0x3ffe3f20
0x400d51e2: _fopen_r at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fopen.c:126

0x400d5298: fopen at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fopen.c:168

0x40080dd9: start_cpu0_default at D:/workspace/vscode/esp-idf.v4/components/esp32/cpu_start.c:363

0x40080ffb: call_start_cpu0 at D:/workspace/vscode/esp-idf.v4/components/esp32/cpu_start.c:262 (discriminator 1)

Rebooting...
ets Jul 29 2019 12:21:46
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

syu-source, 2020-06-26:
After setting `CONFIG_ESP_CONSOLE_UART_NONE` 
backtrace changed.

### Crash code from cpu_start.c
```
#ifndef CONFIG_ESP_CONSOLE_UART_NONE
    const char* default_uart_dev = "/dev/uart/" STRINGIFY(CONFIG_ESP_CONSOLE_UART_NUM);
    _GLOBAL_REENT->_stdin  = fopen(default_uart_dev, "r");
    _GLOBAL_REENT->_stdout = fopen(default_uart_dev, "w");
    _GLOBAL_REENT->_stderr = fopen(default_uart_dev, "w");
#else
    _GLOBAL_REENT->_stdin  = (FILE*) &__sf_fake_stdin;
    _GLOBAL_REENT->_stdout = (FILE*) &__sf_fake_stdout;
    _GLOBAL_REENT->_stderr = (FILE*) &__sf_fake_stderr;
#endif
```

### Monitor log:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7204
load:0x40078000,len:14220
load:0x40080400,len:4584
0x40080400: _init at ??:?

entry 0x400806f8
I (30) boot: ESP-IDF v4.1-dev-2182-gdc14d027c 2nd stage bootloader
I (30) boot: compile time 13:14:02
I (30) boot: chip revision: 3
I (34) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 3, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x04cc8 ( 19656) map
I (121) esp_image: segment 1: paddr=0x00014cf0 vaddr=0x3ffb0000 size=0x01960 (  6496) load
I (125) esp_image: segment 2: paddr=0x00016658 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at D:/workspace/vscode/esp-idf.v4/components/freertos/xtensa/xtensa_vectors.S:1789

I (132) esp_image: segment 3: paddr=0x00016a64 vaddr=0x40080404 size=0x08934 ( 35124) load
I (156) esp_image: segment 4: paddr=0x0001f3a0 vaddr=0x00000000 size=0x00c78 (  3192) 
I (157) esp_image: segment 5: paddr=0x00020020 vaddr=0x400d0020 size=0x12294 ( 74388) map
0x400d0020: _stext at ??:?

I (194) boot: Loaded app from partition at offset 0x10000
I (194) boot: Disabling RNG early entropy source...
I (195) cpu_start: Pro cpu up.
I (198) cpu_start: Application information:
I (203) cpu_start: Project name:     blink
I (208) cpu_start: App version:      1
I (213) cpu_start: Compile time:     Jun 26 2020 14:05:10
I (219) cpu_start: ELF file SHA256:  f91d6083d165e91b...
I (225) cpu_start: ESP-IDF:          v4.1-dev-2182-gdc14d027c-dirty
I (232) cpu_start: Single core mode
I (236) heap_init: Initializing. RAM available for dynamic allocation:
I (243) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (249) heap_init: At 3FFB2050 len 0002DFB0 (183 KiB): DRAM
I (255) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (262) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (268) heap_init: At 40088D38 len 000172C8 (92 KiB): IRAM
I (274) cpu_start: Pro cpu start user code
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000be94  PS      : 0x00060930  A0      : 0x80058cce  A1      : 0x3ffe3730
A2      : 0x3ffaf964  A3      : 0x00060923  A4      : 0x00060920  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0xfffffff6  A8      : 0x00000000  A9      : 0x3ffe3710  
A10     : 0x3ffaf904  A11     : 0x00060b23  A12     : 0x00000000  A13     : 0x3ffe3bc8
A14     : 0x00000000  A15     : 0x01000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: f91d6083d165e91bff2f31a517f6d958a3d1a6c9709b8452aa68071247bc47bc

Backtrace: 0x4000be91:0x3ffe3730 0x40058ccb:0x3ffe3750 0x400db3d1:0x3ffe3770 0x400de3a5:0x3ffe3a80 0x40085299:0x3ffe3ab0 0x400826af:0x3ffe3b10 0x400827ad:0x3ffe3b30 0x400d3755:0x3ffe3b60 0x40080e50:0x3ffe3ba0 0x40080fe7:0x3ffe3bf0 0x4007945b:0x3ffe3c30 0x40079515:0x3ffe3c60 0x40079533:0x3ffe3ca0 0x4007982d:0x3ffe3cc0 0x4008072a:0x3ffe3df0 0x40007c15:0x3ffe3eb0 0x4000073d:0x3ffe3f20
0x400db3d1: _vfprintf_r at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:858 (discriminator 3)

0x400de3a5: vprintf at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34

0x40085299: esp_log_write at D:/workspace/vscode/esp-idf.v4/components/log/log.c:190

0x400826af: detect_spi_flash_chip at D:/workspace/vscode/esp-idf.v4/components/spi_flash/esp_flash_api.c:220 (discriminator 9)

0x400827ad: esp_flash_init at D:/workspace/vscode/esp-idf.v4/components/spi_flash/esp_flash_api.c:133

0x400d3755: esp_flash_init_default_chip at D:/workspace/vscode/esp-idf.v4/components/spi_flash/esp_flash_spi_init.c:207

0x40080e50: start_cpu0_default at D:/workspace/vscode/esp-idf.v4/components/esp32/cpu_start.c:429

0x40080fe7: call_start_cpu0 at D:/workspace/vscode/esp-idf.v4/components/esp32/cpu_start.c:262 (discriminator 1)
```

igrr, 2020-06-26:
Hi @syu-source, please try applying the following fix to your copy of IDF: https://github.com/espressif/esp-idf/commit/aac3e56a9a2644ec8a5810a1d55cff43b71e728e

We will backport the fix to release/v4.1 branch.

syu-source, 2020-06-26:
After this patch, still the same error
@igrr I checked soc/soc_caps.h where SOC_CPU_CORES_NUM  is defined, and it is 2.
[image]

Should I change it to 1 or there is a config somewhere?

igrr, 2020-06-26:
No, please keep SOC_CPU_CORES_NUM == 2 — the goal of the linked fix is to
initialize syscall_table_ptr_app even when running in single core mode.

> After this patch, still the same error
>
Just to be sure that this is not related to something else: could you try
building and running the same example using the latest master branch, or
using the latest release/v4.1 branch?

syu-source, 2020-06-26:
@igrr The root cause is   **aac3e56**, and it is not included in my commit. 

Thanks a lot for quick response, you saved my day.  It was really frustrating.

syu-source, 2020-06-26:
> No, please keep SOC_CPU_CORES_NUM == 2 — the goal of the linked fix is to initialize syscall_table_ptr_app even when running in single core mode.
> After this patch, still the same error
> Just to be [...]

Errr... Just curious, why? I mean the fix could simple delete #IF , which makes it easier to understand. 
And why are there two syscall_tables in the first place, since both with the same value regardless actual number of CPU CORES.

This bug suffered me badly... I would like to know little more.

igrr, 2020-06-26:
The fix is written like this to also work on ESP32-S2, which only has one syscall table. The reason for having two on ESP32 was to allow for potential non-SMP use cases, where one core would be using an RTOS-based set of syscalls, and the other core would use a non-OS set of syscalls.
In ESP32 revision 3, a bug in the ROM build process has resulted in both cores using APP CPU syscall table pointer. As written, the fix makes IDF compatible with all ESP32 revisions, and also with ESP32-S2.

closed 2020-06-26 by igrr

pedrominatel, 2020-07-05:
> Hi @syu-source, please try applying the following fix to your copy of IDF: [aac3e56](https://github.com/espressif/esp-idf/commit/aac3e56a9a2644ec8a5810a1d55cff43b71e728e)
> 
> We will backport the fix to release/v4.1 branch.
> 
> Correction: it seems your commit is ahead of release/v4.1 branch. Actually the fix is not needed on release/v4.1, and is already included in master and release/v4.2.

Thanks!

I had the same issue on my custom board using the U4 version.

Regards,
Pedro