espressif/esp-idf #5504: could not use esp_https_ota with certificate bundle

JakubRakus, 2020-06-26:
## Environment

- Module or chip used: ESP32-WROOM-32D
- IDF version v4.2-dev-1126-gd85d3d969
- Build System:         idf.py
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     Linux

## Problem Description

There is no way to use crt bundle (enabled via menuconfig) with `esp_https_ota`.

If I try to attach the bundle with `esp_crt_bundle_attach` and then set `ota config.use_global_ca_store = true` the ota process stops with error `Server certificate not found in esp_http_client config` because function `esp_https_ota_begin` wants `config.cert_pem`. Trying to set `config.cert_pem` to some dummy cert but it also does not work - `esp_https_ota_begin` couldn't use certs from global store.

Also I've tried to pass `x509_crt_imported_bundle_bin_start` to `config.cert_pem` but it lloks like cert bundle format is not valid pem. Function `esp_crt_bundle_init` loads certs from bin to `static struct crt_bundle_t` and makes some magic with cert headers and numbers.

### Expected Behavior

 `esp_https_ota_begin` should check if `use_global_ca_store` is set to true and then try to use certificates from previously attached crt bundle

and/or

there should be a function in `esp_crt_bunlde` that gives a pointer to certs in pem file format which can be passed to `ota config.cert_pem`

ESP-Marius, 2020-06-29:
Hi, thanks for reporting this.

Yes, right now it's not possible. I'll look into it and push an update.

JakubRakus, 2020-08-24:
Any progress on this?

ESP-Marius, 2020-08-26:
@JakubRakus Sorry, for the slow turn-around. It's still on my to-do list. I'll try to prioritize it.

closed 2021-04-15 by espressif-bot