espressif/esp-idf #5506: Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Code Worked Earlier.

tejakolla, 2020-06-26:
I am seeing the standard "Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled." error.

The same code was worked earlier, Now I'm getting this error. I tried uploading to multiple different ESP32 modules but all of them have the same problem.

The crash occurs while using the  "rmt_get_ringbuf_handle" in the IR receiving part.

Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
.
.
.

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4008b979  PS      : 0x00060e30  A0      : 0x8008bd65  A1      : 0x3ffd0950
0x4008b979: prvReceiveGeneric at ......../esp-idf/components/esp_ringbuf/ringbuf.c:969

A2      : 0x00000000  A3      : 0x3ffd09a0  A4      : 0x00000014  A5      : 0x3ffd09a4
A6      : 0x00000014  A7      : 0x000006dc  A8      : 0x8008b968  A9      : 0x3ffd0930
A10     : 0x000006c8  A11     : 0x00000000  A12     : 0x00000014  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000003c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff9

Backtrace: 0x4008b979:0x3ffd0950 0x4008bd62:0x3ffd0990 0x400d53a9:0x3ffd09d0 0x400d40a8:0x3ffd0a20 0x40130133:0x3ffd0a40 0x4012f365:0x3ffd0a80 0x4012f404:0x3ffd0b10 0x4012f7d4:0x3ffd0b30 0x4012e8b2:0x3ffd0b50 0x4012e924:0x3ffd0b90 0x4008c145:0x3ffd0bb0

wilhelmw201, 2020-06-29:
Usually this is because the program was trying to dereference a zero pointer. Did you initialize the driver correctly? (both rmt_config and rmt_driver_install have to be called)

tejakolla, 2020-06-29:
@wilhelmw201  thank you for your response. I initialized it correctly. After few changes done in code, now I'm getting below **rmt_rx_stop
E (51804) rmt: rmt_driver_uninstall(673): RMT CHANNEL ERR
Guru Meditation Error: Core  0 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d49e4  PS      : 0x00060d30  A0      : 0x800d4c10  A1      : 0x3ffce700  
0x400d49e4: ir_bin at /Users/Loki/Documents/esp_pro/esp-homekit-sdk2/accessories/aws-iot1/main/rx.c:91

A2      : 0x00000200  A3      : 0x00000000  A4      : 0x3ffb80a4  A5      : 0x00000000  
A6      : 0x00000c5c  A7      : 0x3ffce710  A8      : 0x00000004  A9      : 0x00000001  
A10     : 0x3ffce700  A11     : 0x00000001  A12     : 0x00000003  A13     : 0x00000048  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x400d49e4:0x3ffce700 0x400d4c0d:0x3ffce740 0x400d3eec:0x3ffce790 0x4012c6eb:0x3ffce7b0 0x4012b91d:0x3ffce7f0 0x4012b9bc:0x3ffce880 0x4012bd8c:0x3ffce8a0 0x4012ae6a:0x3ffce8c0 0x4012aedc:0x3ffce900 0x4008c145:0x3ffce920**.        Is it related to zero pointer ??

costaud, 2020-06-30:
Would you please share some code to reproduce this issue ? @tejakolla

tejakolla, 2020-06-30:
@costaud 

**#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "freertos/semphr.h"
#include "esp_err.h"
#include "esp_log.h"
#include "driver/rmt.h"
#include "driver/periph_ctrl.h"
#include "soc/rmt_reg.h"
#include "app_config.h"

extern ReceivedData receivedData;

struct rmt_rx_inits {
	rmt_channel_t  channel;
	gpio_num_t     gpio_num;
	uint8_t        clk_div;
	uint8_t        mem_block_num;
	bool           config_filter_en;
	uint8_t        config_filter_tics_thresh;
	uint16_t       config_idle_threshold;
	char           tag[10];
}rx_inputs[1] = {
	{ 1, 16, 80, 2, false, 0, 50000, "Ch0" },
	// { 1, 14, 80, 1, false, 0, 50000, "Ch1" }
};

void rx_init() {
    rmt_config_t rmt_rx;
    size_t i;
    size_t len = sizeof(rx_inputs) / sizeof( rx_inputs[0] );
    for ( i=0; i<len; i++ ) {
		rmt_rx.channel       = rx_inputs[i].channel;
		rmt_rx.gpio_num      = rx_inputs[i].gpio_num;
		rmt_rx.clk_div       = rx_inputs[i].clk_div;
		rmt_rx.mem_block_num = rx_inputs[i].mem_block_num;
		rmt_rx.rmt_mode      = RMT_MODE_RX;
		rmt_rx.rx_config.filter_en           = rx_inputs[i].config_filter_en;
		rmt_rx.rx_config.filter_ticks_thresh = rx_inputs[i].config_filter_tics_thresh;
		rmt_rx.rx_config.idle_threshold      = rx_inputs[i].config_idle_threshold;
		rmt_config(&rmt_rx);
		rmt_driver_install(rx_inputs[i].channel, 1000, 0);
	}
}

int dur( uint32_t level, uint32_t duration ) {
	if ( level == 0 ) { return duration; }
	else { return -1.0 * duration; }
}

size_t num_channels = sizeof(rx_inputs) / sizeof( rx_inputs[0] );
size_t c, i;
size_t rx_size = 0;
rmt_item32_t* items = NULL;
RingbufHandle_t rb;

bool rx_break;
int rx_loop_counter = 0;
int rx_msg[];
int rx_start() {       // Main call happens here
    rx_break = false;
    int start1;
    for ( c=0; c<num_channels; c++ ) {
		rmt_rx_start(rx_inputs[c].channel, 1);
	}
    while (1) {
        rx_loop_counter++;
        if(rx_loop_counter > 2000){
            break;
        }else{
            for ( c=0; c<num_channels; c++ ) {
                rmt_get_ringbuf_handle(rx_inputs[c].channel, &rb);
                items = (rmt_item32_t*) xRingbufferReceive(rb, &rx_size, 20);
                if(items) {
                    int dummy = rx_size/4;
                    if(dummy%2 == 0){
                        rx_break = true;
                        start1 = dur( items[0].level0, items[0].duration0);
                        for ( i=0; i<rx_size/4; i++ ) {
                            rx_msg[2*i] = dur( items[i].level0, items[i].duration0 );
                            rx_msg[2*i+1] = dur( items[i].level1, items[i].duration1 );
                            if(-1*rx_msg[2*i+1] > start1){
                                rx_msg[2*i+1] = 0;
                                break;
                            }
                        }
                        rx_msg[2*i+2] = "\0";
                    }else{
                    }
                    vRingbufferReturnItem(rb, (void*) items);
                }
            }
        }
        if(rx_break){
            rx_loop_counter = 0;
            break;
        }
        vTaskDelay( 100 / portTICK_PERIOD_MS );
	}
    for ( c=0; c<num_channels; c++ ) {
        rmt_rx_stop(rx_inputs[c].channel);
       rmt_driver_uninstall(rx_inputs[c].channel);
        rx_loop_counter = 0;
    }
    if(rx_break){
        rx_break = false;
        return ir_bin(rx_msg,i-1);
    }else{
        return -1;
    }
}**

Along  with this code I'm running three tasks. when rx_start() happens, stopping the code execution part inside the tasks.

closed 2025-02-24 by suda-morris