espressif/esp-idf #5507: BLE_Mesh_Provisioner_Example_Walkthrough.md contains a reference to a missing document

Durobot, 2020-06-26:
[esp-idf/examples/bluetooth/esp_ble_mesh/ble_mesh_provisioner/tutorial/BLE_Mesh_Provisioner_Example_Walkthrough.md](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_ble_mesh/ble_mesh_provisioner/tutorial/BLE_Mesh_Provisioner_Example_Walkthrough.md) is referencing a document that is not there. In "Introduction":

This demo shows how a BLE Mesh device can function as a Provisioner. If you are new to BLE Mesh, please start by checking [Demo for ESP BLE Mesh Node](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node/README.md).

This folder, `https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node/`, does not  contain README.md. Instead it contains two projects' folders, **onoff_client** and **onoff_server**, each with its own (very short) README.md file.

Alvin1Zhang, 2020-06-29:
@Durobot Thanks for pointing out, we will look into.

Alvin1Zhang, 2020-07-10:
Thanks for reporting and sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/9e23affd65d28f32b6c6c6364fbd803ddc563063. Thanks.

closed 2020-07-10 by Alvin1Zhang