espressif/esp-idf #5509: HTTP & MQTT components donot work when Wifi Ethernet and PPPOS clients are initialized (v4.1 beta 2)

joeyExec, 2020-06-28:
## Environment

-Module or chip used: ESP32-PICO-D4
-IDF version : v4.1 beta 2-cd4fa46f5372a552a67452313d246d4d09d2b39f
-Build System: [idf.py]
-Compiler version (run xtensa-esp32-elf-gcc --version to find it):
-xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
-Operating System: [Windows]
-(Windows only) environment type: [Plain Command Prompt].
-Using an IDE?: [No]

## Problem Description
HTTP & MQTT components do not work when Wifi Ethernet and PPPOS clients are initialized.

//Detailed problem description goes here.

My system needs to initialize Wifi, Ethernet and LTE modules. When all of the modules initialized and have gotten their respective IP adresses MQTT & HTTP components always fail with a transport layer error. Once I deinitialize other netifs and chose to initialize only module, HTTP and MQTT components start working again.

### Expected Behavior
MQTT and HTTP components to work with a connected underlying netif inetrface.

### Actual Behavior
MQTT and HTTP components stop working once i initialize second interface after wifi interface.

### Steps to reproduce

1. Initialize Wifi and get IP
2. Initialize Ethernet and get IP
3. Start MQTT client and it would never connect to the broker.

Debug Logs
Error logs from MQTT component

E (904821) TRANS_TCP: DNS lookup failed err=202 res=0x0
E (904821) MQTT_CLIENT: Error transport connect

Error logs from HTTP component

E (899821) TRANS_TCP: DNS lookup failed err=202 res=0x0
E (899822) HTTP_CLIENT: Connection failed, sock < 0
E (899822) wifi station: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT

## Other items if possible
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4841465/sdkconfig.txt)

Alvin1Zhang, 2020-06-29:
Thanks for reporting, we will look into.

david-cermak, 2020-06-30:
Hi @joeyExec 

Could you please apply this patch to the IDF version you use and repeat your test?
[0001-esp-netif-ppp-netif-to-use-netif_is_link_up-to-updat.patch.txt](https://github.com/espressif/esp-idf/files/4852137/0001-esp-netif-ppp-netif-to-use-netif_is_link_up-to-updat.patch.txt)

There was an issue with interface routing, which was fixed in acdadeb, but this hasn't been backported to 4.1 yet (seen it's not included in your version).

Alvin1Zhang, 2020-08-14:
@joeyExec Thanks for reporting, have you tried the patch provided by @david-cermak ? Would you please help share if any updates? Thanks.

closed 2021-01-06 by Alvin1Zhang