espressif/esp-idf #5510: esp-idf-tools-setup-2.3.exe fails to install 4.0

Mair, 2020-06-28:
using the installation tool is failing when selecting version 4.0. It does work with version 4.1
Issue is related to cloning sub-modules. Confirmed by 2 different users

log
```
...
                          Submodule path 'components/coap/libcoap': checked out 'cfec0d072c5b99ed3e54828ca50ea2f6b91e1f50
                          Submodule 'ext/tinydtls' (https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls) registered for path 'components/coap/libcoap/ext/tinydtls
                          Cloning into 'C:/esp32/esp-idf/components/coap/libcoap/ext/tinydtls'..
                          fatal: unable to update url base from redirection
                            asked for: https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls/info/refs?service=git-upload-pac
                             redirect: https://status.eclipse.org
                          fatal: clone of 'https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls' into submodule path 'C:/esp32/esp-idf/components/coap/libcoap/ext/tinydtls' faile
                          Failed to clone 'ext/tinydtls'. Retry schedule
                          Cloning into 'C:/esp32/esp-idf/components/coap/libcoap/ext/tinydtls'..
                          fatal: unable to update url base from redirection
                            asked for: https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls/info/refs?service=git-upload-pac
                             redirect: https://status.eclipse.org
                          fatal: clone of 'https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls' into submodule path 'C:/esp32/esp-idf/components/coap/libcoap/ext/tinydtls' faile
                          Failed to clone 'ext/tinydtls' a second time, abortin
                          Submodule path 'components/esp_wifi/lib_esp32': checked out 'e8a234dd8ddfd7926f58c9c682cfd906073e9e4c
                          Submodule path 'components/esptool_py/esptool': checked out 'de30f21a222ec62f5a023dd955439b4f57702768
                          Submodule path 'components/expat/expat': checked out 'a7bc26b69768f7fb24f0c7976fae24b157b85b13
                          Submodule path 'components/json/cJSON': checked out '3c8935676a97c7c97bf006db8312875b4f292f6c
                          Submodule path 'components/libsodium/libsodium': checked out '70170c28c844a4786e75efc626e1aeebc93caebc
                          Submodule path 'components/lwip/lwip': checked out '50e4e0d1821a6e5deab0ead9f35ffb4e60625ea6
                          Submodule path 'components/mbedtls/mbedtls': checked out '9ef92c551eb8d92677034c3ec8078a8076febf41
                          Submodule path 'components/mqtt/esp-mqtt': checked out '6bc94add892437d0fd50f26bfabe78c646648c13
                          Submodule path 'components/nghttp/nghttp2': checked out '3bcc416e13cc790e2fb45fcfe9111d38609c5032
                          Submodule 'third-party/mruby' (https://github.com/mruby/mruby) registered for path 'components/nghttp/nghttp2/third-party/mruby
                          Submodule 'third-party/neverbleed' (https://github.com/h2o/neverbleed.git) registered for path 'components/nghttp/nghttp2/third-party/neverbleed
                          Cloning into 'C:/esp32/esp-idf/components/nghttp/nghttp2/third-party/mruby'...
                          remote: Enumerating objects: 762, done.        
                          remote: Counting objects: 100% (762/762), done.        
                          remote: Compressing objects: 100% (164/164), done.        
                          remote: Total 1086 (delta 608), reused 710 (delta 570), pack-reused 324        
                          Receiving objects:  97% (1054/1086), 412.00 KiB | 199.00 KiB/sReceiving objects:  97% Receiving objects: 100% (1086/1086), 521.69 KiB | 214.00 KiB/s, done.
                          Resolving deltas: 100% (694/694), completed with 172 local objects
                          Cloning into 'C:/esp32/esp-idf/components/nghttp/nghttp2/third-party/neverbleed'..
                          Submodule path 'components/nghttp/nghttp2/third-party/mruby': checked out '22464fe5a0a10f2b077eaba109ce1e912e4a77de
                          Submodule path 'components/nghttp/nghttp2/third-party/neverbleed': checked out 'da5c2ab419a3bb8a4cc6c37a6c7f3e4bd4b41134
                          Submodule path 'components/protobuf-c/protobuf-c': checked out 'dac1a65feac4ad72f612aab99f487056fbcf5c1a
                          Submodule path 'components/spiffs/spiffs': checked out 'f5e26c4e933189593a71c6b82cda381a7b21e41c
                          Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25
                          Submodule path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2': checked out '7e8e249990ec491ec15990cf95b6d871a66cf64a
                          Failed to recurse into submodule path 'components/coap/libcoap
```

igrr, 2020-06-28:
It looks like this was related to a scheduled upgrade of Eclipse Gerrit server, see https://bugs.eclipse.org/bugs/show_bug.cgi?id=541221#c137. It is back up now.

closed 2020-06-28 by igrr

elasticdotventures, 2020-07-22:
https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls
is "not found" for me in Melbourne, Australia.