espressif/esp-idf #5511: 帮忙解答一下使用PSRAM一些问题点？

guo652917087, 2020-06-28:
你好，
      我在使用ESP32-D0WD-V3和 master最新分支测试psram，遇到一些疑惑，能帮忙回答下吗
      1. 我在使用tft屏幕测试，有些图片本来存储在spiffs，可是经常出现“Cache disabled but cached memory region accessed”，后来想用EXT_RAM_ATTR 申明图片内容放在片外ram，索性不使用flash，但是图片内容在启动后都变0值了，这个有办法像DRAM存储静态数据吗（启动后非零）？
       2. 第一次烧写启动时，如果连接路由器的wifi耗时较久时，也经常出现“Cache disabled but cached memory region accessed”，我猜测是： wifi初始化 rf 参数时，写入nvs数据，同时又在使用PSRAM，引起这个报错。
       3. 如果需要操作 读写 flash，又在使用 PSRAM，这个怎么操作才正确？是需要应用层加互斥锁吗？还是说你们的SDK底层spi片选操作已经有互斥锁的保护，我只需要考虑应用层限制单次读写的时间？
       4. 出现“Cache disabled but cached memory region accessed”，经常出现自动重启无法恢复，无法加载app层的程序（禁用PSRAM后，出错重启都能正常加载），必须使用硬件复位EN脚，才能正常工作，那这个情况是不是必须要求加看门狗？

costaud, 2020-06-29:
1/2/3: 请检查应用代码中是否有注册为ESP_INTR_FLAG_IRAM 的中断在运行。 在注册为ESP_INTR_FLAG_IRAM的中断处理函数中，是否调用了非IRAM的函数或者非DRAM的变量或数据(包括PSRAM中的数据)。

4: 能否提供可复现该‘无法恢复’问题的测试代码，或者提供一下log。

guo652917087, 2020-06-30:
@costaud ,
   无法自动恢复的日志如下：

I (2744) cpu_start: Pro cpu start user code
I (2749) spiram: Adding pool of 4076K of external SPI memory to heap allocator
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x4009c609  PS      : 0x00060d34  A0      : 0x800d3c64  A1      : 0x3ffe3be0  
A2      : 0x0300000c  A3      : 0x0100000c  A4      : 0x00000019  A5      : 0x383fc000  
A6      : 0x00000016  A7      : 0x00000000  A8      : 0x3ff00104  A9      : 0x3ffe3bb0  
A10     : 0x3ffc2960  A11     : 0x3ffc2968  A12     : 0x3ffc2960  A13     : 0x00060d20  
A14     : 0x000000fe  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x4009c606:0x3ffe3be0 0x400d3c61:0x3ffe3bf0 0x400817b5:0x3ffe3c10 0x40081ae1:0x3ffe3c40 0x4007922a:0x3ffe3c80 |<-CORRUPTED

ELF file SHA256: 6df671f07155e12e

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7152
ho 0 tail 12 room 4
load:0x40078000,len:13688
ho 0 tail 12 room 4
load:0x40080400,len:4000
entry 0x40080688
I (32) boot: ESP-IDF v4.2-dev-1303-gcd4fa46f5 2nd stage bootloader
I (33) boot: compile time 14:26:58
I (33) boot: chip revision: 3
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 8MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 0007d000
I (75) boot:  1 otadata          OTA data         01 00 00086000 00002000
I (82) boot:  2 factory          factory app      00 00 00090000 004e2000
I (90) boot:  3 storage          Unknown data     01 82 00572000 001f4000
I (97) boot: End of partition table
I (101) boot: Defaulting to factory image
I (106) esp_image: segment 0: paddr=0x00090020 vaddr=0x3f400020 size=0x2c47c8 (2901960) map
I (1220) esp_image: segment 1: paddr=0x003547f0 vaddr=0x3ffbdb60 size=0x04280 ( 17024) load
I (1227) esp_image: segment 2: paddr=0x00358a78 vaddr=0x40080000 size=0x00404 (  1028) load
I (1228) esp_image: segment 3: paddr=0x00358e84 vaddr=0x40080404 size=0x07194 ( 29076) load
I (1248) esp_image: segment 4: paddr=0x00360020 vaddr=0x400d0020 size=0x10adc4 (1093060) map
I (1665) esp_image: segment 5: paddr=0x0046adec vaddr=0x40087598 size=0x15cd0 ( 89296) load
I (1704) esp_image: segment 6: paddr=0x00480ac4 vaddr=0x00000000 size=0x0f4bc ( 62652) 
I (1743) boot: Loaded app from partition at offset 0x90000
I (1744) boot: Disabling RNG early entropy source...
I (1744) psram: This chip is ESP32-D0WD
I (1749) spiram: Found 64MBit SPI RAM device
I (1753) spiram: SPI RAM mode: flash 40m sram 40m
I (1759) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1766) cpu_start: Pro cpu up.
I (1770) cpu_start: Application information:
I (1775) cpu_start: Project name:     gateway
I (1780) cpu_start: App version:      v4.2-dev-1303-gcd4fa46f5
I (1786) cpu_start: Compile time:     Jun 30 2020 14:25:40
I (1793) cpu_start: ELF file SHA256:  6df671f07155e12e...
I (1799) cpu_start: ESP-IDF:          v4.2-dev-1303-gcd4fa46f5
I (1805) cpu_start: Starting app cpu, entry point is 0x4008188c
I (1796) cpu_start: App cpu up.
[... 77 lines cut ...]
I (2702) heap_init: Initializing. RAM available for dynamic allocation:
I (2702) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (2707) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (2713) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (2719) heap_init: At 3FFCB560 len 00014AA0 (82 KiB): DRAM
I (2725) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (2732) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (2738) heap_init: At 4009D268 len 00002D98 (11 KiB): IRAM
I (2744) cpu_start: Pro cpu start user code
I (2749) spiram: Adding pool of 4076K of external SPI memory to heap allocator
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x4009c609  PS      : 0x00060d34  A0      : 0x800d3c64  A1      : 0x3ffe3be0  
A2      : 0x0300000c  A3      : 0x0100000c  A4      : 0x00000019  A5      : 0x383fc000  
A6      : 0x00000016  A7      : 0x00000000  A8      : 0x3ff00104  A9      : 0x3ffe3bb0  
A10     : 0x3ffc2960  A11     : 0x3ffc2968  A12     : 0x3ffc2960  A13     : 0x00060d20  
A14     : 0x000000fe  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x4009c606:0x3ffe3be0 0x400d3c61:0x3ffe3bf0 0x400817b5:0x3ffe3c10 0x40081ae1:0x3ffe3c40 0x4007922a:0x3ffe3c80 |<-CORRUPTED

ELF file SHA256: 6df671f07155e12e

Rebooting...

guo652917087, 2020-06-30:
断电或者按复位键，启动如下：

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7152
ho 0 tail 12 room 4
load:0x40078000,len:13688
ho 0 tail 12 room 4
load:0x40080400,len:4000
entry 0x40080688
I (33) boot: ESP-IDF v4.2-dev-1303-gcd4fa46f5 2nd stage bootloader
I (33) boot: compile time 14:26:58
I (33) boot: chip revision: 3
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 8MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 0007d000
I (75) boot:  1 otadata          OTA data         01 00 00086000 00002000
I (82) boot:  2 factory          factory app      00 00 00090000 004e2000
I (90) boot:  3 storage          Unknown data     01 82 00572000 001f4000
I (97) boot: End of partition table
I (101) boot: Defaulting to factory image
I (106) esp_image: segment 0: paddr=0x00090020 vaddr=0x3f400020 size=0x2c47c8 (2901960) map
I (1220) esp_image: segment 1: paddr=0x003547f0 vaddr=0x3ffbdb60 size=0x04280 ( 17024) load
I (1227) esp_image: segment 2: paddr=0x00358a78 vaddr=0x40080000 size=0x00404 (  1028) load
I (1228) esp_image: segment 3: paddr=0x00358e84 vaddr=0x40080404 size=0x07194 ( 29076) load
I (1249) esp_image: segment 4: paddr=0x00360020 vaddr=0x400d0020 size=0x10adc8 (1093064) map
I (1665) esp_image: segment 5: paddr=0x0046adf0 vaddr=0x40087598 size=0x15cd0 ( 89296) load
I (1704) esp_image: segment 6: paddr=0x00480ac8 vaddr=0x00000000 size=0x0f4b8 ( 62648) 
I (1744) boot: Loaded app from partition at offset 0x90000
I (1744) boot: Disabling RNG early entropy source...
I (1744) psram: This chip is ESP32-D0WD
I (1749) spiram: Found 64MBit SPI RAM device
I (1754) spiram: SPI RAM mode: flash 40m sram 40m
I (1759) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1766) cpu_start: Pro cpu up.
I (1770) cpu_start: Application information:
I (1775) cpu_start: Project name:     gateway
I (1780) cpu_start: App version:      v4.2-dev-1303-gcd4fa46f5
I (1787) cpu_start: Compile time:     Jun 30 2020 14:25:40
I (1793) cpu_start: ELF file SHA256:  7f08508d3f5c0d87...
I (1799) cpu_start: ESP-IDF:          v4.2-dev-1303-gcd4fa46f5
I (1805) cpu_start: Starting app cpu, entry point is 0x4008188c
I (0) cpu_start: App cpu up.
I (2700) spiram: SPI SRAM memory test OK
I (2702) heap_init: Initializing. RAM available for dynamic allocation:
I (2702) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (2707) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (2713) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (2719) heap_init: At 3FFCB160 len 00014EA0 (83 KiB): DRAM
I (2725) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (2732) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (2738) heap_init: At 4009D268 len 00002D98 (11 KiB): IRAM
I (2744) cpu_start: Pro cpu start user code
I (2749) spiram: Adding pool of 4076K of external SPI memory to heap allocator
I (2771) spi_flash: detected chip: gd
I (2772) spi_flash: flash io: dio
I (2772) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (2780) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (2920) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3020) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
1970-01-01 00:00:00 INFO <filesystem> Initializing SPIFFS
1970-01-01 00:00:00 INFO <filesystem> Partition size: total: 1875221, used: 1523068
I (3200) system_api: Base MAC address is not set
I (3200) system_api: read default base MAC address from EFUSE
I (3220) esp_eth.netif.glue: fc:f5:c4:3c:59:3f
I (3220) esp_eth.netif.glue: ethernet attached to netif
I (3240) wifi:wifi driver task: 3ffdfd94, prio:23, stack:6656, core=0
I (3280) wifi:wifi firmware version: cc57434
I (3280) wifi:wifi certification version: v7.0
I (3280) wifi:config NVS flash: enabled
I (3280) wifi:config nano formating: disabled
I (3300) wifi:Init dynamic tx buffer num: 32
I (3300) wifi:Init data frame dynamic rx buffer num: 32
I (3300) wifi:Init management frame dynamic rx buffer num: 32
I (3310) wifi:Init management short buffer num: 32
I (3310) wifi:Init static tx buffer num: 32
I (3320) wifi:Init static rx buffer size: 2212
I (3320) wifi:Init static rx buffer num: 10
I (3320) wifi:Init dynamic rx buffer num: 32
I (3420) phy: phy_version: 4181, c9a03e2, Mar 10 2020, 12:45:06, 0, 0
I (3420) wifi:mode : sta (fc:f5:c4:3c:59:3c)

guo652917087, 2020-06-30:
我把日志输出等级改为debug：

Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x4009c6d5  PS      : 0x00060d34  A0      : 0x800d3f70  A1      : 0x3ffe3be0  
A2      : 0x0300000c  A3      : 0x0100000c  A4      : 0x00000019  A5      : 0x383fc000  
A6      : 0x00000016  A7      : 0x00000000  A8      : 0x3ff00104  A9      : 0x3ffe3bb0  
A10     : 0x3ffc29e8  A11     : 0x3ffc29f0  A12     : 0x3ffc29e8  A13     : 0x00060d20  
A14     : 0x000000fe  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x4009c6d2:0x3ffe3be0 0x400d3f6d:0x3ffe3bf0 0x400817d5:0x3ffe3c10 0x40081b01:0x3ffe3c40 0x4007922a:0x3ffe3c80 |<-CORRUPTED

ELF file SHA256: 680e8fc3ba095922

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7152
ho 0 tail 12 room 4
load:0x40078000,len:13688
ho 0 tail 12 room 4
load:0x40080400,len:4000
entry 0x40080688
I (32) boot: ESP-IDF v4.2-dev-1303-gcd4fa46f5 2nd stage bootloader
I (33) boot: compile time 14:58:08
I (33) boot: chip revision: 3
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 8MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 0007d000
I (75) boot:  1 otadata          OTA data         01 00 00086000 00002000
I (82) boot:  2 factory          factory app      00 00 00090000 004e2000
I (90) boot:  3 storage          Unknown data     01 82 00572000 001f4000
I (97) boot: End of partition table
I (101) boot: Defaulting to factory image
I (106) esp_image: segment 0: paddr=0x00090020 vaddr=0x3f400020 size=0x2cd134 (2937140) map
I (1233) esp_image: segment 1: paddr=0x0035d15c vaddr=0x3ffbdb60 size=0x02ebc ( 11964) load
I (1238) esp_image: segment 2: paddr=0x00360020 vaddr=0x400d0020 size=0x112420 (1123360) map
I (1668) esp_image: segment 3: paddr=0x00472448 vaddr=0x3ffc0a1c size=0x0144c (  5196) load
I (1670) esp_image: segment 4: paddr=0x0047389c vaddr=0x40080000 size=0x00404 (  1028) load
I (1675) esp_image: segment 5: paddr=0x00473ca8 vaddr=0x40080404 size=0x1cf30 (118576) load
I (1735) esp_image: segment 6: paddr=0x00490be0 vaddr=0x00000000 size=0x0f3a0 ( 62368) 
I (1775) boot: Loaded app from partition at offset 0x90000
I (1775) boot: Disabling RNG early entropy source...
I (1776) psram: This chip is ESP32-D0WD
I (1780) spiram: Found 64MBit SPI RAM device
I (1785) spiram: SPI RAM mode: flash 40m sram 40m
I (1790) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1797) cpu_start: Pro cpu up.
I (1801) cpu_start: Application information:
I (1806) cpu_start: Project name:     gateway
I (1811) cpu_start: App version:      v4.2-dev-1303-gcd4fa46f5
I (1818) cpu_start: Compile time:     Jun 30 2020 14:56:52
I (1824) cpu_start: ELF file SHA256:  680e8fc3ba095922...
I (1830) cpu_start: ESP-IDF:          v4.2-dev-1303-gcd4fa46f5
I (1837) cpu_start: Starting app cpu, entry point is 0x400818ac
I (1827) cpu_start: App cpu up.
I (2731) spiram: SPI SRAM memory test OK
D (2733) memory_layout: Checking 12 reserved memory ranges:
D (2733) memory_layout: Reserved memory range 0x3f800000 - 0x3fc00000
[... 162 lines cut ...]
I (3008) heap_init: At 3FFCB1E8 len 00014E18 (83 KiB): DRAM
I (3014) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (3021) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (3027) heap_init: New heap initialised at 0x4009d334
I (3033) heap_init: At 4009D334 len 00002CCC (11 KiB): IRAM
I (3039) cpu_start: Pro cpu start user code
I (3044) spiram: Adding pool of 4076K of external SPI memory to heap allocator
D (3059) clk: RTC_SLOW_CLK calibration value: 3235622
D (3069) intr_alloc: Connected src 46 to int 2 (cpu 0)
D (3069) intr_alloc: Connected src 17 to int 3 (cpu 0)
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x4009c6d5  PS      : 0x00060d34  A0      : 0x800d3f70  A1      : 0x3ffe3be0  
A2      : 0x0300000c  A3      : 0x0100000c  A4      : 0x00000019  A5      : 0x383fc000  
A6      : 0x00000016  A7      : 0x00000000  A8      : 0x3ff00104  A9      : 0x3ffe3bb0  
A10     : 0x3ffc29e8  A11     : 0x3ffc29f0  A12     : 0x3ffc29e8  A13     : 0x00060d20  
A14     : 0x000000fe  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x4009c6d2:0x3ffe3be0 0x400d3f6d:0x3ffe3bf0 0x400817d5:0x3ffe3c10 0x40081b01:0x3ffe3c40 0x4007922a:0x3ffe3c80 |<-CORRUPTED

ELF file SHA256: 680e8fc3ba095922

Rebooting...

costaud, 2020-06-30:
从log看，还是某个定义为IRAM的终端引起的。
请检查代码中用了哪些外设中断，那些是出时候时带IRAM_FLAG的？

guo652917087, 2020-06-30:
@costaud ，
   软件用到以太网、wifi、ble、tft屏（HSPI）、UART、LEDC、SPIFFS，其中spi和uart的中断IRAM我已经在menuconfig里选上，wifi相关的IRAM由于IRAM不足，我只选择了 【*】WiFi IRAM speed optimization。我自己写的程序，没有用到ESP_INTR_FLAG_IRAM。

costaud, 2020-06-30:
menuconfig里将spi/uart 中断IRAM 取消。
如果问题消失，再定位是spi还是uart中断造成的。 
spi的callback函数是否在IRAM中 以及 uart buffer是否是在DRAM中。

guo652917087, 2020-07-01:
把几个IRAM的选型取消，但IRAM的内存没有释放到esp_get_free_heap_size里，我想把PSAM的部分内容都用在片上内存。
我看到BT有个配置文件
CONFIG_FREERTOS_UNICORE=y
CONFIG_ESP32_IRAM_AS_8BIT_ACCESSIBLE_MEMORY=y
CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_IRAM_8BIT=y

应该是使用IRAM内存的方法吧？只能单核吗？

closed 2020-07-02 by guo652917087