espressif/esp-idf #5512: Invalid escape sequence \o

klaus-liebler, 2020-06-28:
**This issue is related to the last two posts in https://github.com/espressif/esp-idf/issues/4398, but this new issue has not been filed yet... So, I do this now...**

## Environment

- Development Kit:      ESP32-DevKitC
- Module or chip used:  ESP32
- IDF version: ESP-IDF v4.2-dev-1905-g625bd5eb1 (tested with an older ESP-IDF v4.2-dev-1660 as well)
- Build System: CMake|idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     Windows 10 Pro 64bit ESP Command Prompt
- Using an IDE?: VS Code, but build is in ESP command prompt
- Power Supply:         USB

## Problem Description

Building a trivial project just works fine. Then I tried to add the 3rd party component https://github.com/bbulkow/FastLED-idf as follows

```
idf_component_register(
    SRCS "main.cc" "functionblocks.cc" "plcmanager.cc"
    REQUIRES "FastLED-idf" <---- I just added this line
    EMBED_FILES "../../devtools/fbexec.data")
```

Of course, I cloned the FastLED-idf repository into {project_dir}\components

### Expected Behavior
Successful build

### Actual Behavior
```
c:\repos\labathome\firmware>idf build
Executing action: all (aliases: build)
Running cmake in directory c:\repos\labathome\firmware\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 c:\repos\labathome\firmware"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.26.0.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/opt/espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/opt/espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/opt/espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/opt/espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/opt/espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.26.0.windows.1")
CMake Error at C:/opt/esp-idf/tools/cmake/component.cmake:221 (message):
  CMake Warning (dev) at build_properties.temp.cmake:2 (set):

    Syntax error in cmake code at

      C:/repos/labathome/firmware/build/build_properties.temp.cmake:2

    when parsing string

      C:\opt\espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe

    Invalid escape sequence \o

    Policy CMP0010 is not set: Bad variable reference syntax is an error.  Run
    "cmake --help-policy CMP0010" for policy details.  Use the cmake_policy
    command to set the policy and suppress this warning.

  Call Stack (most recent call first):

    C:/opt/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:1 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  CMake Error at
  C:/opt/espressif/tools/cmake/3.16.4/share/cmake-3.16/Modules/ExternalProject.cmake:1029
  (define_property):

    define_property command is not scriptable

  Call Stack (most recent call first):

    C:/opt/esp-idf/tools/cmake/kconfig.cmake:1 (include)
    C:/opt/esp-idf/tools/cmake/idf.cmake:41 (include)
    C:/opt/esp-idf/tools/cmake/project.cmake:7 (include)
    C:/repos/labathome/firmware/components/FastLED-idf/CMakeLists.txt:4 (include)
    C:/opt/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:60 (include)
    C:/opt/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:76 (__component_get_requirements)

Call Stack (most recent call first):
  C:/opt/esp-idf/tools/cmake/build.cmake:405 (__component_get_requirements)
  C:/opt/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:3 (project)

-- Configuring incomplete, errors occurred!
See also "C:/repos/labathome/firmware/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

### Steps to reproduce

1. clone simple project
2. REQUIRES 3rd party lib

Alvin1Zhang, 2020-06-29:
Thanks for reporting, we will look into.

keelung-yang, 2020-06-29:
Same issue on latest master after adding a line to official modbus example: mb_slave\main\CMakeLists.txt
`add_definitions(-DINCLUDE_xTaskGetHandle)`

[image]

roblatour, 2020-07-12:
I had the same problem, the fix above worked for me.

Aietes, 2020-07-13:
Same problem, the fix by @keelung-yang worked. Thank you!

renzbagaporo, 2020-07-13:
Hi, sorry about this. An official fix has been merged internally.

nxthich, 2020-07-19:
@keelung-yang this workaround worked for me. Thank you!

gecko242, 2020-07-20:
@keelung-yang Thanks for the workaround, this fixed it for me too :)

mvacha, 2020-07-22:
If you are using VS Code and this issue persists even after patching the `idf.py` file, try changing all backslashes `\` to forward slashes `/` in Settings/ESP-IDF/Python Bin Path Win (`idf.pythonBinPathWin`):
[image]
```
"idf.pythonBinPathWin": "c:/espressif-tools-latest/python_env/idf4.3_py3.8_env/Scripts/python.exe"
```
This resolved the issue for me entirely.

luzhichao1997606, 2020-07-31:
@mvacha 
This resolved the issue for me entirely.
thanks！

ankit-thealchemist, 2020-08-02:
> If you are using VS Code and this issue persists even after patching the `idf.py` file, try changing all backslashes `\` to forward slashes `/` in Settings/ESP-IDF/Python Bin Path Win [...]

This resolves mine problem in ESP-IDF extension in vs code. Thanks!

Erol444, 2021-02-03:
> If you are using VS Code and this issue persists even after patching the `idf.py` file, try changing all backslashes `\` to forward slashes `/` in Settings/ESP-IDF/Python Bin Path Win [...]

Worked for me as well, using Windows and VS code ESP-IDF extension. Thanks!

Alvin1Zhang, 2021-08-25:
Thanks for reporting and updates, sorry for slow turnaround, fix is available at https://github.com/espressif/esp-idf/commit/f5b1f43a2b0d88493224ed3587bf67f8c52eaa5b, feel free to reopen.

closed 2021-08-25 by Alvin1Zhang