espressif/esp-idf #5514: VFS root base_path faulty error-checking

BrianPugh, 2020-06-28:
[According to documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/vfs.html#_CPPv416esp_vfs_registerPKcPK9esp_vfs_tPv), in `esp_vfs_register` the base_path must meet the following specifications:

> base_path: file path prefix associated with the filesystem. Must be a zero-terminated C string, up to ESP_VFS_PATH_MAX characters long, *and at least 2 characters long*. Name *must start with a “/” and must not end with “/”*. For example, “/data” or “/dev/spi” are valid. These VFSes would then be called to handle file paths such as “/data/myfile.txt” or “/dev/spi/0”.

The logic for validating the `base_path` is [here](https://github.com/espressif/esp-idf/blob/625bd5eb1806809ff3cc010ee20d1f750aa778a1/components/vfs/vfs.c#L78)

The current logic has the following fault(s):

An empty string, i.e. `""` will have a length of 0 and *probably* passes these checks. Is this desired behavior? With a length of 0, checking ` base_path[len - 1] == '/'` results in checking `base_path[-1] == '/'`. Let's use SPIFFS as an example; since `base_path` is the first element in the `esp_vfs_spiffs_conf_t` structure, `base_path[-1]` points to totally unrelated data and could have a chance of triggering that logic. Needless to say, this is probably undesired.  Instead of:

```
if ((len > 0 && base_path[0] != '/') || base_path[len - 1] == '/') {
```
if probably should have been:
```
if (base_path[0] != '/' || (len > 0 && base_path[len - 1] == '/')) {
```

Or, just in general, why not just first explicitly check that `len > 2` ?

igrr, 2020-06-28:
> An empty string, i.e. "" will have a length of 0 and probably passes these checks. Is this desired behavior?

This is intended behavior, but looks like the documentation hasn't been updated to describe it.
The feature was requested in https://github.com/espressif/esp-idf/issues/135, implemented in 5b678ee with a follow-up in https://github.com/espressif/esp-idf/pull/726. Related test case is https://github.com/espressif/esp-idf/blob/625bd5eb1806809ff3cc010ee20d1f750aa778a1/components/vfs/test/test_vfs_paths.c#L185-L190

> With a length of 0, checking base_path[len - 1] == '/' results in checking base_path[-1] == '/'

You are right, this looks like a bug. Thanks for pointing this out, will fix, and also update the docs to mention the case of empty base_path.

BrianPugh, 2020-06-28:
So related to that PR, is `"/"` supposed to be a valid mount point? Because that fails these checks as well.

igrr, 2020-06-28:
No, "/" is not a valid mount point: https://github.com/espressif/esp-idf/blob/625bd5eb1806809ff3cc010ee20d1f750aa778a1/components/vfs/test/test_vfs_paths.c#L272.

closed 2020-08-03 by espressif-bot