espressif/esp-idf #5515: Add configurable logical processor option (e.g. ninja -j 8) to idf.py build

thomasnugent, 2020-06-29:
Hi all, [this link](https://www.reddit.com/r/esp32/comments/8fvy99/advice_boost_espidf_compile_time_on_windows_10) suggests to specify the number of logical processors to use in a `make` or `ninja` call with a parameter like `-j 8` for 8 logical processors. Firstly, that thread is quite old, and I don't think the new CMake system was in place. But I still find compile time is quite long and often requires ~550 C/CXX objects to be built.

Since `idf.py` wraps these calls, I was wondering if [an option](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#idf-py) could be added to `idf.py build` (such as `idf.py build -j 8`) to do this without digging into the `ninja` calls for example.

Is there already a way to do this? Is there a benefit? I'm still getting my head around it all. In the meantime, is there a way I can do this manually? I can't seem to find where the `idf.py` calls `ninja`.

Further, are there more ways to increase compile time on Windows apart from:

* Specifying the minimum number of components in the project-level `CMakeLists.txt`: for e.g. 
```set(COMPONENTS bootloader esptool_py main)```; and
* Setting the `--ccache` flag when building?

e.g. of my very simple project's build output:

```
[512/547] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
```

Alvin1Zhang, 2020-06-30:
Thanks for raising this feature request.

projectgus, 2020-06-30:
Hi @thomasnugent,

Without an explicit argument, `ninja` will default to "number of CPUs" parallel jobs (unlike `make` which defaults to 1, which is why this advice was more valuable when using make as the build system).

You can run `ninja` directly by changing to the project build directory and running it instead of `idf.py` (this will work after the first time `idf.py` has been run). ie `cd build` and then `ninja -j 8`. [More details here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#using-cmake-directly).

If you see a big performance boost when building with more parallel jobs than the default then we'd certainly be happy to add an option to `idf.py` to pass this through. If it doesn't seem to help significantly then we'll probably not add it, as there's already a way to get the same behaviour by running ninja directly. (It may not help in the same way, ninja was explicitly designed to be a more efficient build system then make and to perform better on WIndows.) 

The options you mention to reduce build times are the best choices. The other thing which can heavily influence build times on Windows is Windows Defender, which can sometimes use >50% of the total CPU just monitoring the build process! The ESP-IDF Tools installer will prompt to optionally add some exception rules to Defender when it runs.

Some questions to try and figure out if there's any other "low hanging fruit" for improving build times:

* Approximately how long does a clean build of your project take, and what computer are you building on? (CPU, disk type, etc.)
* If you run Windows Task Monitor while building, do you see any idle CPU and do you see any other processes apart from ninja and `xtensa-esp32-elf-gcc` using high % of CPU? (This is where the Windows Defender service may suddenly show up and hog the CPU.)
* If you run `ccache -s` (for stats), do the stats show an OK "cache hit rate", ie 40% or more? (Meaning that ccache is being used successfully.)

thomasnugent, 2020-07-03:
Hi @projectgus, thanks so much for your response! 

I couldn't manage to add the correct exclusions on my current Antivirus provider (Kaspersky) -- if anyone has tips for this that would be awesome. I tried adding cmd.exe to trusted apps, and added all esp-idf relevant folders to excluded folders, but had no luck. Windows Defender works much better and build time has halved!

I did find though that the **Windows Search feature** eats up most of my CPU. Any tips on reducing its impact?
And Windows Defender (specifically the **Antimalware Service**) still looks like it's using the CPU.

[image]

From a fullclean, my program takes 33s to build. That's without specifying `set(COMPONENTS bootloader esptool_py main)`. Cache hit rate looks good now also. These are most of my specs if they give context at all.

[image]

projectgus, 2020-07-03:
Hi @thomasnugent ,

For Windows Search, you can open Windows Search Settings (will show up if you type into the start menu search) and add your project directory to the Excluded Paths list.

Interesting that you're still seeing some overhead from Windows Defender. If the ESP-IDF directory and the ESP-IDF Tools directory are in the exclusions list then I'm not sure there is anything else that will easily reduce the overhead.

33 seconds seems like an OK time for a clean build, by the way. My development machine is Linux and on paper it's about 25% faster CPU (i7-10510U, 25% estimate comes from eyeballing Geekbench numbers). A `fullclean` build of unmodified ESP-IDF "hello world" example takes me 27s.

You should not need to do this kind of "full build" often, only when changing the project's configuration settings which is hopefully not a frequent thing.

chegewara, 2020-07-03:
On windows 33 sec is very good time. I have ryzen 5 1600 (6 core), PCIE disk and 32GB ram on linux and full build is not much faster. You have only 6 GB ram and most of it is taken by windows. If i had 33 sec build on windows i wouldnt move to linux. My windows build time was few minutes on laptop some time ago.

thomasnugent, 2020-07-06:
Thanks again guys, I've stopped the Windows Search and am really happy with my build time now! This has helped significantly :) 

To answer my original question: `ninja -j 8` doesn't seem to improve performance compared to using good 'ol `idf.py build`. Windows Defender exceptions and specifying specific components worked a treat though!

projectgus, 2020-07-10:
Hi @thomasnugent ,

Good to hear. Do you think this can be closed, then?

Angus

thomasnugent, 2020-07-10:
Yep absolutely, thanks @projectgus!

closed 2020-07-10 by thomasnugent