espressif/esp-idf #5517: Thread safety of multiple devices on a single SPI bus

KaeLL, 2020-06-29:
- Development Kit:      none
- Module or chip used:  ESP32D0WDQ6 (revision 1)
- IDF version: v4.2-dev-1905-g625bd5eb1
- Build System:        make
- Compiler version: esp-2020r2-8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 5V

I'm having issues with [this](https://github.com/espressif/esp-idf/blob/625bd5eb1806809ff3cc010ee20d1f750aa778a1/components/driver/spi_master.c#L933) assertion triggering intermittently.

I have 2 SPI devices on the same bus (SPI3). In the warning [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#driver-features), it says I have to just `add a mutex lock around the shared Device using xSemaphoreCreateMutex`, which is exactly what I'm doing, but the app breaks. The way I'm doing it is by not locking all spi transactions behind a single universal mutex, but rather having a mutex for each SPI device, and locking the one linked to that specific device.

So for example, instead of this:
```c
SemaphoreHandle_t spi_semphr;

void spi_trans( spi_device_handle_t spi_device, uint8_t *buf_w, uint8_t *buf_r, uint8_t len )
{
	xSemaphoreTake( spi_semphr, portMAX_DELAY );

	ESP_ERROR_CHECK( spi_device_polling_transmit( spi_device,
		&( spi_transaction_t ){ .length = 8 * len, .tx_buffer = buf_w, .rx_buffer = buf_r } ) );

	xSemaphoreGive( spi_semphr );
}
```
I'd do something like this:
```c
SemaphoreHandle_t spi_dev1_semphr;
SemaphoreHandle_t spi_dev2_semphr;

spi_device_handle_t spi_dev1;
spi_device_handle_t spi_dev2;

void spi_trans_dev1( uint8_t *buf_w, uint8_t *buf_r, uint8_t len )
{
	xSemaphoreTake( spi_dev1_semphr, portMAX_DELAY );

	ESP_ERROR_CHECK( spi_device_polling_transmit( spi_dev1,
		&( spi_transaction_t ){ .length = 8 * len, .tx_buffer = buf_w, .rx_buffer = buf_r } ) );

	xSemaphoreGive( spi_dev1_semphr );
}

void spi_trans_dev2( uint8_t *buf_w, uint8_t *buf_r, uint8_t len )
{
	xSemaphoreTake( spi_dev2_semphr, portMAX_DELAY );

	ESP_ERROR_CHECK( spi_device_polling_transmit( spi_dev2,
		&( spi_transaction_t ){ .length = 8 * len, .tx_buffer = buf_w, .rx_buffer = buf_r } ) );

	xSemaphoreGive( spi_dev2_semphr );
}
```
I can avoid this issue by reverting to use one global mutex for all transactions, which brings me to the questions: what's going on? Am I missing something? How thread-safe is the spi master library really?

lyris85, 2020-07-10:
you should use only 1 mutex for each spi bus, not device. the mutex prevent you to access the bus when it is busy. if you have 2 mutex, then device 1 and device 2 can access the bus at the same time and you get your assertion.

KaeLL, 2020-07-10:
> you should use only 1 mutex for each spi bus, not device. the mutex prevent you to access the bus when it is busy. if you have 2 mutex, then device 1 and device 2 can access the bus at the same time and you get your assertion.

Reading must be difficult, so no worries
[image]

lyris85, 2020-07-11:
I think the concept you are missing ismutex. Here is a short explaination https://www.webopedia.com/TERM/M/mutex.html

KaeLL, 2020-07-11:
Thanks, I guess? In return, could you do me a favor and actually read what I wrote beyond the title and presume the rest? Thanks in advance.

KaeLL, 2020-10-09:
@igrr I don't like tagging anyone but as it seems to be a trivial thing, its importance to me isn't. If the docs are right, I could avoid redundantly locking the entire SPI bus, but I also wouldn't have the assert triggering on me. Just throw me a bone or something.

KaeLL, 2020-12-15:
@ginkgm Can you take a look at this? Is lyris85 right? In that case, the doc printed above is wrong because it means 1 task per SPI device, not 1 task per SPI bus, and it should be reworded to "... As long as __all devices on the same bus are__ accessed by only one task ... if multiple tasks try to access __any of the__ SPI devices...".
Or the docs are correct and the spi_master component has always been bugged because it doesn't implicitly handle the 1 task/device approach.
Btw, if my first hypothesis is correct, why is that the case? Why doesn't spi_master offers a way to implicitly handle concurrent accesses to the same bus, through the `spi_device_polling_transmit` and `spi_device_transmit` wrappers or similar? They fulfill this role of being the simplest and easiest APIs for SPI transactions anyway.

KaeLL, 2020-12-22:
Answered [here](https://www.esp32.com/viewtopic.php?f=13&t=17562). I'll close it, even though the suggestion for improving the documentation remains.

closed 2020-12-22 by KaeLL

philippe44, 2022-01-03:
I'm probably slow but I'm not sure I understand the conclusion. I'm having that same problem https://github.com/espressif/esp-idf/issues/8179 but my understanding of the doc is that you can access multiple devices (for simplification I mean one device = a peripheral with its own CS line, represented by a handle obtained by spi_bus_add_device) without worrying as long as each device is accessed by a single task. 

In other words, the driver does not handle for you concurrent accesses to the same handle that you obtained from spi_bus_add_device() which seems reasonable to me, as it is purely under the control of the user's application.

Now, what I'm observing is that I need to put a global mutex on the SPI *host* (SPI0..3) for it to work. That sounds counter-intuitive to me because that means, for example, that SPI-to-ethernet devices cannot coexist with anything, unless I patch esp-idf's spi_device_polling_transmit

KaeLL, 2022-01-05:
@philippe44 conclusion is that I'm right. @Spritetm cleared it up. The wording in the docs section I posted the print of is wrong, and "Device" really means bus there, not actually each of the up to 3 devices on the same bus. So it's either using a bus-wide mutex, or a single task governing each bus, or rolling up your own spi_master with actual device-wide mutual exclusion.

philippe44, 2022-01-05:
@KaeLL thanks for answering. What puzzles me is that all SPI-Ethernet driver are problematic then because they require exclusive usage of an SPI bus, and when you use SPIRAM @ 80MHz, there is only one left. It sounds to me like a big flaw. Of course, I can (and this is what I did) change spi_master to include a mutex per host (if I understand the concept :smile:)

Now, consider the following code
```
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/timers.h"
#include "esp_task.h"
#include "esp_log.h"
#include "esp_heap_caps.h"
#include "driver/spi_master.h"

static char TAG[] = "spi-bug";

spi_device_interface_config_t config;
spi_device_handle_t handle1 = NULL, handle2 = NULL;

bool inTask;
int taskCount, timerCount;

static esp_err_t spi_write(spi_device_handle_t handle, size_t len) {
    spi_transaction_t transaction = { };
	uint8_t *buffer = heap_caps_malloc(len, MALLOC_CAP_SPIRAM);

	transaction.tx_buffer = buffer;
	transaction.length = len*8;

	esp_err_t ret = spi_device_polling_transmit(handle, &transaction);
	free(buffer);
	return ret;
}

static void task(void *arg) {
	while (1) {
		taskCount++;
		inTask = true;		
		spi_write(handle2, 1024);
		inTask = false;
	}
}

static void timer(void *arg) {
	static int count = 0;
	if (inTask) {
		//ESP_LOGI(TAG, "Timer (likely) interrupting task");		
	}
	timerCount++;
	spi_write(handle1, 128);
}

void app_main()
{
	spi_bus_config_t spi_config = {
		.mosi_io_num = 23,
		.miso_io_num = 19,
		.sclk_io_num = 18,
		.quadwp_io_num = -1,
        .quadhd_io_num = -1
	};

	esp_err_t err = spi_bus_initialize( SPI2_HOST, &spi_config, 1 );
	ESP_LOGI(TAG, "SPI host %d init %d", SPI2_HOST, err);
	
	memset(&config, 0, sizeof(spi_device_interface_config_t));
    config.clock_speed_hz = SPI_MASTER_FREQ_8M;
	config.clock_speed_hz = 1*1000*1000;
    config.spics_io_num = 22;
    config.queue_size = 10;
	config.flags = SPI_DEVICE_NO_DUMMY;
	err = spi_bus_add_device(SPI2_HOST, &config, &handle1);
	ESP_LOGI(TAG, "SPI device %p %d", handle1, err);
	
	memset(&config, 0, sizeof(spi_device_interface_config_t));
	config.clock_speed_hz = SPI_MASTER_FREQ_8M;
	config.clock_speed_hz = 1*1000*1000;
    config.spics_io_num = 22;
    config.queue_size = 1;
	config.flags = SPI_DEVICE_NO_DUMMY;
	err = spi_bus_add_device(SPI2_HOST, &config, &handle2);
	ESP_LOGI(TAG, "SPI device %p %d", handle2, err);
	
	xTaskCreate(task, "task", 4096, NULL, ESP_TASK_PRIO_MIN + 1, NULL);
    xTimerStart(xTimerCreate("timer", pdMS_TO_TICKS(10), pdTRUE, NULL, timer), portMAX_DELAY);
	
	while(1) {
		ESP_LOGI(TAG, "TaskCount %d TimerCount %d", taskCount, timerCount);		
		vTaskDelay(pdMS_TO_TICKS(2000));
    }
}
```
It's not supposed to be pretty but, knowing that timer task is @ priority 10, shouldn't it fail very often? The low-priority task is saturating the bus with "long" transfers and the timer is interrupting it as frequently as possible. That has been running for a day w/o any issue where the real app with the w5500 driver and a SPI display fails after a few seconds.

Spritetm, 2022-01-06:
> @philippe44 conclusion is that I'm right. @Spritetm cleared it up. The wording in the docs section I posted the print of is wrong, and "Device" really means bus there, not actually each of the up to [...]

How do you read that in my response there? I'm actually saying the exact opposite: the API is supposed to be threadsafe as long as you keep each one of the 3 devices in the same thread.

philippe44, 2022-01-06:
That makes sense to me but I was not sure anymore what to think. So yes, I absolutely have a case https://github.com/espressif/esp-idf/issues/8179 where 2 threads access 2 different *devices* but on the same bus and randomly, after a few seconds, one thinks that it as acquired the lock but if you check right when exiting the spi_bus_lock_acquire_start(), you're not the acquiring device. There is only one call to spi_device_polling_transmit() in my code, the other ones are made by w5500 which have a mutex. Just to be sure, I've traces that there is no re-entrance in my own call to spi_device_polling_transmit(). As long as I put a per-host mutex inside spi_device_polling_transmit() then it will work for hours.

KaeLL, 2022-01-06:
@Spritetm
> How do you read that in my response there?

You know what? I have no idea. I'm actually asking myself that same question right now. That's me being stupid again. Thanks for pointing it out and sorry about that. It'll probably happen again.

> If you have a case where it's not, that is a bug and it should be fixed.

Great, because I've had it for a year and half now, and I do recall assuming the docs were wrong because, well, I had no reason to believe otherwise, but had suggestions to believe it so.
No one else was complaining about, @lyris85 was the only one who bothered commenting and he reinforced the idea of a bus-wide mutex; the chip runs code, not documentation, so I'm inclined to believe the former rather than the latter, especially regarding a behavior that isn't necessarily a bug by itself, but an inefficiency. So I guess you can see where I'm coming from.

kTakumo, 2025-09-18:
For whom it may concern: 

I dont know if this is relevant for anyone else coming across this situation, but I have a problem when two logical devices on the same bus run in different tasks on different cores. Its leading to frequent deadlocks or worse. One task is using interrupt transfers and polling transfers while the other is only using polling transfers. The deadlock happens in the polling transmission function, sometimes also some assertions are triggered. I debugged the situation quiet a bit and found this in the code: 

`3. Mixed mode

   The driver is written under the assumption that polling and interrupt
   transactions are not happening simultaneously. When sending polling
   transactions, it will check whether the ISR is active, which includes the
   case the ISR is sending the interrupt transactions of the acquiring
   device. If the ISR is still working, the routine sending a polling
   transaction will get blocked and wait until the semaphore from the ISR
   which indicates the ISR is free now.

   A fatal case is, a polling transaction is in flight, but the ISR received
   an interrupt transaction. The behavior of the driver is unpredictable,
   which should be strictly forbidden.`

When analyzing the spi_device_acquire_bus  function i noticed that it uses the following function: 
`
// Debug only
// NOTE if the acquiring is not fully completed, `spi_bus_lock_get_acquiring_dev`
// may return a false `NULL` cause the function returning false `false`.
static inline SPI_MASTER_ISR_ATTR bool spi_bus_device_is_polling(spi_device_t *dev)
{
    return get_acquiring_dev(dev->host) == dev && dev->host->polling;
}
`

So I assume that it is possible to start an interrupt transfer from one device although the other device is preparing an polling transfer if the timing is right.  That leads to the exact situation that is "strictly forbidden". So the looks to me that the SPI-Masterbus driver can fail if mixing different devices from different threads and mixing polling and interupt transfers.