espressif/esp-idf #5518: MQTT错误

Bruce1221, 2020-06-30:
## Environment

- Development Kit:         [none]
- Module or chip used:  [ESP32]
- IDF version:                  [v4.0.0]
- Build System:               [idf.py]

## Problem Description
使用idf中的例子，可以连接到mqtt服务器并发送消息，但是发送完成后会报错并重连，错误如下
mqtt_message_receive: received a message with an invalid header=0x92
之前没有使用idf的esp8266连接是没有问题的
### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp

static int mqtt_flag = ESP_FAIL;
QueueHandle_t mqtt_recv_queue;
QueueHandle_t mqtt_send_queue;

esp_mqtt_client_handle_t client;

// 事件处理函数
static void mqtt_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, void *event_data) 
{
    //printf("Event dispatched from event loop base=%s, event_id=%d \n", base, event_id);
    // 获取MQTT客户端结构体指针
    esp_mqtt_event_handle_t event = event_data;
    esp_mqtt_client_handle_t client = event->client;
    char sub_topic[64] = { 0 };
    // 通过事件ID来分别处理对应的事件
    switch (event->event_id) 
    {
        // 建立连接成功
        case MQTT_EVENT_CONNECTED:
            mqtt_flag = ESP_OK;
            size_t len = sizeof(sub_topic);
            config_read_str("mqtt_sub", sub_topic, &len);
            esp_mqtt_client_subscribe(client, sub_topic, 0);           
            break;
        // 客户端断开连接
        case MQTT_EVENT_DISCONNECTED:
            mqtt_flag = ESP_FAIL;
            printf("MQTT_client have disconnected. \n");
            break;
        // 主题订阅成功
        case MQTT_EVENT_SUBSCRIBED:
            printf("mqtt subscribe ok. msg_id = %d \n",event->msg_id);
            break;
        // 取消订阅
        case MQTT_EVENT_UNSUBSCRIBED:
            printf("mqtt unsubscribe ok. msg_id = %d \n",event->msg_id);
            break;
        //  主题发布成功
        case MQTT_EVENT_PUBLISHED:
            printf("mqtt published ok. msg_id = %d \n",event->msg_id);
            break;
        // 已收到订阅的主题消息
        case MQTT_EVENT_DATA:
            //printf("mqtt received topic: %.*s \n",event->topic_len, event->topic);
            //printf("topic data: %.*s\r\n", event->data_len, event->data);
            ;uint16_t data_len = (event->data_len);
            uint16_t topic_len = (event->topic_len);
            char *topicBuf = (char*)calloc(1, topic_len+1);
            char *dataBuf = (char*)calloc(1, data_len+1);
            memcpy(dataBuf, event->data, data_len);
            dataBuf[data_len] = 0;
            memcpy(topicBuf, event->topic, topic_len);
            topicBuf[topic_len] = 0;
            xQueueSendFromISR(mqtt_recv_queue, dataBuf, NULL);
            free(topicBuf);
            free(dataBuf);
            break;
        // 客户端遇到错误
        case MQTT_EVENT_ERROR:
            mqtt_flag = ESP_FAIL;
            printf("MQTT_EVENT_ERROR \n");
            break;
        default:
            //printf("Other event id:%d \n", event->event_id);
            break;
    }
}

void mqtt_app_start(void)
{
    char mqtt_host[64] = { 0 };
    uint32_t mqtt_port;
    char mqtt_user[32] = { 0 };
    char device_id[32] = { 0 };
    size_t len;

    len = sizeof(mqtt_host);
    config_read_str("mqtt_host", mqtt_host, &len);
    config_read_u32("mqtt_port", &mqtt_port);
    len = sizeof(mqtt_user);
    config_read_str("mqtt_user", mqtt_user, &len);
    len = sizeof(device_id);
    config_read_str("device_id", device_id, &len);

    esp_mqtt_client_config_t mqtt_cfg = {
        .host = mqtt_host,
        .port = mqtt_port,
        .username = mqtt_user,
        .password = NULL,
        .client_id = device_id
    };

    client = esp_mqtt_client_init(&mqtt_cfg);
    esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt_event_handler, client);
    esp_mqtt_client_start(client);
}

```

ESP-Marius, 2020-06-30:
Hi @Bruce1221 

Which MQTT server are you connecting to? Do you have the same problem if you try with the MQTT server used in our examples (`mqtt.eclipse.org`)?

0x92 is not a valid header according to the specifications, see [this](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718021), but some MQTT servers are known to not follow these.

See https://github.com/espressif/esp-mqtt/issues/134 for a similar issue where a user reports that the Thingboard MQTT server sends an invalid 0x92 header.

Bruce1221, 2020-07-02:
OK，i will close it.

closed 2020-07-02 by Bruce1221