espressif/esp-idf #5519: Add xTaskGetHandle() support

keelung-yang, 2020-06-30:
_pcTaskGetTaskName()_, `get name by handle`, has been ported from FreeRTOS,
_xTaskGetHandle()_, `get handle by name`, should be ported too.

I know "This function takes a relatively long time to complete", according to FreeRTOS document.

But it's useful in some application scenes, such as start/stop tasks by name in UI:
- No need to maintain relations between name and handle in application code, then keep code concise. 
- No need to care about the performance of xTaskGetHandle(), since start/stop are manipulated from UI by human.

```c
int dmodbus_sim(diag_sim_t * sim)
{
    char task_name[16];
    sprintf(task_name, "dm-sim-%08x", (uint32_t)sim->id);
    TaskHandle_t task_handle = xTaskGetHandle(task_name);
    if (task_handle == NULL) {
        task_handle = xTaskCreate(ftx_task, task_name, 2048, &sim->ftx, tskIDLE_PRIORITY + 1, &task_handle);
        if (task_handle == NULL)
            return MB_DIAG_ERROR_OS;
    }

    switch (sim->cmd) {
    case MB_DIAG_SIM_CMD_START:
        vTaskResume(task_handle);
        break;
    case MB_DIAG_SIM_CMD_STOP:
        vTaskSuspend(task_handle);
        break;
    default:
        return MB_DIAG_ERROR_INVALID_SIM_CMD;
    }

    return MB_DIAG_ERROR_OK;
}
```

projectgus, 2020-07-10:
Hi @keelung-yang ,

Thanks for being patient while someone got back to you.

We would like to add this function and we're looking into it.

To fill you in on the current situation:

Right now we're also in the process of updating [ESP-IDF to FreeRTOS 10](https://github.com/espressif/esp-idf/tree/feature/freertos_upstream_plus_tlsf_allocator_preview) (currently available in this preview branch but will be merged to master soon). So of course the codebase will gain an implementation of this function when we do that. We haven't added tests for enalbing xTaskGetHandle() enabled in the FreeRTOS config with SMP yet, so possibly there will be some additional bugfixes needed before it actually works correctly in SMP modes.

Angus

Dazza0, 2021-09-23:
@keelung-yang ESP-IDF updated to FreeRTOS 10 as of commit a3c90bf59add307d3a41f5fb4c17385ee67dcf20. `xTaskGetHandle()` should be available.

closed 2021-09-23 by Dazza0

Maksons, 2021-11-12:
@Dazza0  I update ESP-IDF to latest master branch but xTaskGetHandle is not available still. 
So question, could solution from ESP forum
`INCLUDE_xTaskGetHandle must be set to 1 in FreeRTOSConfig.h for xTaskGetHandle() to be available.`
be included by default ?
Or could FreeRTOS component have config option for that ?

Dazza0, 2021-11-15:
@Maksons Thanks for reporting this issue. It seems like we've forgotten add some of the new configurations after `FreeRTOS.h` after updating to v10.4.3. I've created https://github.com/espressif/esp-idf/issues/7902 to track this separately