espressif/esp-idf #5520: Example idf_as_lib does not build on windows (mingw64)

jeremyong, 2020-06-30:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Module or chip used:  ESP32-WROOM-32
- IDF version v4.0.1
- Build System:        CMake
- Compiler version 8.2.0
- Operating System:     Windows
- (Windows only) environment type: mingw64
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

There are seemingly an endless stream of issues with attempting to build this project and it's making me wonder if this is meant to build on Windows at all.

### Expected Behavior

The project builds.

### Actual Behavior

1. First, cmake will immediately complain that the compiler cannot build a suitable test program. I managed to bypass this by setting `CMAKE_C_COMPILER_WORKS` and `CMAKE_CXX_COMPILER_WORKS` both to `ON` prior to declaring the project as a workaround, but it doesn't seem ideal.
2. The cmake module paths are not set, and the `include` directives do not find anything in the `$ENV{IDF_PATH}/tools/cmake` folder or the `ENV{IDF_PATH}/tools/cmake/third_party` folder. Appending both to `CMAKE_MODULE_PATH` gets you slightly farther.
3. The variable `${idf_path}` is simply not set correctly within the `build.cmake` file. When logging this values from `project.cmake`, the variable correctly reflects the value of `$ENV{idf_path}`, but somewhere during project initialization, the variable is mangled and just becomes `C` which causes a number of things to break (like finding the kconfig tools for example. Within this file, I simply monkey patched the variable to be `ENV{idf_path}` which continues to work. Frankly, I don't see the point of having a duplicated variable for something that should not change during the configuration period. Furthermore, the scripts set `idf_path` twice for ostensibly no reason.
4. `GetGitRevisionDescription` is simply broken. When trying to configure the inline file, it fails because two paths get mangled together at some point. This really should be done with a version file.
5. Kconfig fails to add the external project mconf-idf. Personally, I don't even care about kconfig and would like a way to disable this altogether. It adds a completely pointless dependency and isn't even a particularly sane way to manage configurations in the first place (at the very least, the toolchain should let the user to decide how to do this, not impose a configuration management solution).
6. There are more errors and my evening is gone so maybe I'll document it later

### Steps to reproduce

1. Try building idf_as_lib (outside the example directory tree) on windows

#### Commentary

I'm admittedly somewhat new to the ESP ecosystem, but I have to say, this is just nuts. It should be just a simple toolchain file, some libraries I need to link, and a few scripts for flashing the board and opening a serial connection. The documentation for this is sparse and it's making my life impossibly difficult, especially when it comes to integrating with a ton of existing C++ code (where the rest of the universe operates using simple, modern cmake). Are there any broad plans to fix your toolchain to something sensible? Sorry if I'm sounding frustrating but after 8 hours of wading through your cmake garbage, I'm getting cold feet.

jeremyong, 2020-06-30:
Also, I should mention quickly that using `idf.py` "works" on my machine so I believe everything is installed properly (files downloaded, dependencies met, submodules pulled, etc). I also reinstalled (twice) just to be sure with both the official installer but also getting each component manually. Nothing worked.

jeremyong, 2020-06-30:
[image]

I'm going to call it quits for now, but functions like this are sort of the issue. This function (along with others) are used pervasively to mutate variables in `PARENT_SCOPE`. :(

jeremyong, 2020-07-01:
I'm going to close, I think the main issue is that MinGW/MSYS doesn't seem supported (a number of assumptions are made about how windows paths work).

closed 2020-07-01 by jeremyong

projectgus, 2020-07-10:
Hi @jeremyong ,

Sorry noone got back to you when you originally opened this.

You're right that we don't test or support the CMake-based build system on MinGW/MSYS (we did support *only* MSYS2 in the old GNU Make based build system, but most Windows developers found this foreign so we migrated away). There are a lot of variables in how MSYS2 is set up around things like CMake generator type, Python type (mingw or msys), Windows vs Unix path formats, etc. that it can get really complex to support.

Providing the IDF_PATH environment variable as a Unix-style path, ie `/c/something` rather than `c:\something` or `c:/something` may help with some of the path parsing issues you mentioned if CMake and/or Python doesn't realise it's running on Windows. However I expect this will not be enough.

Is using either the plain Windows command prompt or WSL an option for you?

Angus