espressif/esp-idf #5521: LwIP and mDNS Issue

mriksman, 2020-06-30:
## Environment

- Build System:         [idf.py]
- Operating System:     [Windows]
- (Windows only) environment type: [PowerShell].
- Using an IDE?: [Yes VSCode]
- Power Supply:         [USB]

## Problem Description

Cannot resolve mDNS hostname with `getaddrinfo` - _if_ the host is an ESP module. It will resolve other devices (like a HP printer).

### Expected Behavior

`getaddrinfo` should resolve mDNS addresses of remote ESP devices.

### Actual Behavior

When I call this;
```c    
struct addrinfo hints, *res;
int status;
memset(&hints, 0, sizeof(hints));
hints.ai_family = AF_INET; 
hints.ai_socktype = SOCK_STREAM;
char *host = "esp-942daf.local";
char *port = "5556";
status = getaddrinfo(host, port, &hints, &res)
```
I get `EAI_FAIL`. 

Looking at WireShark, I am seeing the requests;
`192.168.1.187	224.0.0.251	MDNS	Standard query 0x3090 A esp_942daf.local, "QM" question`
But no response.

I then call immediately after, the following;
```c
struct ip4_addr addr;
addr.addr = 0;
mdns_query_a("esp-942daf", 2000,  &addr);
```
And it correctly resolves the address. Notably, this time the request is of type `QU`;
`192.168.1.187	224.0.0.251	MDNS	Standard query 0x0000 A esp-942daf.local, "QU" question`

I also tried another mDNS hostname; a HP printer with name `HPC8D9D240DB2B.local`, and `getaddrinfo` resolves it correctly, using a `QM` request.

The ESP module will also respond to a Windows `QM` request.

So, why isn't the ESP module responding to a `QM` mDNS query from another ESP module? Why is LwIP and mDNS using different query types? What else can I try?

See the pictures below; I've spearated it into sections;
1. `esp_http_client` fails.
2. `getaddrinfo` to `HPC8D9D240DB2B.local` works
3. `getaddrinfo` to `esp-942daf.local` fails
4. `mdns_query_a` to `esp-942daf` works
5. Ping from Windows 10 PC

[image]

[image]

Alvin1Zhang, 2020-07-01:
Thanks for reporting, we will look into.

mriksman, 2020-07-01:
@Alvin1Zhang 
I took the example from https://github.com/espressif/esp-idf/blob/master/examples/protocols/mdns/main/mdns_example_main.c and modified the code slightly. I placed one copy on an ESP8266 and another on an ESP32, and modified their mDNS names, services, and their hostname query addresses on each (swapped). I get the same issue as discussed before; so the issue exists on both the latest `master` on both SDKs.

```c
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_netif.h"
#include "esp_wifi.h"
#include "mdns.h"
#include "driver/gpio.h"
#include "netdb.h"

#define EXAMPLE_BUTTON_GPIO     0

static const char *TAG = "mdns-test";

static void  query_mdns_host_with_gethostbyname(char * host);
static void  query_mdns_host_with_getaddrinfo(char * host);
static void query_mdns_host(const char * host_name);

static void initialise_mdns(void)
{
    char* hostname = "esp32-mdns";
    //initialize mDNS
    ESP_ERROR_CHECK( mdns_init() );
    //set mDNS hostname (required if you want to advertise services)
    ESP_ERROR_CHECK( mdns_hostname_set(hostname) );
    ESP_LOGI(TAG, "mdns hostname set to: [%s]", hostname);
    //set default mDNS instance name
    ESP_ERROR_CHECK( mdns_instance_name_set("esp32-instance") );

    //structure with TXT records
    mdns_txt_item_t serviceTxtData[3] = {
        {"board","esp32"},
        {"u","user"},
        {"p","password"}
    };

    //initialize service
    ESP_ERROR_CHECK( mdns_service_add("ESP32-WebServer", "_http", "_tcp", 80, serviceTxtData, 3) );
    //add another TXT item
    ESP_ERROR_CHECK( mdns_service_txt_item_set("_http", "_tcp", "path", "/foobar") );
    //change TXT item value
    ESP_ERROR_CHECK( mdns_service_txt_item_set("_http", "_tcp", "u", "admin") );

}

static void initialise_button(void)
{
    gpio_config_t io_conf = {0};
    #ifdef CONFIG_IDF_TARGET_ESP32
        io_conf.intr_type = GPIO_PIN_INTR_DISABLE;
        io_conf.pin_bit_mask = BIT64(EXAMPLE_BUTTON_GPIO);
    #elif CONFIG_IDF_TARGET_ESP8266
        io_conf.intr_type = GPIO_INTR_DISABLE;
        io_conf.pin_bit_mask = BIT(EXAMPLE_BUTTON_GPIO);
    #endif
    
    io_conf.mode = GPIO_MODE_INPUT;
    io_conf.pull_up_en = 1;
    io_conf.pull_down_en = 0;
    gpio_config(&io_conf);
}

static void check_button(void)
[... 134 lines cut ...]

    memset(&hints, 0, sizeof(hints));
    hints.ai_family = AF_UNSPEC;
    hints.ai_socktype = SOCK_STREAM;

    int err = getaddrinfo(host, NULL, &hints, &res);

    if(err){
        ESP_LOGE(TAG, "getaddrinfo Query Failed: %d", err);
        return;
    }

    while (res) {
        ESP_LOGI(TAG, "getaddrinfo: %s resolved to: %s", host,
                    res->ai_family == AF_INET?
                    inet_ntoa(((struct sockaddr_in *) res->ai_addr)->sin_addr):
                    inet_ntoa(((struct sockaddr_in6 *) res->ai_addr)->sin6_addr));
        res = res->ai_next;
    }
}
```

[image]

[image]

mriksman, 2020-07-01:
I notice this on the ESP32 debug logs, when the ESP8266 tries to use `getaddrinfo` and `gethostbyname` to resolve the ESP32 hostname. So the ESP32 _is_ receiving the queries;
```
D (837384) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbff94
D (838414) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbff94
D (839334) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbff94
D (841384) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbff94
```
The 4x entries corresponds to the 4 attempted queries.

mriksman, 2020-07-01:
I turned on debugging, and saw the following. The first one is `mdns_query_a`, which works, and you can see the `Q: *U*`.
```
RX[0][0]: From: 172.20.10.9:5353, To: 224.0.0.251, Packet[18590]: 
    Q: *U* esp32-mdns.local. A IN
D (19447) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffc0094

TX[0][0]: To: 172.20.10.9:5353, Packet[18600]: AUTHORITATIVE
    A: esp32-mdns.local. A IN FLUSH 120 [4] 172.20.10.3
```

Then `gethostbyname` which fails (it retries 4 times)
```
RX[0][0]: From: 172.20.10.9:21213, To: 224.0.0.251, Packet[18650]:  0x0100
    Q: esp32-mdns.local. A IN
D (19507) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffc0094

TX[0][0]: To: 172.20.10.9:21213, Packet[18660]: AUTHORITATIVE
    A: esp32-mdns.local. A IN 120 [4] 172.20.10.3

RX[0][0]: From: 172.20.10.9:21213, To: 224.0.0.251, Packet[19210]:  0x0100
    Q: esp32-mdns.local. A IN
D (20067) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffc0094

TX[0][0]: To: 172.20.10.9:21213, Packet[19210]: AUTHORITATIVE
    A: esp32-mdns.local. A IN 120 [4] 172.20.10.3

RX[0][0]: From: 172.20.10.9:21213, To: 224.0.0.251, Packet[20230]:  0x0100
    Q: esp32-mdns.local. A IN
D (21087) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffc0094

TX[0][0]: To: 172.20.10.9:21213, Packet[20240]: AUTHORITATIVE
    A: esp32-mdns.local. A IN 120 [4] 172.20.10.3

RX[0][0]: From: 172.20.10.9:21213, To: 224.0.0.251, Packet[22280]:  0x0100
    Q: esp32-mdns.local. A IN
D (23137) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffc0094

TX[0][0]: To: 172.20.10.9:21213, Packet[22290]: AUTHORITATIVE
    A: esp32-mdns.local. A IN 120 [4] 172.20.10.3
```

Then `getaddrinfo` which fails (it retries 4 times)
```
RX[0][0]: From: 172.20.10.9:2985, To: 224.0.0.251, Packet[25250]:  0x0100
    Q: esp32-mdns.local. A IN
D (26107) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffc0094

TX[0][0]: To: 172.20.10.9:2985, Packet[25250]: AUTHORITATIVE
    A: esp32-mdns.local. A IN 120 [4] 172.20.10.3

RX[0][0]: From: 172.20.10.9:2985, To: 224.0.0.251, Packet[26270]:  0x0100
    Q: esp32-mdns.local. A IN
D (27127) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffc0094

TX[0][0]: To: 172.20.10.9:2985, Packet[26280]: AUTHORITATIVE
    A: esp32-mdns.local. A IN 120 [4] 172.20.10.3

RX[0][0]: From: 172.20.10.9:2985, To: 224.0.0.251, Packet[27190]:  0x0100
    Q: esp32-mdns.local. A IN
D (28047) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffc0094

TX[0][0]: To: 172.20.10.9:2985, Packet[27200]: AUTHORITATIVE
    A: esp32-mdns.local. A IN 120 [4] 172.20.10.3

RX[0][0]: From: 172.20.10.9:2985, To: 224.0.0.251, Packet[29240]:  0x0100
    Q: esp32-mdns.local. A IN
D (30097) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffc0094

TX[0][0]: To: 172.20.10.9:2985, Packet[29250]: AUTHORITATIVE
    A: esp32-mdns.local. A IN 120 [4] 172.20.10.3
```

david-cermak, 2020-07-02:
Hi @mriksman 

Thanks for the detailed report and your investigation. There is indeed an issue with resolving mDNS names in IDF. And yes, we use two mdns libraries here, `lwip`'s mdns for the `getaddrinfo()` and similar, and our own implementation in `components/mdns` for advertising services (and explicit name/service resolving).

The problem is not in the port number nor in the unicast field. The issue is that the answers generated in `components/mdns` do not contain question section per [RFC6762-chapter6](https://tools.ietf.org/html/rfc6762#page-14):
```
Multicast DNS responses MUST NOT contain any questions
```
But the lwip resolver checks if the received dns response is valid in https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/core/dns.c#L1249 so expects non-empty question field.

Interestingly most of the mdns devices reply with repeated question field which (again according to the RFC6762) must be silently ignored. That's why the lwip resolver works okay with other devices, such as printers.

Thank you for making us aware of this problem. Will have to fix it somehow. As a workaround might I suggest using the internal mdns library and resolve ESP devices using  the `mdns_query_a()` API, in the meantime?

mriksman, 2020-07-05:
@david-cermak Thanks. For now, I'll do as you propose. 
I assume this will be fixed in `esp-lwip` and thus be available to ESP8266 RTOS SDK as well?

Pro, 2020-08-18:
I also ran into this issue.

It can even be reproduced on one single device with the following steps:

1. Start mdns and use `mdns_hostname_set` to set to a specific hostname, e.g. "MyDevice.local"
2. Start all the wifi stuff
3. Use getaddrinfo with "MyDevice.local" which should just return the local ip address, but it fails with `202`. I know that you can also just pass NULL for the own hostname, but still this works on other devices, e.g., windows or linux

closed 2020-09-16 by espressif-bot

mriksman, 2022-01-19:
@david-cermak It's been a while since I looked at this issue (and programmed in general). I am curious; are there any benefits using `mdns_query_a` as opposed to `getaddrinfo` or `gethostbyname`?

david-cermak, 2022-01-19:
Short answer is that the mdns API implements full featured mDNS querier while the lwip just sends single-short queries (https://datatracker.ietf.org/doc/html/rfc6762#section-5.1).
The very same thing as when querying using `avahi` on linux, versus running `dig @224.0.0.251`