espressif/esp-idf #5522: SPIFFS Partition Could not be found

mntolia, 2020-06-30:
Environment
Development Kit: ESP32-Wrover-Kit
Kit version (for WroverKit/PicoKit/DevKitC): v4
Module or chip used: ESP32-WROOM-32
IDF version : 4
Build System: CMake
Operating System: Linux
Using an IDE?: PlatformIO on Visual Studio Code
Power Supply: USB

I am using Arduino as an ESP-IDF Component

## Problem Description

I am getting an error 261
When I initiate SPIFFS
```
  Serial.println("Setting Up Flash Memory Reader");
  if (!SPIFFS.begin(true))
  {
    Serial.println("An Error has occurred while mounting SPIFFS");
    return;
  }
```

### Steps to reproduce

1. I uploaded the program with Platformio
2. I uploaded the SPIFFS data via Arduino. 

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void setup()
{
  Serial.println("Setting Up Flash Memory Reader");
  if (!SPIFFS.begin(true))
  {
    Serial.println("An Error has occurred while mounting SPIFFS");
    return;
  }
}
```
void setup()
{
  Serial.begin(115200);
  Serial.println("Default SSID" + DEFAULT_SSID);
  Serial.println("===================Setting Up Start=======================");

  //Relay Setup
  Serial.println("Setting Up Relay");
  pinMode(relay_pin, OUTPUT);

  //Rtc
  Rtc.Begin();
  Serial.println(printDateTime(Rtc.GetDateTime()));
  //Serial.println("Printing Current Date & Time From DS3231");
  //printDateTime(Rtc.GetDateTime());

  //preferences settings
  fmem.begin(flash_namespace);
  //fmem.clear();
  uint8_t size_of_one_time = fmem.getBytesLength(onetime_keystore_key);
  uint8_t size_of_reccurring = fmem.getBytesLength(recurring_keystore_key);
  Serial.println("Current Onetime Key Bytes Length");
  Serial.println(size_of_one_time);
  Serial.println("Current Recurring Key Bytes Length");
  Serial.println(fmem.getBytesLength(recurring_keystore_key));
  Serial.println("Current Relay State");
  Serial.println(fmem.getUInt(relay_state_key, 0));
  //Switch Relay To that state
  switch_relay(fmem.getUInt(relay_state_key, 0));

  if (size_of_one_time == 0)
  {
    onetime = false;
  }
  if (size_of_reccurring == 0)
  {
    recurring = false;
  }

  load_keystore();
  Serial.printf("\nOne Time Keystore Size: %d", onetime_keystore.size());
  Serial.printf("\nRecurring Time Keystore Size: %d", recurring_keystore.size());

  for (String i : recurring_keystore)
  {
    Serial.println(i);
  }

  //LED Testing
  Serial.println("Setting Up LED");
  pinMode(led_pin, OUTPUT);

  Serial.println("Setting Up Hotspot");
  setup_hotspot();

  Serial.println("Setting Up Flash Memory Reader");
  if (!SPIFFS.begin(true))
  {
    Serial.println("An Error has occurred while mounting SPIFFS");
    return;
  }

  Serial.println("Running Switch Reset in Core 0");

  //Running Timer Checker on Core 0
  xTaskCreatePinnedToCore(time_checker_task, "Time Chcker", 10000, NULL, 0, &Task1, 0);
  Serial.println("===================Setting Up Stop=======================");
}

## Debug Logs

https://gist.github.com/mntolia/d709bccaba7974cd8670f983c5a0726a

## Other items if possible

- [ ] SDKconfig file: https://gist.github.com/mntolia/0fb93ca96206ff5c7fee349f1beedcfd

chegewara, 2020-06-30:
How is your partitions.csv looks like?
Do you have `board_build.partitions = partitions.csv` n platformio.ini file?

mntolia, 2020-06-30:
> How is your partitions.csv looks like?
> Do you have `board_build.partitions = partitions.csv` n platformio.ini file?

You are right.
Darn it. I put one less 0 in the spiff's partition. Sorry for the trouble.

Alvin1Zhang, 2020-07-01:
@chegewara Thanks for the notes. @mntolia Thanks for reporting, feel free to reopen if the issue still happens. Thanks.

closed 2020-07-01 by Alvin1Zhang