espressif/esp-idf #5524: ESP32 Incorrect timing on the last few items when sending more than certain rmt_item32 at a time through RMT

weixiongmei, 2020-07-01:
INSTRUCTIONS
============
```C
#define RMT_TX_CHANNEL RMT_CHANNEL_0
#define CONFIG_EXAMPLE_RMT_TX_GPIO 15

#define RATIO (float)(40000000 / 1e9)
#define WS2812_T0H_NS (uint32_t)(350 * RATIO)
#define WS2812_T0L_NS (uint32_t)(850 * RATIO)
#define WS2812_T1H_NS (uint32_t)(600 * RATIO)
#define WS2812_T1L_NS (uint32_t)(600 * RATIO)
#define WS2812_RESET_US (uint32_t)(80 * 1000 * RATIO)

static const rmt_item32_t morse_on[] = {
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},

    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},

    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
};

static const rmt_item32_t morse_off[] = {
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
};
/*
 * Initialize the RMT Tx channel
 */
static void rmt_tx_init(void)
{
  config.tx_config.loop_en = false;
  config.rmt_mode = RMT_MODE_TX;
  config.tx_config.carrier_en = false;
  config.tx_config.idle_level = RMT_IDLE_LEVEL_LOW;
  config.tx_config.idle_output_en = false;
  config.tx_config.carrier_duty_percent = 50;
  config.clk_div = 2;

  ESP_ERROR_CHECK(rmt_config(&config));
  ESP_ERROR_CHECK(rmt_driver_install(config.channel, 0, 0));
}
void app_main()
{
  rmt_tx_init();
  uint8_t size = 24; //When sending less than 20 items the timing has no problem.
  while (true)
  {
   rmt_write_items(RMT_TX_CHANNEL, morse_on, size, true);
    vTaskDelay(100 / portTICK_PERIOD_MS);
  }
```

## Environment
IDF: v4.2-dev-1905-g625bd5eb1
Toolchain version: esp-2020r2
Compiler version: 8.2.0

- Development Kit:      ESP32-Wrover-Kit
- Build System:         Make
- Operating System:     macOS
- Power Supply:         USB

## Problem Description
When I tried to send more than 20 items by using esp_err_trmt_write_items, the timing of the last few rmt_item32 being incorrect. 

## Screenshots Sending Less Than 20 Items
[image]
## Screenshots Sending Over 20 Items
[image]
[image]

Alvin1Zhang, 2020-07-01:
Thanks for reporting, we will look into.

wilhelmw201, 2020-07-02:
Hi, weixiongmei,

Thank you for your detailed description, however I was unable to reproduce the problem with your code. Here is the code that I used:
`
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "driver/rmt.h"

#define RMT_TX_CHANNEL RMT_CHANNEL_0

#define RATIO (float)(40000000 / 1e9)
#define WS2812_T0H_NS (uint32_t)(350 * RATIO)
#define WS2812_T0L_NS (uint32_t)(850 * RATIO)
#define WS2812_T1H_NS (uint32_t)(600 * RATIO)
#define WS2812_T1L_NS (uint32_t)(600 * RATIO)
#define WS2812_RESET_US (uint32_t)(80 * 1000 * RATIO)

static const rmt_item32_t morse_on[] = {
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},

    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},

    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
    {{{WS2812_T0H_NS, 1, WS2812_T0L_NS, 0}}},
};

static void rmt_tx_init(void)
{
  rmt_config_t config;
  config.channel = 0;
  config.mem_block_num = 1;
  config.gpio_num = 15;

  config.tx_config.loop_en = false;
  config.rmt_mode = RMT_MODE_TX;
  config.tx_config.carrier_en = false;
  config.tx_config.idle_level = RMT_IDLE_LEVEL_LOW;
  config.tx_config.idle_output_en = false;

  config.clk_div = 2;

  ESP_ERROR_CHECK(rmt_config(&config));
  ESP_ERROR_CHECK(rmt_driver_install(config.channel, 0, 0));
  
}
void app_main()
{
  rmt_tx_init();
  uint8_t size = 24; //When sending less than 20 items the timing has no problem.
  while (true)
  {
   rmt_write_items(RMT_TX_CHANNEL, morse_on, size, true);
    vTaskDelay(1000 / portTICK_PERIOD_MS);
  }
}

`

I was able to get a stable signal of 360ns high - 840 ns low on my computer with a esp32wroverb. Could you please test with the above program?

weixiongmei, 2020-07-02:
@wilhelmw201 Hi, I tried your code, the timing of the last few items still incorrect, is your external memory enabled? I upload the my sdkconfig file to see is it causing the problem or not. Thanks~~

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/4861785/sdkconfig.zip)

wilhelmw201, 2020-07-02:
Hi weixiongmei,

thank you for your response. The problem is caused by the power management unit. For now, please disable PM, or use a power management lock to keep the PM from interfering with the APB clock when RMT is not complete. 

An example of using the power management lock:
```
esp_pm_lock_handle_t hd;
esp_pm_lock_create(ESP_PM_APB_FREQ_MAX, 0, "my_rmt_apb_lock", &hd);
```
then,
```
esp_pm_lock_acquire(hd);
rmt_write_items(RMT_TX_CHANNEL, my_data, size, true); // this function needs to block, or sleep for some ticks or something
esp_pm_lock_release(hd);
```

weixiongmei, 2020-07-02:
@wilhelmw201, Thank you so much, it's fixed~~

closed 2022-11-16 by Alvin1Zhang