espressif/esp-idf #5527: Inconsistent CMake behaviour

ammaree, 2020-07-01:
## Environment

- IDF version (run ``git describe --tags`` to find it): v4.2-dev-1905-g625bd5eb1
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Eclipse]
- Power Supply:         [external 5V]

## Problem Description
In order to include an accurate epoch based BUILD timestamp within each executable we create a timestamp.h file from the main project CMakeLists.txt file using the following code.
'
####################### Using UnxUtils date() get Linux Epoch datetime stamp #######################
set( DATE_CMD C:/Dropbox/devs/bin/UnxUtils/usr/local/wbin/date.exe )
set( DATE_PAR +%s )
execute_process( COMMAND  ${DATE_CMD}  ${DATE_PAR}  RESULT_VARIABLE  DATE_RES  OUTPUT_VARIABLE  DATE_OUT  OUTPUT_STRIP_TRAILING_WHITESPACE )
message( "Date Result     : " ${DATE_RES} )
message( "Build Timestamp : " ${DATE_OUT} )

####################### Echo the datetime stamp to project timestamp.h file ########################
set( ECHO_CMD C:/Dropbox/devs/bin/UnxUtils/usr/local/wbin/echo.exe )
set( ECHO_PAR "#define     APP_TIMESTAMP      ${DATE_OUT}" )
set( ECHO_OUT  ${PA}/main/timestamp.h )
execute_process( COMMAND  ${ECHO_CMD}  ${ECHO_PAR}  RESULT_VARIABLE  ECHO_RES  OUTPUT_FILE  ${ECHO_OUT}  OUTPUT_STRIP_TRAILING_WHITESPACE )
message( "Echo Result     : " ${ECHO_RES} )
message( "Timestamp File  : " ${ECHO_OUT} )
'

The format of the timestamp.h file is a single macro definition as follows:
'#define     APP_TIMESTAMP      1593606089'

### Expected Behavior
Every time idf.py is run the value of the macro in timestamp.h should be updated.
Afterwardsthe single source file where the macro is used should be recompiled and the binary recreated by the linker.
 
### Actual Behavior
A. If idf.py is run for the first time, the timestamp.h is created with the correct timestamp value and BIN is built.
B. If idf.py is rerun some time later, without changing anything, timestamp.h is NOT update and nothing is [re]built.
C. If the source file (where the value of the macro is used) is changed, timestamp.h is NOT update and nothing is [re]built.
D. If the main project CMakeLists.txt is changed, even just a space added, timestamp.h IS updated and .BIN is [re]built.
E.  If the component (where the value of the macro is used) CMakeLists.txt is changed, even just a space added, timestamp.h IS updated and .BIN is [re]built.

Surely in cases B and C the main CMakeLists.txt file should be "evaluated" resulting in
a) new epoch value
b) new (updated?) timestamp.h
c) recompile of the application.c where the macro value is used
d) a new .BIN as a result of the new libmain.a?

igrr, 2020-07-01:
Hi @ammaree, builds using CMake build system generally have two phases. First phase, "configuration", is when CMakeLists.txt files are evaluated, and build scripts are generated. At the second phase, build scripts are executed (usually by Make or Ninja).

The `execute_process` CMake function only runs at CMakeLists evaluation time, i.e. at the configuration stage. This is inconveniently mentioned in the last sentence of this CMake manual page: https://cmake.org/cmake/help/latest/command/execute_process.html:

> Both commands run while CMake is processing the project prior to build system generation. Use add_custom_target() and add_custom_command() to create custom commands that run at build time.

The reason why timestamp.h is updated when you modify any CMakeLists.txt file is that by default, CMake adds all CMakeLists.txt files as dependencies for "reconfiguration". That is, if any of the CMakeLists.txt files are modified, the target build system (Ninja or CMake) will re-run CMake configuration process to update the build scripts. It is important to understand that CMake doesn't evaluate CMakeLists files every time when you do `idf.py build` (or `cd build; cmake --build .`) Normally `idf.py build` will only run `make` or `ninja`, i.e. the build tool which CMake has generated the build script for. This, by the way, is the reason why it is not recommended to use globbing in CMakeLists files ([docs link](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#file-globbing-incremental-builds)).

So what you need is to have a step that is executed at build time, not at CMake configuration time:

1. use `add_custom_target` and `add_custom_command` functions to generate the header file
2. tell CMake that the header file is generated using `set_source_files_properties`
3. tell CMake which files depend on the header using `add_dependencies`

For an existing example, you can have a look at this answer to a similar question: https://stackoverflow.com/a/3794650.

ammaree, 2020-07-01:
@igrr thanks for the quick feedback. 

Just to clarify, using the link you referenced and considering that the "timestamp.h" file is included from my "${PROJECT}/main/application.c" file and located in the same "${PROJECT}/main/" directory.....

a) The functionality of the getsvn.cmake section would be replaced  in a "get_timestamp.cmake" file containing the 2 sections of code under my "Problem Description" above?

b) The "get_timestamp.cmake" file would be located in the same directory with the "application.c" and "timestamp.h" files?

c) The functionality of the CmakeLists.txt section from the link referenced would be replaced with adapted logic in which of the existing CMakeLists.txt files, the top level "${PROJECT}/" or "${PROJECT}/main/" version?

d) If in the top level CMakeLists.txt file, where should it be placed relative to the lines..

> include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(${PROJECT_NAME})

ammaree, 2020-07-01:
@igrr have tried the following.

Created following "get_timestamp.cmake" file in "${PROJECT}/main" directory

> ####################### Using UnxUtils date() get Linux Epoch datetime stamp #######################
set( DATE_CMD C:/Dropbox/devs/bin/UnxUtils/usr/local/wbin/date.exe )
set( DATE_PAR +%s )
execute_process( COMMAND  ${DATE_CMD}  ${DATE_PAR}  RESULT_VARIABLE  DATE_RES  OUTPUT_VARIABLE  DATE_OUT  OUTPUT_STRIP_TRAILING_WHITESPACE )
####################### Echo the datetime stamp to project timestamp.h file ########################
set( ECHO_CMD C:/Dropbox/devs/bin/UnxUtils/usr/local/wbin/echo.exe )
set( ECHO_PAR "#define     APP_TIMESTAMP      ${DATE_OUT}" )
set( ECHO_OUT  ${PA}/main/timestamp.h )
execute_process( COMMAND  ${ECHO_CMD}  ${ECHO_PAR}  RESULT_VARIABLE  ECHO_RES  OUTPUT_FILE  ${ECHO_OUT}  OUTPUT_STRIP_TRAILING_WHITESPACE )
message( "################################" )
message( "Date Result     : " ${DATE_RES} )
message( "Build Timestamp : " ${DATE_OUT} )
message( "Echo Result     : " ${ECHO_RES} )
message( "Timestamp File  : " ${ECHO_OUT} )
message( "################################" )

Added following lines first above and then below the 2 lines quoted in my previous post:
>ADD_CUSTOM_TARGET(timestamp_tag ALL)
ADD_CUSTOM_COMMAND(TARGET timestamp_tag COMMAND ${CMAKE_COMMAND} -DSOURCE_DIR=${CMAKE_CURRENT_SOURCE_DIR} -P ${CMAKE_CURRENT_SOURCE_DIR}/get_timestamp.cmake)
add_dependencies(main timestamp_tag)

Returns the following error:
> Cannot add target-level dependencies to non-existent target "main".
  The add_dependencies works for top-level logical targets created by the
  add_executable, add_library, or add_custom_target commands.  If you want to
  add file-level dependencies see the DEPENDS option of the add_custom_target
  and add_custom_command commands.

Also tried to replace the "main" target with the "${PROJECT_NAME}" but same problem, just the name of the non-existent target changes

Any suggestions/corrections?

igrr, 2020-07-01:
Could you please try https://github.com/igrr/idf-cmake-timestamp-demo and see if it works for you?

ammaree, 2020-07-01:
@igrr 
It works perfectly, you are a star. Thanks a lot.

closed 2020-07-01 by ammaree