espressif/esp-idf #5528: lan8720_reset_hw() out of spec, doesn't pull low for >100us

brianbienvenu, 2020-07-01:
`lan8720_reset_hw()` doesn't contain any delay when pulsing the reset pin low. In the real world, this makes a ~250ns wide negative pulse. The chipin production **LAN8720A** datasheet asks for t_rstia >100us:
[image]

The datasheet for the EOL SMSC LAN8720 also specified `T11.1 Reset Pulse Width >100us`.

I first noticed this issue in v4.0.1, it's unchanged in master today:
https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth_phy_lan8720.c#L262

I added a busy wait `ets_delay_us(100);` to resolve this issue:
```
static esp_err_t lan8720_reset_hw(esp_eth_phy_t *phy)
{
    phy_lan8720_t *lan8720 = __containerof(phy, phy_lan8720_t, parent);
    if (lan8720->reset_gpio_num >= 0) {
        gpio_pad_select_gpio(lan8720->reset_gpio_num);
        gpio_set_direction(lan8720->reset_gpio_num, GPIO_MODE_OUTPUT);
        gpio_set_level(lan8720->reset_gpio_num, 0);
        ets_delay_us(100);
        gpio_set_level(lan8720->reset_gpio_num, 1);
    }
    return ESP_OK;
}
```

suda-morris, 2020-07-02:
@brianbienvenu Well spotted! Will fix it ASAP.

Alvin1Zhang, 2020-07-30:
Thanks for reporting and sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/5a99038b1af7dc97ff1d23027d9763fdd882784b. Feel free to reopen if the issue still happens.

closed 2020-07-30 by Alvin1Zhang