espressif/esp-idf #5531: xQueueGenericReceive failed at /esp-idf/components/freertos/queue.c:1461

devUsama, 2020-07-02:
## Environment

- Development Kit : custom
- Kit version 
- Module or chip used : ESP32-WROOM-32D
- IDF version : v4.1-beta2-142-g6c5f19c26
- Build System : idf.py
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: No
- Power Supply: Battery

## Problem Description

**xQueueGenericReceive Failed at  C:/esp-idf-v4.1/esp-idf/components/freertos/queue.c:1461.

I (5120) MQTT: Other event id:7
Guru Meditation Error: Core  1 panic'ed (LoadStoreAlignment). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008f4d2  PS      : 0x00060433  A0      : 0x8008eb33  A1      : 0x3ffe3b70
0x4008f4d2: uxPortCompareSet at C:/esp-idf-v4.1/esp-idf/components/freertos/include/freertos/portmacro.h:357
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at C:/esp-idf-v4.1/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at C:/esp-idf-v4.1/esp-idf/components/freertos/portmux_impl.h:100
 (inlined by) vTaskEnterCritical at C:/esp-idf-v4.1/esp-idf/components/freertos/tasks.c:4201

A2      : 0x6c776f8b  A3      : 0x00001000  A4      : 0x3ffe3bb0  A5      : 0x0000000c
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x0000abab  A9      : 0x0000abab
A10     : 0xb33fffff  A11     : 0x0000cdcd  A12     : 0x00060420  A13     : 0x00000001
A14     : 0x00060420  A15     : 0x3ffe3a40  SAR     : 0x00000004  EXCCAUSE: 0x00000009
EXCVADDR: 0x6c776f8b  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe

ELF file SHA256: 584596b26a336667

Backtrace: 0x4008f4cf:0x3ffe3b70 0x4008eb30:0x3ffe3b90 0x400ddd96:0x3ffe3bd0 0x400d9953:0x3ffe3bf0 0x4010374a:0x3ffe3c20 0x40103791:0x3ffe3c40 0x40103f14:0x3ffe3c60 0x4008e075:0x3ffe3ca0
0x4008f4cf: uxPortCompareSet at C:/esp-idf-v4.1/esp-idf/components/freertos/include/freertos/portmacro.h:357
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at C:/esp-idf-v4.1/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at C:/esp-idf-v4.1/esp-idf/components/freertos/portmux_impl.h:100
 (inlined by) vTaskEnterCritical at C:/esp-idf-v4.1/esp-idf/components/freertos/tasks.c:4201

0x4008eb30: xQueueGenericReceive at C:/esp-idf-v4.1/esp-idf/components/freertos/queue.c:1461 (discriminator 4)

0x400ddd96: MY_ERROR_HANDLING_freeheep at c:\esp-idf-v4.1\tank_motor_v4.1\build/../components/iot-core/src/errorHandling.c:339

0x400d9953: mqtt_event_handler at c:\esp-idf-v4.1\tank_motor_v4.1\build/../main/app_main.c:1371

0x4010374a: esp_mqtt_dispatch_event at C:/esp-idf-v4.1/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:812

0x40103791: esp_mqtt_dispatch_event_with_msgid at C:/esp-idf-v4.1/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:803

0x40103f14: esp_mqtt_task at C:/esp-idf-v4.1/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1226

0x4008e075: vPortTaskWrapper at C:/esp-idf-v4.1/esp-idf/components/freertos/port.c:143**

### Expected Behavior
xQueueGenericReceive at esp-idf/components/freertos/queue.c:1461 should not failed!

### Actual Behavior
If I erase controller flash by using **idf.py erase_flash** and flash the firmware again issue get resolved and at first it get worked, but afterwards if i reset controller the same issue get back.

Alvin1Zhang, 2020-07-02:
Thanks for reporting, we will look into.

igrr, 2020-07-02:
Could you please try enabling heap poisoning? This looks like a heap memory corruption issue. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#configuration

devUsama, 2020-07-02:
It seems like i got this issue in esp_mqtt_client_start() function.

**Heap Trace Logs around esp_mqtt_client_start() function**

41 allocations trace (100 entry buffer)
240 bytes (@ 0x3ffd0d4c) allocated CPU 0 ccount 0x16a70e98 caller 0x40100e3a:0x400d74fe
0x40100e3a: esp_mqtt_client_init at C:/esp-idf-v4.1/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:589

0x400d74fe: mqtt_app_start at c:\esp-idf-v4.1\tank_motor_v4.1\build/../main/app_main.c:1420

20 bytes (@ 0x3ffb7a6c) allocated CPU 0 ccount 0x16a72240 caller 0x40100e79:0x400d74fe
0x40100e79: esp_mqtt_client_init at C:/esp-idf-v4.1/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:591 (discriminator 2)

0x400d74fe: mqtt_app_start at c:\esp-idf-v4.1\tank_motor_v4.1\build/../main/app_main.c:1420

80 bytes (@ 0x3ffd0e50) allocated CPU 0 ccount 0x16a72c74 caller 0x4008e866:0x4008eabb
0x4008e866: xQueueGenericCreate at C:/esp-idf-v4.1/esp-idf/components/freertos/queue.c:391

0x4008eabb: xQueueCreateMutex at C:/esp-idf-v4.1/esp-idf/components/freertos/queue.c:498

12 bytes (@ 0x3ffba298) allocated CPU 0 ccount 0x16a74138 caller 0x4012321a:0x40100eb8
0x4012321a: esp_transport_list_init at C:/esp-idf-v4.1/esp-idf/components/tcp_transport/transport.c:75

0x40100eb8: esp_mqtt_client_init at C:/esp-idf-v4.1/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:604

12 bytes (@ 0x3ffbd874) allocated CPU 0 ccount 0x16a74ce4 caller 0x4012325c:0x40100eb8
0x4012325c: esp_transport_list_init at C:/esp-idf-v4.1/esp-idf/components/tcp_transport/transport.c:78 (discriminator 2)

0x40100eb8: esp_mqtt_client_init at C:/esp-idf-v4.1/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:604

64 bytes (@ 0x3ffbdad8) allocated CPU 0 ccount 0x16a75a10 caller 0x40123326:0x40123d52
0x40123326: esp_transport_init at C:/esp-idf-v4.1/esp-idf/components/tcp_transport/transport.c:136

0x40123d52: esp_transport_tcp_init at C:/esp-idf-v4.1/esp-idf/components/tcp_transport/transport_tcp.c:183

4 bytes (@ 0x3ffb7a94) allocated CPU 0 ccount 0x16a76824 caller 0x40123d5e:0x40100eef
0x40123d5e: esp_transport_tcp_init at C:/esp-idf-v4.1/esp-idf/components/tcp_transport/transport_tcp.c:184

0x40100eef: esp_mqtt_client_init at C:/esp-idf-v4.1/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:607 (discriminator 2)

5 bytes (@ 0x3ffba708) allocated CPU 0 ccount 0x16a778c8 caller 0x40123290:0x40100f39
0x40123290: esp_transport_list_add at C:/esp-idf-v4.1/esp-idf/components/tcp_transport/transport.c:87

0x40100f39: esp_mqtt_client_init at C:/esp-idf-v4.1/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:613 (discriminator 2)

64 bytes (@ 0x3ffd0eb4) allocated CPU 0 ccount 0x16a78528 caller 0x40123326:0x401245e8
0x40123326: esp_transport_init at C:/esp-idf-v4.1/esp-idf/components/tcp_transport/transport.c:136

0x401245e8: esp_transport_ws_init at C:/esp-idf-v4.1/esp-idf/components/tcp_transport/transport_ws.c:454

40 bytes (@ 0x3ffd0f08) allocated CPU 0 ccount 0x16a79444 caller 0x401245f6:0x40100f3e
0x401245f6: esp_transport_ws_init at C:/esp-idf-v4.1/esp-idf/components/tcp_transport/transport_ws.c:455

0x40100f3e: esp_mqtt_client_init at C:/esp-idf-v4.1/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:613 (discriminator 2)

2 bytes (@ 0x3ffbd894) allocated CPU 0 ccount 0x16a7a438 caller 0x40094b20:0x4000bbc5
0x40094b20: _malloc_r at C:/esp-idf-v4.1/esp-idf/components/newlib/heap.c:53

1024 bytes (@ 0x3ffd0f44) allocated CPU 0 ccount 0x16a7b128 caller 0x4012467a:0x40100f3e
0x4012467a: esp_transport_ws_init at C:/esp-idf-v4.1/esp-idf/components/tcp_transport/transport_ws.c:464 (discriminator 2)

0x40100f3e: esp_mqtt_client_init at C:/esp-idf-v4.1/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:613 (discriminator 2)

5 bytes (@ 0x3ffd1358) allocated CPU 0 ccount 0x16a7daac caller 0x40094b20:0x4000bbc5
0x40094b20: _malloc_r at C:/esp-idf-v4.1/esp-idf/components/newlib/heap.c:53

3 bytes (@ 0x3ffd1374) allocated CPU 0 ccount 0x16a7e940 caller 0x40123290:0x40100f8c
0x40123290: esp_transport_list_add at C:/esp-idf-v4.1/esp-idf/components/tcp_transport/transport.c:87

0x40100f8c: esp_mqtt_client_init at C:/esp-idf-v4.1/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:623 (discriminator 2)
[... 134 lines cut ...]

Backtrace: 0x4008d6c9:0x3ffd78e0 0x4008da5d:0x3ffd7900 0x401695c3:0x3ffd7920 0x4010c877:0x3ffd7950 0x4011d2d5:0x3ffd7980 0x401675b9:0x3ffd79c0 0x40123c51:0x3ffd7a00 0x40123cb5:0x3ffd7a20 0x4017feb1:0x3ffd7a60 0x4010009e:0x3ffd7a80 0x4008e249:0x3ffd7ac0
0x4008d6c9: invoke_abort at C:/esp-idf-v4.1/esp-idf/components/esp32/panic.c:157

0x4008da5d: abort at C:/esp-idf-v4.1/esp-idf/components/esp32/panic.c:174

0x401695c3: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x4010c877: tcpip_send_msg_wait_sem at C:/esp-idf-v4.1/esp-idf/components/lwip/lwip/src/api/tcpip.c:455 (discriminator 1)

0x4011d2d5: netconn_gethostbyname_addrtype at C:/esp-idf-v4.1/esp-idf/components/lwip/lwip/src/api/api_lib.c:1338 (discriminator 2)

0x401675b9: lwip_gethostbyname at C:/esp-idf-v4.1/esp-idf/components/lwip/lwip/src/api/netdb.c:100

0x40123c51: resolve_dns at C:/esp-idf-v4.1/esp-idf/components/tcp_transport/transport_tcp.c:39

0x40123cb5: tcp_connect at C:/esp-idf-v4.1/esp-idf/components/tcp_transport/transport_tcp.c:62

0x4017feb1: esp_transport_connect at C:/esp-idf-v4.1/esp-idf/components/tcp_transport/transport.c:165

0x4010009e: esp_mqtt_task at C:/esp-idf-v4.1/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1237

0x4008e249: vPortTaskWrapper at C:/esp-idf-v4.1/esp-idf/components/freertos/port.c:143

Rebooting...

closed 2020-07-10 by devUsama