espressif/esp-idf #5532: BLE MESH NET_BUF_PULL

MrCurio, 2020-07-02:
## Environment

- Development Kit:    ESP32-DevKitC
- Kit version DevKitC: [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version : v4.2-dev-1660-g7d7521367 and also tried with LATEST
- Build System: [Make]
- Compiler version (esp-2020r1)
- Operating System:  [Linux]
- Using an IDE?: [No|
- Power Supply: [USB|

## Problem Description

Hi everyone,
I am lately working with **ESP-BLE-MESH** and i am having problems with the Network Buffer Library (net_buf_simple variables) implementation. I want to use a network buffer in order to save some **changing** data (let's say a sensor reading) and i want to update this data constantly into the buffer. 
For doing so, i have read the documentation of Zephyr (ESP-IDF doesn't have) and i found out that i need to add the data to the buffer and then, pull to extract it. The problem appears when i try to pull from the buffer. It seems that pull operation is trying to act over a non-existing member of the struct net_buf_simple and, i get the following error: 

`esp-idf/components/bt/esp_ble_mesh/mesh_common/include/mesh_buf.h:1380:58: error: 'struct net_buf_simple' has no member named 'b'
 #define net_buf_pull(buf, len) net_buf_simple_pull(&(buf)->b, len)
`

Is this a bug?? 

Finally, i had to use : **net_buf_simple_pull(&sensor_data_0, sensor_data_0.len)** instead for making it work.

### Expected Behavior
Pulling data from the buffer and positioning the data pointer to the beginning of the buffer.
### Actual Behavior
Is not pulling data cause the compiler complains about the member 'b' of the implementation

### Steps to reproduce

1. Define a static Network Buffer -> NET_BUF_SIMPLE_DEFINE_STATIC(sensor_data_0, 1);
2. Add data to the end of the buffer -> net_buf_simple_add_u8(&sensor_data_0, indoor_temp++);
3. Pull data from the buffer -> net_buf_pull(&sensor_data_0, sensor_data_0.len);
4. Add again data to the end of the buffer -> net_buf_simple_add_u8(&sensor_data_0, indoor_temp++);

### Code to reproduce this issue

```cpp

NET_BUF_SIMPLE_DEFINE_STATIC(sensor_data_0, 1);

void app_main(){
      net_buf_simple_add_u8(&sensor_data_0, indoor_temp++);
      parse_buf(&sensor_data_0); //Displays buffer information 
      net_buf_pull(&sensor_data_0, sensor_data_0.len);
      parse_buf(&sensor_data_0); //Displays buffer information 
      net_buf_simple_add_u8(&sensor_data_0, indoor_temp++);
      parse_buf(&sensor_data_0); //Displays buffer information 
}

```
## Debug Logs

```
/esp/esp-idf/components/bt/esp_ble_mesh/mesh_common/include/mesh_buf.h:1380:58: error: 'struct net_buf_simple' has no member named 'b'
 #define net_buf_pull(buf, len) net_buf_simple_pull(&(buf)->b, len)

/esp/esp-idf/examples/bluetooth/esp_ble_mesh/esp32-ble-mesh/nest_ble/sensor_server/main/main.c:824:5: note: in expansion of macro 'net_buf_pull'
     net_buf_pull(&sensor_data_0, sensor_data_0.len);

```

Thank you!
MrCurio

Alvin1Zhang, 2020-07-02:
Thanks for the detailed report and letting us know, we will look into. Thanks.

Campou, 2020-07-06:
Hi @MrCurio 

For ESP-BLE-Mesh, the `net_buf` related functions are mainly for stack internal use, not recommended to use these functions in the application layer.

Currenlty some of them are used in the examples, later I think we will remove these from examples and made these internal functions as private.

If you're using it now, please use `net_buf_simple_pull` instead of `net_buf_pull`.

Thanks.

MrCurio, 2020-07-20:
Hi @Campou , 
I have noticed that the `net_buf` is for stack internal use. 
Is there any alternative i could use in the application layer?
I am having troubles implementing these buffers when reading from sensors (usually end in a corrupted backtrace) when adding and pulling from the `net_buf.`

May i get some help?

Thank you

chegewara, 2020-07-20:
You can use this macro to create net_buf:
`NET_BUF_SIMPLE(10)`

MrCurio, 2020-07-20:
What does it change, from the point of view of memory and corruption problems, the use of `NET_BUF_SIMPLE()` in comparison with `NET_BUF_SIMPLE_DEFINE()`?

I've made some tests creating a task for reading sensors and updating their net_buf (by adding and pulling operations) and after a while it ends with a corruption problem related to UART. Seems like a problem in UART until i delete the part of the net_buf. After that works perfect.

Could you please provide a minimal/functional sketch where these buffers are filled and cleared constantly with changing data(sensor data or equivalent code operation)?

I'd really appreciate the example since I'm stuck with these buffers. It's basically part of the proof-of-concept I'm developing.

Thank you in advance

chegewara, 2020-07-20:
Sorry, i missed that part you are using `NET_BUF_SIMPLE_DEFINE()`

MrCurio, 2020-07-20:
Hi @chegewara,
Well, actually I'm trying all of them (static, non-static...) cause i always end up with corruption problems after 2-3 min running.

Campou, 2020-07-22:
Hi @MrCurio 

Just made some simple changes to the above code, not sure what exactly does the function `parse_buf` do on your side, so I just output the current sensor data, sorry.
```
NET_BUF_SIMPLE_DEFINE_STATIC(sensor_data_0, 1);

static void parse_buf(struct net_buf_simple *buf)
{
    ESP_LOG_BUFFER_HEX("sensor_data", buf->data, buf->len);
}

void app_main(void)
{
    uint8_t indoor_temp = 26;

    for (int i = 0; i < 5; i++) {
        net_buf_simple_add_u8(&sensor_data_0, indoor_temp++);
        parse_buf(&sensor_data_0); //Displays buffer information 

        net_buf_simple_pull_u8(&sensor_data_0);

        net_buf_simple_add_u8(&sensor_data_0, indoor_temp++);
        parse_buf(&sensor_data_0); //Displays buffer information 

        net_buf_simple_pull_u8(&sensor_data_0);
    }
}
```

And the following is log output:
[image]

In the struct `net_buf_simple`, only the `data` pointer and `len` will be changed when add/pull data to/from the buffer, the `size` and `__buf` are only changed during buffer initialization.

Thanks.

MrCurio, 2020-07-23:
Hi @Campou,
I just changed my `net_buf_pull` function for the `net_buf_simple_reset` following the example you provided and everything seems to work just fine!

I will close this issue :)

Thank you!

closed 2020-07-23 by MrCurio