espressif/esp-idf #5533: Get TSF Information in STA-Mode

hpfmn, 2020-07-02:
**Is your feature request related to a problem? Please describe.**

I'd like to synchronise a bunch of esp32 which are connected to an AP.

**Describe the solution you'd like**

It would be great to have a method to get the current TSF.

**Describe alternatives you've considered**

Maybe it would be possible to connect one node via ethernet and have a mesh with the others but seems unnecessary complicated.

**Additional context**

There is a ```wdev_get_tsf_time``` symbol in the wifi lib but it is not exposed via a header file.

liuzfesp, 2020-07-06:
Hi @hpfmn, could you let me know which Chip you are using? ESP32 or ESP32S2? 

For ESP32S2, we can provide an API to get TSF timer value of station.
For ESP32, we are NOT recommended to use TSF timer for station timer syncrynization.

hpfmn, 2020-07-06:
I'm using an ESP32. Why is there a difference? But we might use ESP32S2 in the future - so I guess having an API is always a thing.

closed 2020-08-03 by espressif-bot