espressif/esp-idf #5534: FreeRTOS task name length error

ammaree, 2020-07-02:
When FreeRTOS is configured for a short task name (< 12) a compile error is generated in tasks.c line 2038.
With line 2032 defining i as BaseType_t the range of values evaluate as -1 to 1 and hence any value for configMAX_TASK_NAME_LEN smaller than 12 will generate an error.

Changing the BaseType_t to UBaseType_t restricts the range of values to 0->1 and hence values for configMAX_TASK_NAME_LEN should work down to 5+1, although other task names like "esp_timer" already enforce a minimum length of 8+1.

Maybe a decision should be made to restrict names of ESP-IDF created tasks to 5+1 in length?

`
void vTaskStartScheduler( void )
{
BaseType_t xReturn;
BaseType_t i;

	/* Add the per-core idle tasks at the lowest priority. */
	for ( i=0; i<portNUM_PROCESSORS; i++) {
		//Generate idle task name
		char cIdleName[configMAX_TASK_NAME_LEN];
		snprintf(cIdleName, configMAX_TASK_NAME_LEN, "IDLE%d", i);
`

Dazza0, 2020-07-03:
> Changing the BaseType_t to UBaseType_t restricts the range of values to 0->1 and hence values for configMAX_TASK_NAME_LEN should work down to 5+1, although other task names like "esp_timer" already enforce a minimum length of 8+1.

I would make it +3 instead (to print a `uint8_t`) because there can be instances where multiple numbered instances of a task are created. I can change the Idle task `snprintf()` to use `%hu`.

> Maybe a decision should be made to restrict names of ESP-IDF created tasks to 5+1 in length?

I think that would be too short as some component names are longer than 5 characters (thus creating a task related to a component would already exceed that length). It's a trade-off between saving few bytes of memory vs unambiguous task naming. Is there a reason why `configMAX_TASK_NAME_LEN` needs to be shortened so much?

ammaree, 2020-07-03:
In order to fit a full task list with priorities, stack low and RTOS stats (ticks and xx.xx%) on a 320x240 display we need to keep task names to <= 10, preferably 8 characters.
>
T# | Pc/Pb |-TaskName-| S | LoStk|MCU| %Util| Ticks |
---|-------|----------|---|------|---|------|-------|
 1   | 22/22 | esp_timer   | B | 3,308|   0   |  2.37|    58M|
 2  | 24/24 | ipc0            | B |    468|   0   |  0.00|    36K|
 3 | 24/24 |ipc1      | B |   464| 1 |  0.00|    88K|
 5 |  1/ 1 |main      | B | 2,548| 0 |  1.82|    44M|
 6 |  0/ 0 |IDLE0     | P |   952| 0 | 82.80| 2,037M|
 7 |  0/ 0 |IDLE1     | P |   956| 1 | 26.10|   642M|
 8 |  1/ 1 |Tmr Svc   | B |   682| 0 |  0.11| 2,796K|
12 | 20/20 |sys_evt   | B | 1,440| 0 |  0.00|  3,810|
13 | 23/23 |wifi      | B | 4,672| 0 | 10.10|   248M|
14 | 18/18 |tiT       | B | 2,160| X | 15.41|   379M|
15 |  2/ 2 |SNTP      | B |   856| X |  0.00|    49K|
16 |  4/ 4 |HTTP      | B | 1,132| X |  0.39| 9,761K|
17 |  3/ 3 |TNET      | R |   808| X |  0.85|    21M|
18 |  5/ 5 |mqttTX    | B | 1,880| X |  0.12| 3,191K|
19 |  5/ 5 |mqttRX    | B | 1,736| X | 35.11|   863M|
20 |  8/ 8 |Actuator  | B |   772| X | 31.51|   775M|
21 |  7/ 7 |Events    | B | 1,424| X |  0.11| 2,896K|
22 |  0/ 0 |Sensors   | P |   900| 0 |107.41| 2,642M|
23 |  0/ 0 |GUI       | P | 7,080| 1 | 60.21| 1,481M|

ammaree, 2022-08-30:
@Dazza0 

Any progress on this, almost 1yr later?

closed 2024-02-14 by ammaree