espressif/esp-idf #5535: Error in installing requirements.txt in msys32

ROYMURLI, 2020-07-02:
Hi,
I am trying to install the requirements.txt using "pip install -r requirements.txt" for ESP-IDF in msys32 but getting this error.
```
$ pip install -r requirements.txt
failed to create process (C:\msys32\mingw32\bin\python2.exe "D:\dozee\dozee_compiler\msys32\mingw32\bin\pip-script.py" "install" "-r" "requirements.txt").
```

dobairoland, 2020-07-03:
Your Python seems to been broken. You can try it reinstall it with:
```
pacman -S mingw-w64-i686-python-pip mingw-w64-i686-python
```

You can also download a pre-ZIPped environment where everything is setup: [for IDF 4.0 or later](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started-legacy/windows-setup.html#toolchain-setup) or [for IDF 3.3 or earlier](https://docs.espressif.com/projects/esp-idf/en/v3.3.2/get-started/windows-setup.html#toolchain-setup).

ROYMURLI, 2020-07-03:
I did the same thing which is given in "https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started-legacy/windows-setup.html#toolchain-setup". Facing above problem. 
Also getting this error:
$ pacman -S mingw-w64-i686-python-pip mingw-w64-i686-python
error: target not found: mingw-w64-i686-python-pip
error: target not found: mingw-w64-i686-python

dobairoland, 2020-07-03:
I'm sorry I've forgot that you have the old MSYS. This should work:
```
pacman -S mingw-w64-i686-python2-pip mingw-w64-i686-python2
```

> I did the same thing which is given in "https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started-legacy/windows-setup.html#toolchain-setup". Facing above problem.

Here is the new environment which hasn't made it to the documentation yet: 
https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_esp2020r2_toolchain-20200601.zip

ROYMURLI, 2020-07-03:
Thanks, It worked but I got an error in "Installing build dependencies". The error message is attached here. I am trying to solve this. 
IDFv- v4.2-dev-1905-g625bd5eb1
[error.txt](https://github.com/espressif/esp-idf/files/4869259/error.txt)

dobairoland, 2020-07-03:
`pacman -S mingw-w64-i686-gcc` should install the compiler required for dependencies.

ROYMURLI, 2020-07-03:
Same error is coming with v4.2-dev-1905-g625bd5eb1. However it is working fine with v4.1-beta2.
[error1.txt](https://github.com/espressif/esp-idf/files/4869409/error1.txt)

dobairoland, 2020-07-03:
> Same error is coming with v4.2-dev-1905-g625bd5eb1. However it is working fine with v4.1-beta2.

Because v4.2 has more dependencies one of which has to be compiled.

If the compiler doesn't help then probably there is nothing else you could do but to update the packages: `pacman -Syu`. However, be warned that it might break other things.

The new ZIP i've linked was tested with v4.2.

ROYMURLI, 2020-07-03:
Its working with v4.1-beta. On compiling my code it is giving error for "sprintf"
'''
 error: 'Content-Type: application/js...' directive writing 32 bytes into a region of size between 1 and 8192 [-Werror=format-overflow=]
        sprintf(payload, "%sContent-Type: application/json\r\n",payload);

 note: 'sprintf' output between 33 and 8224 bytes into a destination of size 8192
        sprintf(payload, "%sContent-Type: application/json\r\n",payload);
'''
I think it is a compiler version mismatch. It was working fine earlier using very old msys2. How can I fix this?

dobairoland, 2020-07-07:
Not `gcc` but `xtensa-esp32-elf-gcc` is used. The right version is:
```
$ xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r2) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

All toolchain releases can be found [here](https://github.com/espressif/crosstool-NG/releases).

ROYMURLI, 2020-07-10:
Thanks, It worked.

closed 2020-07-10 by ROYMURLI