espressif/esp-idf #5539: MQTT takes too much to reconnect

jonaphael, 2020-07-03:
## Environment

- Development Kit:     ESP32-PICO-Kit
- Module or chip used:  ESP32-PICO-D4
- IDF version v4.0.1
- Build System:         idf.py
- Compiler version: 8.2.0
- Operating System:  Linux
- Using an IDE: Visual studio code.
- Power Supply: external

## Problem Description
The MQTT client connected to a local broker is taking to much to reconnects after wifi reconnection.

// Detailed problem description goes here.

### Steps to reproduce
make sure you have 2 access points with the same SSID and password, and the wifi clients reconnect to wifi after losing connection

1. start wifi connect to a station
2. connect MQTT client to the broker
3. get the BSSID where esp32 is connected and disconnect this router.
4 esp32 will connect to the other router - (after been connected and retrieved IP, connect the MQTT client)
5. it will take 1 minutes or more to connect, meanwhile, it will receive disconnects events

## Debug Logs

```
DEBUG: MQTT_TAG: Received disconnect event                                                                                                            
ERROR  TRANS_TCP: tcp_poll_write select error 113, errno = Software caused connection abort, fd = 57 
```
## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2020-07-06:
Thanks for reporting, we will look into.

ESP-Marius, 2020-07-06:
Hi @jonaphael 

I wasn't able to reproduce your issue. I tested on v4.0.1 using our MQTT TCP example and it seems to migrate to the new AP without any issues. After connecting and getting the new IP it takes about 5-10 seconds before the MQTT client reconnects with the broker, which is as expected since `network_timeout_ms` defaults to 10 000. 

Did you try to adjust  `network_timeout_ms` or are you running with the default value?

ESP-Marius, 2020-08-07:
@jonaphael Any updates? Did the issue resolve itself?

Alvin1Zhang, 2020-08-13:
@jonaphael Thanks for reporting, would you please help share if any updates for this issue? Thanks.

ESP-Marius, 2020-09-17:
Closing issue due to no feedback, feel free to reopen with new information if this is still an ongoing issue.

closed 2020-09-17 by ESP-Marius