espressif/esp-idf #5541: sending log to UDP with esp_log_set_vprintf() causes esp_restart() to hang

tommessick, 2020-07-04:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version v4.0.1
- Build System:         Make
- Compiler version (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 5V

## Problem Description
sending log to UDP with esp_log_set_vprintf() causes esp_restart() to hang

//Detailed problem description goes here.
I am working on a system that is used without a console connection.
It communicates via UDP.  I am using esp_log_set_vprintf() to send the
logs to the controlling PC via UDP.  If I call esp_restart() the system does not
reboot a stops responding to UDP.  A hardware reset is required to recover.
If I use esp_log_set_vprintf() to restore the serial logging the system resets as expected.
### Expected Behavior
No hanging.
### Actual Behavior
hanging
### Steps to reproduce

1. Run the program
2. From a Linux PC   netcat -u esp_ip_address 8765
3 The rest of this is via the netcat connection
3. Hit the return key to see the logging on the serial console
4.Type log
5, Hit the return key to see the logging on the netcat session
6. Type reboot
7. System is not responsive

// If possible, attach a picture of your setup/wiring here.
PC <-> WIFI <-> ESP32

### Code to reproduce this issue
https://gist.github.com/tommessick/ebfe58ba53022cb5e598a9049f42336b

This can be dropped in to the wifi/getting_started/station example.

Alvin1Zhang, 2020-07-06:
Thanks for reporting, we will look into.

liuzfesp, 2020-07-06:
Hi @tommessick, the root cause is `sendto` is called in LWIP task and leads to dead-lock.

The problem scenario:

- step1 - netcat type log
- step2 - netcat type reboot
- step3 - ESP32 STA receives reboot and calls esp_restart(), esp_restart() calls esp_wifi_stop() to stop the WiFi STA.
- step4 - esp_wifi_stop() calls ESP_LOGW(...) to display some warning message, it calls netlog()->sendto(...)-> ...,  sendto() will post a message to LWIP task and LWIP task finally calls WiFi API esp_wifi_internal_tx() to send the UDP, however, since the WiFi is stopping, another warning message will be generated in esp_wifi_internal_tx(), so the LWIP calls netlog()->sendto()->.... The problem is we should NOT call sendto() in LWIP task, otherwise the deadlock happens.  Even without the deadlock, the netlog() -> esp_wifi_internal_tx() -> display warning message -> netlog -> esp_wifi_internal_tx() -> display warning message ... ... will repeat until the stack overflow, it just like the ESP_LOGI is called in netlog(): the log itself trigger new log.

A workaround for about deadlock is NOT send to message is sendto() is called in LWIP task, the side-effect is that the LWIP can't display any log.

int netLog(const char *p, va_list args)
{   
    char buffer[256];
    int ret = 0;
    
    if (is_lwip_task()) {
        ets_printf("is lwip task, ignore\n");
        return 0;
    }
...
}

Maybe we you can find better and more complex way to avoid above issue.

liuzfesp, 2020-08-04:
Hi @tommessic, does the recommendation solve your issue? I would like to close this issue. Feel free to reopen it if you think the recommendation doesn't solve your issue.

closed 2020-08-04 by Alvin1Zhang

tommessick, 2020-08-04:
I fixed this in my application by setting the log back to the serial port before rebooting.  esp_restart() should do this.  If you won't do that at least you should note that this is a problem in the documentation for esp_log_set_vprintf().