espressif/esp-idf #5542: Support runtime change log level to higher verbosity level

AxelLin, 2020-07-05:
* The problem:
Currently the esp-idf only supports run-time lower verbosity level.
So if build with CONFIG_LOG_DEFAULT_LEVEL to NONE, it's not possible to see any log by using esp_log_level_set.
i.e. the CONFIG_LOG_DEFAULT_LEVEL also means the maximum log level.
However, this is a very common use case that people don't want to show any log messages in release firmware but would like to
enable log messages for debug when hitting issues.
In current esp-idf, the only resolution is to rebuild with a debug firmware with higher CONFIG_LOG_DEFAULT_LEVEL.

* The suggestion:
Add CONFIG_LOG_MAX_LEVEL setting so people can decide it at compile-time and allow run-time set log level up to CONFIG_LOG_MAX_LEVEL.
This way, it should be ok to set CONFIG_LOG_DEFAULT_LEVEL=NONE for release firmware and then enable log messages for debug.
i.e. To allow runtime change log level to higher verbosity level for debug problem.

chegewara, 2020-07-05:
This is already implemented, just naming is maybe confused to you.
`CONFIG_LOG_DEFAULT_LEVEL` does what you want to do with `CONFIG_LOG_MAX_LEVEL` and to control runtime you have `esp_log_level_set`. All you have to do is to add this line in `app_main()`:
`esp_log_level_set("*", ESP_LOG_NONE);`
and all logs in app are turned off.

AxelLin, 2020-07-05:
> This is already implemented, just naming is maybe confused to you.
> `CONFIG_LOG_DEFAULT_LEVEL` does what you want to do with `CONFIG_LOG_MAX_LEVEL` and to control runtime you have [...]
This is not what I'm asking for...
I want to build with CONFIG_LOG_DEFAULT_LEVEL=none, and also allows esp_log_level_set("*", ESP_LOG_ERROR);

see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/log.html#_CPPv417esp_log_level_setPKc15esp_log_level_t
Note that this function can not raise log level above the level set using CONFIG_LOG_DEFAULT_LEVEL setting in menuconfig.

chegewara, 2020-07-05:
When you set `CONFIG_LOG_DEFAULT_LEVEL=none` then logs are not included in binary file. It is meant to decrease binary size when you dont need then anymore.

> This is not what I'm asking for...
> I want to build with CONFIG_LOG_DEFAULT_LEVEL=none, and also allows esp_log_level_set("*", ESP_LOG_ERROR);
>
> see [...]

You can, but this wont make any difference since those logs are not part of binary, because default log level excluded them.

AxelLin, 2020-07-05:
> When you set `CONFIG_LOG_DEFAULT_LEVEL=none` then logs are not included in binary file. It is meant to decrease binary size when you dont need then anymore.

I'm aware about that.
So I would set CONFIG_LOG_MAX_LEVEL=ESP_LOG_DEBUG for debug build
and set CONFIG_LOG_MAX_LEVEL=ESP_LOG_WARN for release build.
And set CONFIG_LOG_DEFAULT_LEVEL=none for both cases.

> You can, but this wont make any difference since those logs are not part of binary, because default log level excluded them.

The problem of having CONFIG_LOG_DEFAULT_LEVEL=ESP_LOG_INFO and
calling esp_log_level_set("*", ESP_LOG_NONE) is it's too late to switch to ESP_LOG_NONE. 
I still observe some booting messages during boot.
e.g.
I (546) cpu_start: Pro cpu up.
I (546) cpu_start: Application information:
I (546) cpu_start: Project name:     my-app
I (548) cpu_start: App version:      1.2.3.4
I (553) cpu_start: Compile time:     Jul  5 2020 19:51:03
I (559) cpu_start: ELF file SHA256:  1ce23f685baf66d3...
I (565) cpu_start: ESP-IDF:          v4.2-dev-1905-g625bd5eb1806-dir
I (572) cpu_start: Starting app cpu, entry point is 0x40081950
....

projectgus, 2020-07-06:
Hi @AxelLin,

Thanks for explaining the request. Agree this would be useful. There are some considerations in implementing this, but I think we can probably find a way to make everything work smoothly.

I'm afraid that I can't give you an ETA on this feature, but we will look into it.

Angus

AxelLin, 2020-07-14:
hi @projectgus 

I though this could be quite easy:
- **At compile time**: in menuconfig, set the verbosity level using the option :envvar:`CONFIG_LOG_DEFAULT_LEVEL`. All logging statements for verbosity levels higher than :envvar:`CONFIG_LOG_DEFAULT_LEVEL` will be removed by the preprocessor.

So define CONFIG_LOG_MAX_LEVEL to make preprocessor check CONFIG_LOG_MAX_LEVEL instead of CONFIG_LOG_DEFAULT_LEVEL, then it should work.

AxelLin, 2021-02-01:
> Hi @AxelLin,
> 
> Thanks for explaining the request. Agree this would be useful. There are some considerations in implementing this, but I think we can probably find a way to make everything work smoothly.

Hi @projectgus 
It has been quite long time since I created this issue.
As my previous reply, I thought this is not too difficult, but I might be wrong.
Just wondering if any update?

projectgus, 2021-03-11:
@AxelLin Thanks for being patient, this does look like it's relatively small amount of work but we're also pretty busy around here. :). 

The good news is that I have an implementation of this working, and it's currently in internal review. There are some tradeoffs so we might decide it needs further work, but hopefully not too long before it's available.

closed 2021-05-04 by espressif-bot

projectgus, 2021-05-05:
Thanks for being patient!

This option is available here now:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-log-maximum-level

ClaesIvarsson, 2023-02-16:
[boot.log](https://github.com/espressif/esp-idf/files/10753894/boot.log)
boot with verbose log level