espressif/esp-idf #5544: Any suggestion where to look on crash in interrupt handler?

cranphin, 2020-07-05:
## Environment

- Development Kit:      Generic ESP32 board
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.2-dev-1905-g625bd5eb1
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

Hi! I'm hoping to just get a push in the right direction, I have an interrupt handler calling c++ classes, 
all used methods are defined with IRAM_ATTR (I want this part to be fast).
There's also plenty of other stuff going on.
Whenever the interrupt handler is called a lot, the whole ESP32 reboots, it always gives an identical stack trace (see below).
Error is LoadProhibited, and EXCVADDR is always 0x800830f4 (does that address exist? It's always the same).
The line in the stack trace is where one class method calls another, which seems strange.

I'm hoping someone can give me a hint where to look for, as I'm not sure what kind of fault this is even, corrupt memory,
ISR calling flash when it is in use, something else?

Thanks!

(I will make a more minimal sample,  but it all takes time.. :) )

### Code to reproduce this issue

https://github.com/cranphin/MyEsp32Idf/blob/4c29699ce43729500a5d332732862c81a4d9cc55/main/RcReceiver.h
https://github.com/cranphin/MyEsp32Idf/blob/4c29699ce43729500a5d332732862c81a4d9cc55/main/RcReceiver.cpp
https://github.com/cranphin/MyEsp32Idf/blob/4c29699ce43729500a5d332732862c81a4d9cc55/main/main.cpp#L71

## Debug Logs

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40082aba  PS      : 0x00050c31  A0      : 0x800830e8  A1      : 0x3ffb0780  
0x40082aba: _xt_lowint1 at /opt/esp/idf/components/freertos/xtensa/xtensa_vectors.S:1105

A2      : 0x00020000  A3      : 0x00038000  A4      : 0x000637ff  A5      : 0x3ffb0760
A6      : 0x00000002  A7      : 0x00000002  A8      : 0x800d5812  A9      : 0x4008a00c  
0x4008a00c: _frxt_int_enter at /opt/esp/idf/components/freertos/xtensa/portasm.S:119

A10     : 0x00000000  A11     : 0x00060021  A12     : 0x3ff44000  A13     : 0xffffff00  
A14     : 0x3c79a80b  A15     : 0x3ffc1008  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x800830f4  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace:0x40082ab7:0x3ffb0780 0x400830e5:0x3ffbd320 0x400d58be:0x3ffbd340 0x4008a6cd:0x3ffbd360 0x40089d45:0x3ffbd380
0x40082ab7: _xt_lowint1 at /opt/esp/idf/components/freertos/xtensa/xtensa_vectors.S:1105
https://github.com/espressif/esp-idf/blob/625bd5eb1806809ff3cc010ee20d1f750aa778a1/components/freertos/xtensa/xtensa_vectors.S#L1105

0x400830e5: RcReceiver::handleInterrupt(unsigned int) at /project/build/../main/RcReceiver.cpp:74

0x400d58be: esp_vApplicationIdleHook at /opt/esp/idf/components/esp_common/src/freertos_hooks.c:63
https://github.com/espressif/esp-idf/blob/625bd5eb1806809ff3cc010ee20d1f750aa778a1/components/esp_common/src/freertos_hooks.c#L63

0x4008a6cd: prvIdleTask at /opt/esp/idf/components/freertos/tasks.c:3385 (discriminator 1)
https://github.com/espressif/esp-idf/blob/625bd5eb1806809ff3cc010ee20d1f750aa778a1/components/freertos/tasks.c#L3385

0x40089d45: vPortTaskWrapper at /opt/esp/idf/components/freertos/xtensa/port.c:143
https://github.com/espressif/esp-idf/blob/625bd5eb1806809ff3cc010ee20d1f750aa778a1/components/freertos/xtensa/port.c#L143

```

Wielebny666, 2020-07-05:
Maybe handleInterrupt is called before the previous one is completed.

Shouldn't be called at the end?
```
         if (xHigherPriorityTaskWoken == pdTRUE) {
             portYIELD_FROM_ISR ();
         }
```

Why do you have so many calculations in the interrupt procedure?
Send the necessary information to Queue and process it in the task.

cranphin, 2020-07-05:
> Maybe handleInterrupt is called before the previous one is completed.
>
> Shouldn't be called at the end?
>
> ```
>          if (xHigherPriorityTaskWoken == pdTRUE) {
>              portYIELD_FROM_ISR [...]

Hmm, mayby portYIELD_FROM_ISR   is related to my problem.
FreeRTOS manual says it's port depended if you may call it only at the end of a function or not.
But I'm not sure what the rules are with nested functions, should you always call it from the topmost function?
I will experiment :)

I know I calculate a lot in a interrupt handler, but it's still pretty fast on a esp32, sending to a queue also has overhead, and with lots of signals coming in (every rise/drop in a 443 remote signal), filtering a lot of the data in the isr seemed more efficient.
It might not be a good idea, but it has worked well in the past (even on esp8266 :) ).

I don't think handleInterrupt is called before the previous one is done, I don't think a new interrupt may interrupt the method handling the previous interrupt, instead only the flag is set to re-call the method directly after (?)

cranphin, 2020-07-06:
Hmm, well tried moving portYIELD_FROM_ISR to the end of the outermost function (first to be called to handle the interrupt,
and I still get the same result:
'''
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40082aba  PS      : 0x00050a31  A0      : 0x800830f2  A1      : 0x3ffb0780
0x40082aba: _xt_lowint1 at /opt/esp/idf/components/freertos/xtensa/xtensa_vectors.S:1105

A2      : 0x00020000  A3      : 0x00038000  A4      : 0x000637ff  A5      : 0x3ffb0760  
A6      : 0x00000002  A7      : 0x00000002  A8      : 0x3ffc1008  A9      : 0x00000000
A10     : 0x3ffc10c4  A11     : 0x3ffc0fa8  A12     : 0x3ff44000  A13     : 0xffffff00  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x800830fe  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace:0x40082ab7:0x3ffb0780 0x400830ef:0x3ffbd320 0x400d58be:0x3ffbd340 0x4008a6d9:0x3ffbd360 0x40089d51:0x3ffbd380
0x40082ab7: _xt_lowint1 at /opt/esp/idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x400830ef: RcReceiver::handleInterrupt(unsigned int, int&) at /project/build/../main/RcReceiver.cpp:74

0x400d58be: esp_vApplicationIdleHook at /opt/esp/idf/components/esp_common/src/freertos_hooks.c:63

0x4008a6d9: prvIdleTask at /opt/esp/idf/components/freertos/tasks.c:3385 (discriminator 1)

0x40089d51: vPortTaskWrapper at /opt/esp/idf/components/freertos/xtensa/port.c:143
'''

Manually inlining handleDuration (the  method which is called at the place the crash occurs) does make a difference though,
in that case the call to the method previously calling it (handleInterrupt) becomes the error location.

So I guess something funny either entering or leaving the method, I guess a trashed stack would be one option.

Dazza0, 2020-07-07:
@cranphin Usually, return address are stored by the CPU as `0x8...` but actually point to `0x4...` for hardware reasons. Thus, knowing what `0x400830f4` points to would be helpful.

Could you run the following
```
xtensa-esp32-elf-addr2line -e ./build/your_project_name.elf -a 0x400830f4
```

cranphin, 2020-07-07:
> 
> 
> @cranphin Usually, return address are stored by the CPU as `0x8...` but actually point to `0x4...` for hardware reasons. Thus, knowing what `0x400830f4` points to would be helpful.
> 
> Could you run the following
> 
> ```
> xtensa-esp32-elf-addr2line -e ./build/your_project_name.elf -a 0x400830f4
> ```

Ohhh! Usefull to know!
A few compiles further, so the address changed slight, seems to point here:

root@c699ac45ca5d:/project# xtensa-esp32-elf-addr2line -e ./build/MyEsp32Idf.elf -a 0x400830fe
0x400830fe
/project/build/../main/doorbell_recv.cpp:62

I'll see if I can figure out what that might mean, thanks! :)

cranphin, 2020-07-12:
Well, made a smaller example on a branch, which still shows the same issue.
Thought wifi/sdcard/spi display might be interfering, but there's really almost just the RC code left now:
https://github.com/cranphin/MyEsp32Idf/tree/d10f4883d3663bda7386fd192050a479c28105ee

Stacktrace is still mostly identical:
```
Core  0 register dump:
PC      : 0x4008256a  PS      : 0x00050e31  A0      : 0x80082b9e  A1      : 0x3ffb0780  
0x4008256a: _xt_lowint1 at /opt/esp/idf/components/freertos/xtensa/xtensa_vectors.S:1105

A2      : 0x00000040  A3      : 0x00018040  A4      : 0x000637ff  A5      : 0x3ffb0760  
A6      : 0x00000002  A7      : 0x00000002  A8      : 0x800d287e  A9      : 0x40084e34
0x40084e34: _frxt_int_enter at /opt/esp/idf/components/freertos/xtensa/portasm.S:119

A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ff44000  A13     : 0xffffff00  
A14     : 0x00000000  A15     : 0x3ffb0730  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x80082baa  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace:0x40082567:0x3ffb0780 0x40082b9b:0x3ffb6270 0x400d2927:0x3ffb6290 0x400854ee:0x3ffb62b0 0x40084b75:0x3ffb62d0
0x40082567: _xt_lowint1 at /opt/esp/idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x40082b9b: RcReceiver::handleInterrupt(unsigned int, int&) at /project/build/../main/RcReceiver.cpp:74

0x400d2927: esp_vApplicationIdleHook at /opt/esp/idf/components/esp_common/src/freertos_hooks.c:63

0x400854ee: prvIdleTask at /opt/esp/idf/components/freertos/tasks.c:3385 (discriminator 1)

0x40084b75: vPortTaskWrapper at /opt/esp/idf/components/freertos/xtensa/port.c:143

root@c32f84216d9a:/project# xtensa-esp32-elf-addr2line -e ./build/MyEsp32Idf.elf -a 0x40082baa
0x40082baa
/project/build/../main/doorbell_recv.cpp:64
```

Mentioned lines:
https://github.com/cranphin/MyEsp32Idf/blob/d10f4883d3663bda7386fd192050a479c28105ee/main/RcReceiver.cpp#L74
https://github.com/cranphin/MyEsp32Idf/blob/d10f4883d3663bda7386fd192050a479c28105ee/main/doorbell_recv.cpp#L64

I'll keep removing parts :)

ESP-Marius, 2024-07-08:
User deleted account, so marking this one as closed.

closed 2024-07-08 by ESP-Marius